espressif/arduino-esp32 #3519: HTTPClient not verifying CA-Certs when using deprecated API

plehmkuhl-ampere, 2019-11-26:
### Description:
Our project was using the now deprecated methods of HTTPClient to perform requests. 
Whe have noticed the warning `[I][ssl_client.cpp:159] start_ssl_client(): WARNING: Use certificates for a more secure communication!` in our logs even though we provided a CA certificate. 

After investigation we found the culprit of this problem:

HTTPClient::connect establishes the connection to the server prior to setting the certificates to the underlying WiFiClientSecure instance. 

```cpp
    if (!_client) {
        log_d("HTTPClient::begin was not called or returned error");
        return false;
    }

    if(!_client->connect(_host.c_str(), _port, _connectTimeout)) {  // <----- Connect
        log_d("failed connect to %s:%u", _host.c_str(), _port);
        return false;
    }

    // set Timeout for WiFiClient and for Stream::readBytesUntil() and Stream::readStringUntil()
    _client->setTimeout((_tcpTimeout + 500) / 1000);	

    log_d(" connected to %s:%u", _host.c_str(), _port);

#ifdef HTTPCLIENT_1_1_COMPATIBLE
    if (_tcpDeprecated && !_transportTraits->verify(*_client, _host.c_str())) { // <----- Setting of CAs inside of TLSTraits::verify
        log_d("transport level verify failed");
        _client->stop();
        return false;
    }	
#endif
```
After moving the block before the connect the certificate got validated as expected.

Since we have decided to update our calls to the newer APIs I just wanted to share this here. :)

closed 2020-02-08 by stale[bot]

pranavb25, 2020-03-08:
@plehmkuhl-ampere I also noticed the same behaviour. However the latest version of HttpClient still has the same issue. Could you please clarify which are the new APIs that you refer to.

plehmkuhl-ampere, 2020-03-09:
@pranavb25 Sure! We are using the `bool HTTPClient::begin(WiFiClient &client, String url)` method and assign the certificate beforehand to the WiFiClient instance you have to provide there.