espressif/arduino-esp32 #3520: Two strange reboot issues when using a GSM Modem

TomoLar51, 2019-11-27:
Board:							ESP32-WROOM-32 custom board
IDE name:						Platform.io

Hi.

I see two strange reboot issues in my program.
I think the problems are related to RF Noise from an onboard GSM modem, because it happens when the modem communicates.

The modem is connected to my ESP-WROOM-32 using HardwareSerial on PIN 21 and 22.

```
HardwareSerial SerialUART1(1);
SerialUART1.begin(115200, SERIAL_8N1, 21, 22);
```

The modem is powered from its own power supply (24V->5V)
The ESP32 is powered from its own power supply (24V->3.3V)
24V are common power supply.
The modem is sitting on an adapter on top of our main board.

RX0/TX0 and RX2/TX2 are used for other things.

The first issue I see is this:

```
Modem                       : Reset to factory defaults
AT&F0
OK
Modem                       : Success

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

------------------------------------------------------------------
------------------------------ BOOT ------------------------------
------------------------------------------------------------------
ESP32 SDK                   : v3.2.3-14-gd3e562907
System Reset Action CPU 0   : POWERON_RESET
System Reset Action CPU 1   : EXT_CPU_RESET
------------------------------------------------------------------
```

The are no errors logged (Only modem status) by the ESP32 before this line:

```
ets Jun  8 2016 00:22:57
```

The program can run many hours without any problems, but suddenly it keeps rebooting 5-10 times, shortly after the modem is powered up.
Suddenly it can run again, without problems.

What can cause a reboot (POWERON_RESET) of the ESP32? 
I don't see a drop on the 3.3V, while the modem is communicating.
Does the reset reason codes tell you anything?

The second issue I sometimes see is this:	

```
Guru Meditation Error: Core  0 panic'ed (LoadStoreError) at pc=4015506e.

Using PlatformIO debugger and a JTAG adapter, I can see the PC is pointing to 
vhci_notify_host_send_available function, in the esp32-bt-lib.

<vhci_notify_host_send_available>
0x40155064	entry	a1, 32
0x40155067	l32r	a9, 0 <vhci_notify_host_send_available-0x8>
0x4015506a	mov.n	a2, a9
0x4015506c	l32i.n	a8, a9, 0
0x4015506e	l8ui	a10, a8, 34		(Stops on this line)
```

Bluetooth BLE was enabled, but no client connected at the time of crash.

Can this bluetooth function get disturbed by the GSM modem communication?

Thomas

TomoLar51, 2019-11-28:
Hi again.

I can se that the POWERON_RESET message, is equal to the message I se, when press the reset button attached to RST-EN pin.

By investigating our board, I can se that we are missing our ground capasitor on the RST-EN pin.
At the moment, it is only pulled up by a 10K resistor.

By reading this message from the esp32 hardware design guidelines:

**_To avoid reboots caused by external interferences, the CHIP_PU trace should be as short as possible and routed away from the clock lines. A pull-up resistor and a ground capacitor are highly recommended._**

The RF noise from the GSM modem, could maybe be enough to disturb this input.

I will now try to add this missing ground capasitor, and hopefully it will take care off the seen reboots.

Thomas

closed 2020-02-10 by stale[bot]

dhirajp15, 2020-10-01:
Hi ,

I am facing the exact same issue. Using GSM module and esp32 together , when GSM module tries to connect to Network the ESP32 reboots with following logs:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6336
ho 0 tail 12 room 4
load:0x40078000,len:11308
load:0x40080400,len:6680
entry 0x40080760

**Note:** This voltage fluctuation is observed at EN pin , while the VCC is does not show any fluctuation.( I assume it is RF noise) .
Also when I keep GSM antenna away from PCB , it does not reset.

[image]

Using two **47uF**  resistors at EN pin to suppress the noise : 

[image]

Please suggest if there is any better way to fix this issue