espressif/arduino-esp32 #3523: esp32 en button must be pressed for the serial to be recognized

neuberfran, 2019-11-28:
Hii,

My esp32 board need EN  button must be pressed for the serial to be recognized:

![20191127_225405](https://user-images.githubusercontent.com/38701397/69770819-4c5d0d80-
113f-11ea-9840-6adead2370e9.jpg)

Pls help me

TD-er, 2019-12-02:
By pressing that button at boot, you may prevent the ESP from booting, so it will consume less power right after powering on,  allowing the PC to recognize the serial chip.
Can you try using a better USB cable?

neuberfran02, 2019-12-02:
@TD-er I already changed the usb cable and it did not solve. I also put a 0.22uF ceramic capacitor between EN and GND and it didn't solve either. **Please, what else can I try**?
 
[image]

neuberfran, 2020-01-31:
No solved

neuberfran, 2020-03-31:
not solved yet

atanisoft, 2020-03-31:
@neuberfran No fix will be provided in this code base as it would seem you have a hardware issue with the board you are trying to use.

closed 2020-06-13 by stale[bot]