espressif/arduino-esp32 #3524: The access modifier of String::concat(const char*, unsigned int) has a difference with ESP8266 core.

Hieromon, 2019-11-28:
It has been rearranged to the public in esp8266 but remains with the protected in esp32 WString.h.
Is this going to be changed for compatibility with ESP8266?

Hieromon, 2020-01-17:
No one mentions this topic, but is the gap with the API of ESP8266 core will not dissolve?

atanisoft, 2020-01-17:
Feel free to submit a PR to expose this.

Hieromon, 2020-01-21:
I've further verified the difference for the interface of `String::concat` between ESP32 arduino-core and the [Arduino language reference](https://www.arduino.cc/reference/en/language/variables/data-types/string/functions/concat/).
`String::concat(const char*, unsigned int)` is not stipulated with the Arduino String. Consequently,  the definition in esp32 conforms to the Arduino reference, rather esp8266 is the exception.
An implementation of `String::concat` as an esp32 arduino core is appropriate.
So, closed.

closed 2020-01-21 by Hieromon