espressif/arduino-esp32 #3525: mqtt_client library name conflict with esp-idf v3.3

john-yan, 2019-11-28:
**What you are trying to do?**
Use mqtt_client library from ESP-IDF v3.3

**Describe your system( Hardware, computer, O/S, core version, environment).**
Mac, macOS Catalina. ESP-IDF v3.3.

**Describe what is failing.**
Compilation is failing

**Show the shortest possible code that will duplicate the error.**
Whatever code that intend to use `#inclide "mqtt_client.h"`

**Show the EXACT error message(it doesn't work is not enough).**
I lost track the exact message because it was intermittent as explain below.
The error was complaint about definitions like `esp_mqtt_event_handle_t` not found.

The reason for this failure is caused by the library name conflicts between esp32-arduino and esp-idf.
There are 2 `mqtt_client.h ` files:
1. `esp-idf-3.3/components/mqtt/esp-mqtt/include/mqtt_client.h`
2. `arduino/tools/sdk/include/mqtt/mqtt_client.h`

When I intent to include the 1st library, the 2nd library was actually included because the library names are exactly the same. The failure is intermittent depended on which .h file was included.

lbernstone, 2019-11-28:
arduino-esp32 includes compiled libraries from esp-idf.  It is not intended that you use those libraries directly.  If you want to use the mqtt_client from idf 3.3, you can use the compiled libraries available in the idf-release/v3.3 branch of the repository.

john-yan, 2019-11-28:
Hello @lbernstone, thanks for your help!

I am currently using esp32-arduino as a component by following https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md. The doc didn't mention anything on how to use the prebuilt libraries. Do you have any pointer/doc on how to do that?

lbernstone, 2019-11-28:
Try https://github.com/espressif/arduino-esp32/blob/master/docs/lib_builder.md
The copy-to-arduino.sh script puts everything in the right place, and then it should be the default used when you call a `#include<>`.

john-yan, 2019-11-28:
Hello @lbernstone, I mainly use the ESP-IDF environment but I also need to use a few Arduino libraries so I included arduino-esp32 as a ESP-IDF component. I am not quite use the Arduino IDE or Arduino environment. This tool will put the libraries in the Arduino setup, which is not quite what I am looking for.

chegewara, 2019-11-28:
Did you try to disable in menuconfig mqtt library loaded from arduino?
There is option to enable/disable which libraries are used from arduino and if i remember by default all are used.

john-yan, 2019-11-28:
hello @chegewara, thanks for helping! First of all, I need to point out that in my original PR, I mentioned that the wrong `mqtt_client.h` to include was `components/arduino/tools/sdk/include/mqtt/mqtt_client.h` but it was actually `components/arduino/libraries/AzureIoT/src/az_iot/umqtt/inc/azure_umqtt_c/mqtt_client.h`. I updated my PR. And yes, your solution did work after disable `AzureIoT` in `make menuconfig`. But would that be nice to rename the file to avoid this trap? It really takes a while to figure out what went wrong with this.

chegewara, 2019-11-28:
Working with arduino as component is not easy, many traps. Especially if you want to make platformio project with arduino and esp-idf frameworks (i know its not the same, but similar).

john-yan, 2019-11-28:
Indeed! Not sure if someone will fix or improve this. Should I close this PR?

closed 2020-02-11 by stale[bot]