espressif/arduino-esp32 #3530: MDNS connection issue

suvarna84, 2019-11-29:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:	                1.0.4
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                      no
Upload Speed:					921600
Computer OS:                                         Windows 10

### Description:
I am trying to connect to mdns, it is connecting in my pc but if i try to connect in some other pc's it is not connecting, showing 'This site can't be reached'.
   It is connecting via IP in all pc's  without any problem but when i changed address to DNS name(eg: name.local) it is not connecting in other pc's.

sansillusion, 2019-11-29:
If I remember correctly you need to have apple Bonjour service installed.

enggsajjad, 2020-02-04:
I am also not able to connect using mDNS i.e. esp32.local on Heltec ESP32 board

ernfel, 2020-03-06:
I experience the same issue. With board version 1.0.3 everything works fine:
The device can be accessed by e.g. name.local
After changing to v1.0.4, the mdns is not working anymore

closed 2020-05-19 by stale[bot]