espressif/arduino-esp32 #3537: ESP32 BLE 2018 V 2019 boards

davejel, 2019-12-03:
i have developed a BLE project to the point where i thought it was completed
to test it fully i needed more boards which i bought from Ebay, my original boards were bought in 2018 the new boards late 2019.
using the 2018 boards both server and client function perfectly and exactly as expected.
using a 2018 server and 2019 client i run into problems, 
firstly the client can no longer connect to the server and for a long time the advertised device callback reports nothing. then after several minutes usually about 10 the client callback will report finding the server once, no attempt is made at connection ant the esp locks until reset is pressed
my code for the client is:
`
#include "BLEDevice.h"

static BLEAdvertisedDevice*     myDevice;
// custom service 1 PickleService
static BLEUUID    							PickleService("94698a03-dbb1-42b2-acb6-cc1abdffac35");
// custom characteristic 1 PickleChar
static BLEUUID    							PickleChar("94698a04-dbb1-42b2-acb6-cc1abdffac35");
static BLERemoteCharacteristic* pPickleChar;

static BLEScan* 								pBLEScan;

std::string											PicksAddress = "a4:cf:12:26:10:d6";
std::string                 		deviceAddress;
std::string											serviceUUID;
long														sTmr;

class MyClientCallback : public BLEClientCallbacks {
  void onConnect(BLEClient* pclient) {
		connected = true;
		doScan = false;
	}

  void onDisconnect(BLEClient* pclient) {
    connected = false;
		doScan = true;
    Serial.println("onDisconnect");
  }
};
//================================================

static void notifyCallback(
  BLERemoteCharacteristic* pBLERemoteCharacteristic,
  uint8_t* pData,
  size_t length,
  bool isNotify) {
    pData[length] = '\0';
		NData = (char*)pData;
    notified = true;
}
//================================================
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
  void onResult(BLEAdvertisedDevice advertisedDevice) {
    Serial.print("BLE Advertised Device found: ");
    Serial.println(advertisedDevice.toString().c_str());

    // We have found a device, let us now see if it contains the service we are looking for.
    if (advertisedDevice.haveServiceUUID() && advertisedDevice.isAdvertisingService(PickleService)) {

      BLEDevice::getScan()->stop();
      myDevice = new BLEAdvertisedDevice(advertisedDevice);
      doConnect = true;
      doScan = false;

    } else Serial.println("Not Picks UUID");
  } // onResult
}; // MyAdvertisedDeviceCallbacks

//================================================
bool connectToServer() {
  //  Serial.print("Forming a connection to ");
  //  Serial.println(myDevice->getAddress().toString().c_str());

    BLEClient*  pClient  = BLEDevice::createClient();
[... 132 lines cut ...]
[D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 41
[V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 41
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[I][BLEDevice.cpp:607] removePeerDevice(): remove: 0, GATT role client
[V][FreeRTOS.cpp:120] give(): Semaphore giving: name: RssiCmplEvt (0x3ffde928), owner: <N/A>
[V][FreeRTOS.cpp:120] give(): Semaphore giving: name: SearchCmplEvt (0x3ffde8c8), owner: <N/A>
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 2
[D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 2
[V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 1
[D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 1
[V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
`
if anyone can find a problem in my code causing this i would be VERY grateful and have less grey hairs at the end of it
otherwise is anyone experiencing or had a similar problem. on the espressif site there was a case where one user was experiencing WiFi problems with the 2019 boards another thought is that this might be a similar issue.

it is also worth mentioning that a server app on my phone does register with my client but not connect (as expected) so this seems to be an esp 'thing'
My server esp (2018 board) is also found and connected to by my phone and notify's  data ok
i would be very grateful for any help anyone can give
thanks in advance
Dave

chegewara, 2019-12-03:
Did you test new boards with smartphone if it is easy to find with ble app and/or connect to?
Maybe its issue with hardware and antenna.

davejel, 2019-12-04:
@chegewara yes i tested the new boards with smartphone ans is easy to find / connect
i bought 4 of these boards and all do exactly the same so is not probable that it is component failure, the boards do not look any different in component placement but it might be like you say that there is an issue with hardware or antenna this is the reason i put it here to see if anyone else has found similar problems
i will also put on espressif board to see if anyone there comes back with anything

davejel, 2019-12-04:
@chegewara 
just a thought, is there any way i can increase the server transmit power to see if the client will respond to the increased power level

chegewara, 2019-12-04:
> BLEDevice::init("Middle Server"); // create the server
BLEDevice::setPower(ESP_PWR_LVL_P7);

Yes, this should works, you can test with old boards that are connecting and nRF connect app.

davejel, 2019-12-04:
@chegewara 
thanks for your help, i have already tried and the results are the same, i think i have to return my new boards to the supplier

closed 2019-12-04 by davejel