espressif/arduino-esp32 #3539: general question: preprocessor directive for a board #define MACRO to identify both  esp32 and esp8266 Boards ?

dsyleixa, 2019-12-05:
which is the correct preprocessor directive for a board #define MACRO to identify both esp32 and esp8266 Boards ?

e.g.,, I need to declare the preprocessor directive
#if defined (SAM)
   #include "avr/dtostrf.h"
#elseif defined (? ESP ?) // <<<<<<<<<<<<< ????
  #include <stdlib_noniso.h>
#endif

atanisoft, 2019-12-05:
ESP32 for ESP32 and probably ESP8266 for the other....

dsyleixa, 2019-12-05:
is one ring to rule them all?

atanisoft, 2019-12-05:
Pretty much yes, there is a platform specific define created for each architecture, you can see it in platform.txt for each.

dsyleixa, 2019-12-05:
in boards.txt I don't see board macros which rule both ESP32 and ESP8266 boards

atanisoft, 2019-12-05:
boards.txt is not the same as [platform.txt](https://github.com/espressif/arduino-esp32/blob/master/platform.txt)...

dsyleixa, 2019-12-05:
in platform.txt I don't see a macro both for ESP32 and ESP8266 boards - do you?

atanisoft, 2019-12-05:
obviously you won't see ESP8266 in the ESP32 platform.txt, you will need to look at the ESP8266 platform.txt for that platform. This repo only covers ESP32.

dsyleixa, 2019-12-05:
no, I mean: a mcro which matches boh, all at once.

Like 
SAM
both covers __SAMD__ and __SAM3X8E__

atanisoft, 2019-12-05:
There isn't one

dsyleixa, 2019-12-05:
I'll then try
#elseif defined(ESP32) || defined(ESP8266)  // <<<<<<<<<<<<< ????

atanisoft, 2019-12-05:
I believe that should work, you will need to confirm with the ESP8266 side though to ensure what define is available for that platform.

closed 2020-02-17 by stale[bot]