espressif/arduino-esp32 #3540: Compiling parts of arduino-esp32 with IDF v4.1

jubueche, 2019-12-05:
Hi,

I am currently trying to compile a subset of the libraries on using the ESP-IDF v4.1.
It seems like cmake successfully finds the specified folders, but compilation is still unsuccessful.
Specifically, I am trying to use only SPI and BLE.

**components/arduino-esp32/CMakeLists.txt** :
```

set(CORE_SRCS
"${COMPONENT_DIR}/cores/esp32/base64.cpp"
"${COMPONENT_DIR}/cores/esp32/cbuf.cpp"
"${COMPONENT_DIR}/cores/esp32/esp32-hal-adc.c"
"${COMPONENT_DIR}/cores/esp32/esp32-hal-bt.c"
"${COMPONENT_DIR}/cores/esp32/esp32-hal-cpu.c"
"${COMPONENT_DIR}/cores/esp32/esp32-hal-dac.c"
"${COMPONENT_DIR}/cores/esp32/esp32-hal-gpio.c"
"${COMPONENT_DIR}/cores/esp32/esp32-hal-i2c.c"
"${COMPONENT_DIR}/cores/esp32/esp32-hal-ledc.c"
"${COMPONENT_DIR}/cores/esp32/esp32-hal-matrix.c"
"${COMPONENT_DIR}/cores/esp32/esp32-hal-misc.c"
"${COMPONENT_DIR}/cores/esp32/esp32-hal-psram.c"
"${COMPONENT_DIR}/cores/esp32/esp32-hal-sigmadelta.c"
"${COMPONENT_DIR}/cores/esp32/esp32-hal-spi.c"
"${COMPONENT_DIR}/cores/esp32/esp32-hal-time.c"
"${COMPONENT_DIR}/cores/esp32/esp32-hal-timer.c"
"${COMPONENT_DIR}/cores/esp32/esp32-hal-touch.c"
"${COMPONENT_DIR}/cores/esp32/esp32-hal-uart.c"
"${COMPONENT_DIR}/cores/esp32/esp32-hal-rmt.c"
"${COMPONENT_DIR}/cores/esp32/Esp.cpp"
"${COMPONENT_DIR}/cores/esp32/FunctionalInterrupt.cpp"
"${COMPONENT_DIR}/cores/esp32/HardwareSerial.cpp"
"${COMPONENT_DIR}/cores/esp32/IPAddress.cpp"
"${COMPONENT_DIR}/cores/esp32/IPv6Address.cpp"
"${COMPONENT_DIR}/cores/esp32/libb64/cdecode.c"
"${COMPONENT_DIR}/cores/esp32/libb64/cencode.c"
"${COMPONENT_DIR}/cores/esp32/main.cpp"
"${COMPONENT_DIR}/cores/esp32/MD5Builder.cpp"
"${COMPONENT_DIR}/cores/esp32/Print.cpp"
"${COMPONENT_DIR}/cores/esp32/stdlib_noniso.c"
"${COMPONENT_DIR}/cores/esp32/Stream.cpp"
"${COMPONENT_DIR}/cores/esp32/StreamString.cpp"
"${COMPONENT_DIR}/cores/esp32/wiring_pulse.c"
"${COMPONENT_DIR}/cores/esp32/wiring_shift.c"
"${COMPONENT_DIR}/cores/esp32/WMath.cpp"
"${COMPONENT_DIR}/cores/esp32/WString.cpp"
  )

set(LIBRARY_SRCS
  "${COMPONENT_DIR}/libraries/SPI/src/SPI.cpp")

set(BLE_SRCS
  "${COMPONENT_DIR}/libraries/BLE/src/BLE2902.cpp"
  "${COMPONENT_DIR}/libraries/BLE/src/BLE2904.cpp"
  "${COMPONENT_DIR}/libraries/BLE/src/BLEAddress.cpp"
  "${COMPONENT_DIR}/libraries/BLE/src/BLEAdvertisedDevice.cpp"
  "${COMPONENT_DIR}/libraries/BLE/src/BLEAdvertising.cpp"
  "${COMPONENT_DIR}/libraries/BLE/src/BLEBeacon.cpp"
  "${COMPONENT_DIR}/libraries/BLE/src/BLECharacteristic.cpp"
  "${COMPONENT_DIR}/libraries/BLE/src/BLECharacteristicMap.cpp"
  "${COMPONENT_DIR}/libraries/BLE/src/BLEClient.cpp"
  "${COMPONENT_DIR}/libraries/BLE/src/BLEDescriptor.cpp"
  "${COMPONENT_DIR}/libraries/BLE/src/BLEDescriptorMap.cpp"
  "${COMPONENT_DIR}/libraries/BLE/src/BLEDevice.cpp"
  "${COMPONENT_DIR}/libraries/BLE/src/BLEEddystoneTLM.cpp"
  "${COMPONENT_DIR}/libraries/BLE/src/BLEEddystoneURL.cpp"
  "${COMPONENT_DIR}/libraries/BLE/src/BLEExceptions.cpp"
  "${COMPONENT_DIR}/libraries/BLE/src/BLEHIDDevice.cpp"
  "${COMPONENT_DIR}/libraries/BLE/src/BLERemoteCharacteristic.cpp"
  "${COMPONENT_DIR}/libraries/BLE/src/BLERemoteDescriptor.cpp"
  "${COMPONENT_DIR}/libraries/BLE/src/BLERemoteService.cpp"
  "${COMPONENT_DIR}/libraries/BLE/src/BLEScan.cpp"
  "${COMPONENT_DIR}/libraries/BLE/src/BLESecurity.cpp"
  "${COMPONENT_DIR}/libraries/BLE/src/BLEServer.cpp"
  "${COMPONENT_DIR}/libraries/BLE/src/BLEService.cpp"
  "${COMPONENT_DIR}/libraries/BLE/src/BLEServiceMap.cpp"
  "${COMPONENT_DIR}/libraries/BLE/src/BLEUtils.cpp"
  "${COMPONENT_DIR}/libraries/BLE/src/BLEUUID.cpp"
  "${COMPONENT_DIR}/libraries/BLE/src/BLEValue.cpp"
  "${COMPONENT_DIR}/libraries/BLE/src/FreeRTOS.cpp"
  "${COMPONENT_DIR}/libraries/BLE/src/GeneralUtils.cpp"
  )

set(COMPONENT_SRCS ${CORE_SRCS} ${LIBRARY_SRCS} ${BLE_SRCS})

set(COMPONENT_ADD_INCLUDEDIRS
  "${COMPONENT_DIR}/variants/esp32/"
  "${COMPONENT_DIR}/cores/esp32/"
  "${COMPONENT_DIR}/libraries/BLE/src"
  "${COMPONENT_DIR}/libraries/SPI/src")

set(COMPONENT_PRIV_INCLUDEDIRS "${COMPONENT_DIR}/cores/esp32/libb64")

set(COMPONENT_REQUIRES spi_flash mbedtls mdns)
set(COMPONENT_PRIV_REQUIRES fatfs nvs_flash app_update spiffs bootloader_support openssl bt)

register_component()
```

I don't know if the " are needed, but it seemed like it when I saw how to set the parameters [here](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html#idf-py).

The error that I get is the following:
```
In file included from ../components/Chaze_Advertising/include/Chaze_Advertising.h:5,
                 from ../main/main.cpp:2:
../components/Chaze_ble/include/Chaze_ble.h:27:5: error: 'BLEServer' does not name a type; did you mean 'Server'?
     BLEServer *pServer = NULL;
     ^~~~~~~~~
     Server
../components/Chaze_ble/include/Chaze_ble.h:28:5: error: 'BLECharacteristic' does not name a type; did you mean 'Character_h'?
     BLECharacteristic * pTxCharacteristic;
     ^~~~~~~~~~~~~~~~~
     Character_h
../components/Chaze_ble/include/Chaze_ble.h:29:5: error: 'BLECharacteristic' does not name a type; did you mean 'Character_h'?
     BLECharacteristic * pRxCharacteristic;
     ^~~~~~~~~~~~~~~~~
     Character_h
../components/Chaze_ble/include/Chaze_ble.h:30:5: error: 'BLEService' does not name a type
     BLEService *pService;
     ^~~~~~~~~~
../components/Chaze_ble/include/Chaze_ble.h:36:21: error: 'std::string' has not been declared
     void write(std::string);
```

And the file
**components/Chave_ble/include/Chaze_ble.h**:
```
#ifndef CHAZE_BLE_H
#define CHAZE_BLE_H

#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>
#include <time.h>
#include <ChazeFlashtraining.h>

#include "Arduino.h"
#include "Configuration.h"

#define SERVICE_UUID           "6E400001-B5A3-F393-E0A9-E50E24DCCA9E" // UART service UUID
#define CHARACTERISTIC_UUID_RX "6E400002-B5A3-F393-E0A9-E50E24DCCA9E"
#define CHARACTERISTIC_UUID_TX "6E400003-B5A3-F393-E0A9-E50E24DCCA9E"
#define STACK_CONGESTION_TIMEOUT 20

class Chaze_ble {
public:
    //Constructor
    Chaze_ble();
    ~Chaze_ble();

    // Attributes
    
    BLEServer *pServer = NULL;
    BLECharacteristic * pTxCharacteristic;
    BLECharacteristic * pRxCharacteristic;
    BLEService *pService;
    uint8_t txValue;
    
    //Functions
    void initialize_connection();
    void write(uint8_t);
    void write(std::string);
    void write(uint8_t*,size_t);
    void write(char*,size_t);
    void advertise(void);
    bool is_initialized(void);
};

#endif
```

And the ```CMakeLists.txt``` for ```Chaze_ble```
**components/Chaze_ble/CMakeLists.txt**:
```
idf_component_register(SRCS "Chaze_ble.cpp"
                  INCLUDE_DIRS "include"
                  REQUIRES arduino-esp32 Configuration Chaze_Flashtraining)
```

Is it needed that I also use
```idf_component_register``` for **components/arduino-esp32/CMakeLists.txt**?

Has anyone successfully compiled this using the new version of the IDF?
I know that the authors are currently working on it.

h2zero, 2020-01-21:
Have you had any success with this? I’m looking to try doing the same.

robert-alfaro, 2020-01-26:
IDF v3.3 works great, any need specifically for 4+?

me-no-dev, 2020-01-26:
there is now a 4.0 branch. Actually two branches: 
- `idf-4.0-prep` contains all changes to the core and examples required to compile under 4.0
- `idf-release/v4.0` contains the above changes and also the 4.0 libs/sdk

jubueche, 2020-01-26:
Sounds good. I will try it out next week. @robert-alfaro I wanted to use the slimmer SSL module.

h2zero, 2020-01-27:
Nice, I wanted to try out nimble. I decided to back port it to the current arduino release and it’s working. 

I’m working on a rebuild of the BLE library to use nimble now. Thanks for the heads up so I can test with 4.0 too.

sujeendran, 2020-02-29:
> Nice, I wanted to try out nimble. I decided to back port it to the current arduino release and it’s working.
> 
> I’m working on a rebuild of the BLE library to use nimble now. Thanks for the heads up so I can test with 4.0 too.

@h2zero Hi, could you share the details on how to back port the NimBLE successfully with current arduino release? I tried the same for a long time and failed to compile it.

h2zero, 2020-03-02:
@sujeendran [Here](https://github.com/h2zero/NimBLE-Arduino) is a link to my repo for NimBLE. Just put it in your libraries folder and it should compile fine with the IDF examples. As you will see I have the cpp implemented for the client code that is compatible with the current ardruino BLE library (starting work on server now) with minor changes. Give it a try and post any issues you find.

beegee-tokyo, 2020-03-02:
@h2zero     
If you need tester for the NimBLE server let me know. Looking desperate for a replacement of the bulky ESP32 BLE library. But I need the server part.

h2zero, 2020-03-02:
Thanks, keep an eye on my repo and feel free to provide feedback there. Probably won’t have any server code ready for a few days.

closed 2020-05-15 by stale[bot]