espressif/arduino-esp32 #3546: When FFat is mounted before SD, then filesystem reports wrong system sizes

ullix, 2019-12-08:
I use an ESP32 in Arduino environment (more precise: UECIDE). I use both the internal Flash (about 1.4MB) with FFat and an external SD card (32GB) with SD. 

When I first mount the Flash with FFat, and then the SD with SD, all calls to totalBytes and freeBytes, usedBytes, resp., show only the values valid for FFat. 

When I reverse the order of mounting, so doing SD first, then all comes out correct.

(Note: FFat can't do usedBytes,  SD can't do freeBytes.)

```
#include "FS.h"
#include "FFat.h"
#include "SD.h"

void setup(){
    Serial.begin(115200);
    uint64_t totalB;    // total Bytes
    uint64_t freeB;     // free Bytes
    uint64_t usedB;     // used Bytes = total Bytes - free Bytes    
    uint64_t cardSize;  // only for SD: size of card

    //************ First ******************************************
    Serial.println("\nMounting FFat Flash");
    if (!FFat.begin(false)) {     
        Serial.println("An Error has occurred while mounting FFAT");
        return;
    }
    // end first **************************************************

    //************ Second *****************************************
    Serial.println("\nMounting SD card");
    if(!SD.begin()){
        Serial.println("Card Mount Failed");
        return;
    }
    //end second***************************************************
    
    Serial.println();
    
    totalB = SD .totalBytes();
    usedB  = SD .usedBytes();
    freeB  = totalB - usedB;   // no freeBytes in SD
    cardSize = SD.cardSize() / (1024 * 1024);    
    Serial.printf("SD: total: %llu,  free: %llu,  used: %llu, card size: %llu \n", totalB, freeB, usedB, cardSize);  
    
    totalB = FFat.totalBytes();
    freeB  = FFat.freeBytes();
    usedB  = totalB - freeB;    // no usedBytes in FFat
    Serial.printf("FFat: total: %llu,  free: %llu,  used: %llu \n", totalB, freeB, usedB);

    totalB = SD .totalBytes();
    usedB  = SD .usedBytes();
    freeB  = totalB - usedB;   // no freeBytes in SD
    cardSize = SD.cardSize() / (1024 * 1024);
    Serial.printf("SD: total: %llu,  free: %llu,  used: %llu, card size: %llu \n", totalB, freeB, usedB, cardSize);  
    
    totalB = FFat.totalBytes();
    freeB  = FFat.freeBytes();
    usedB  = totalB - freeB;    // no usedBytes in FFat
    Serial.printf("FFat: total: %llu,  free: %llu,  used: %llu \n", totalB, freeB, usedB);
}

void loop(){}

```

When the code is run as shown, the outcome is:
```

Mounting FFat Flash

Mounting SD card

SD: total: 1454080,  free: 4096,  used: 1449984, card size: 30735
FFat: total: 1454080,  free: 4096,  used: 1449984
SD: total: 1454080,  free: 4096,  used: 1449984, card size: 30735
FFat: total: 1454080,  free: 4096,  used: 1449984
```
Irrespective of whether SD or FFat is selected, only the FFat data are shown. 

When the sections in the code marked 'First' and 'Second' are switched, then the outcome is this:

```
Mounting SD card

Mounting FFat Flash

SD: total: 32211189760,  free: 31970983936,  used: 240205824, card size: 30735
FFat: total: 1454080,  free: 4096,  used: 1449984
SD: total: 32211189760,  free: 31970983936,  used: 240205824, card size: 30735
FFat: total: 1454080,  free: 4096,  used: 1449984
```

This is the correct result!

lbernstone, 2019-12-08:
Let me run some tests before I merge, but this should fix it.
https://github.com/lbernstone/arduino-esp32/commit/648dcca78d259238c3cb9c0b00f83890619b2f44

closed 2020-02-20 by stale[bot]

holgerlembke, 2023-08-13:
Needs to be reopened. Version, 2.11.

Report depends on FS.begin()-order:

FFat.totalbytes(): 1392640 FFat.usedbytes(): 49152
SD.totalbytes(): 1392640 SD.usedbytes(): 49152

SD.totalbytes(): 1015267328 SD.usedbytes(): 73269248
FFat.totalbytes(): 1015267328 FFat.usedbytes(): 73269248

```
// esp32 wrover module
#include <FS.h>
#include <SD.h>
#include <FFat.h>

void setup() {
  Serial.begin(115200);
  delay(1000);

  SD.begin(5);
  FFat.begin();

  Serial.print("FFat.totalbytes(): ");
  Serial.print(SD.totalBytes());
  Serial.print(" FFat.usedbytes(): ");
  Serial.println(SD.usedBytes());

  Serial.print("SD.totalbytes(): ");
  Serial.print(SD.totalBytes());
  Serial.print(" SD.usedbytes(): ");
  Serial.println(SD.usedBytes());
}

void loop() {}
```

holgerlembke, 2023-08-13:
Needs to be opened again. Version 2.0.11

ffat first:
FFat.totalbytes(): 1392640 FFat.usedbytes(): 49152
SD.totalbytes(): 1392640 SD.usedbytes(): 49152

sd first:
FFat.totalbytes(): 1392640 FFat.usedbytes(): 49152
SD.totalbytes(): 1015267328 SD.usedbytes(): 73269248

```
// esp32 wrover module
#include <FS.h>
#include <SD.h>
#include <FFat.h>

void setup() {
  Serial.begin(115200);
  delay(1000);

  SD.begin(5);
  FFat.begin();

  Serial.print("FFat.totalbytes(): ");
  Serial.print(FFat.totalBytes());
  Serial.print(" FFat.usedbytes(): ");
  Serial.println(FFat.usedBytes());

  Serial.print("SD.totalbytes(): ");
  Serial.print(SD.totalBytes());
  Serial.print(" SD.usedbytes(): ");
  Serial.println(SD.usedBytes());
}

void loop() {}
```