espressif/arduino-esp32 #3548: Incorrect values read using analogRead() on IO33 (but works with IDF adc example)

WayneKeenan, 2019-12-08:
### Hardware:
Board:							Adafruit
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        macOS 10.13.6

Other hardware:
I have a STM32 board running the standard `analogWrite` Arduino example, which generates a PWM sweep from 0 to 255, then 255 to 0 in a infinite loop, connected to IO33

Other info:
Toolchain version: esp32-2019r1
Compiler version: 8.2.0
ESP-IDF v4.0-beta1

### Description:

Performing an `anologRead()` on pin33 typically returns 0 or 4095, and not an expected range of values.
But, replacing the Arduino sketch with the IDF `adc` example, with exactly the same setup, works.

### Sketch: 

Standard Arduino `AnalogInSerialOut` sketch output:

```
sensor = 4095	 output = 1020
sensor = 0	 output = 0
sensor = 4095	 output = 1020
sensor = 0	 output = 0
sensor = 4095	 output = 1020
sensor = 4095	 output = 1020
sensor = 0	 output = 0
sensor = 4095	 output = 1020
sensor = 4095	 output = 1020
sensor = 0	 output = 0
sensor = 4095	 output = 1020
sensor = 0	 output = 0
sensor = 0	 output = 0
sensor = 4095	 output = 1020

```

ESP-IDF 4.0-beta1  'adc' example output:
```
Raw: 1791	Voltage: 496mV
Raw: 1919	Voltage: 527mV
Raw: 1855	Voltage: 512mV
Raw: 1473	Voltage: 422mV
Raw: 1023	Voltage: 316mV
Raw: 904	Voltage: 288mV
Raw: 959	Voltage: 301mV
Raw: 837	Voltage: 272mV
Raw: 772	Voltage: 257mV
Raw: 262	Voltage: 137mV
Raw: 264	Voltage: 137mV
Raw: 65	Voltage: 90mV
Raw: 5	Voltage: 76mV
Raw: 5	Voltage: 76mV
Raw: 207	Voltage: 124mV
Raw: 255	Voltage: 135mV
Raw: 383	Voltage: 165mV
Raw: 771	Voltage: 256mV
Raw: 959	Voltage: 301mV
Raw: 1153	Voltage: 346mV
Raw: 1151	Voltage: 346mV
Raw: 1026	Voltage: 316mV
Raw: 1407	Voltage: 406mV
Raw: 1733	Voltage: 483mV
Raw: 2115	Voltage: 573mV
```

WayneKeenan, 2019-12-08:
This also applies to IO32.   
I can see the PWM signal when scoping the Huzzah32's pin headers.

lbernstone, 2020-04-10:
Post your code.  Seems likely your problem is the sensor, not the analogWrite.

WayneKeenan, 2020-04-10:
I'll dig it up (think it was a built in example), but as I mention, just changing one thing (from Arduino sketch to the IDF adc example) shows it working, so there's something in the Arduino lib

WayneKeenan, 2020-04-10:
It was just ADC read sketch, nothing else, no wifi no bluetooth.

WayneKeenan, 2020-06-19:
What was attached to the pin wasn't a sensor, but a test invariant.

WayneKeenan, 2020-08-19:
No wifi involved.

closed 2020-11-01 by stale[bot]