espressif/arduino-esp32 #3551: POWER SLEEP

LDavidFr, 2019-12-09:
I tested lowpower ESP32 pico D4

My code 

```
>   esp_bt_controller_disable();
>   btStop();
>   //esp_wifi_stop(); // BUG
>   adc_power_off();
>   esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);
>   esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_SLOW_MEM, ESP_PD_OPTION_OFF);
>   esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_FAST_MEM, ESP_PD_OPTION_OFF);
>   esp_sleep_pd_config(ESP_PD_DOMAIN_XTAL, ESP_PD_OPTION_OFF);
>   Serial.println("Going to sleep now");
>   Serial.flush();
>   esp_deep_sleep_start();
```

The consumption is 800µA. I need real consumption ... 15µA ;)
You have idea ?

chegewara, 2019-12-09:
https://esp32.com/viewtopic.php?t=10222

LDavidFr, 2019-12-09:
> https://esp32.com/viewtopic.php?t=10222

> https://esp32.com/viewtopic.php?t=10222

LDavidFr, 2019-12-09:
I dont use LDO

closed 2020-02-21 by stale[bot]