espressif/arduino-esp32 #3552: esp_wifi_stop(); BUG

LDavidFr, 2019-12-09:
Hello 
I have bug if i use esp_wifi_stop();
Bug report :
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400e3624  PS      : 0x00060f30  A0      : 0x800d103d  A1      : 0x3ffca260  
A2      : 0x00000000  A3      : 0x3ffbf910  A4      : 0x3ffc2538  A5      : 0xa77bcc2b  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x800d2dfa  A9      : 0x3ffca250  
A10     : 0x00000014  A11     : 0xa77bcc2b  A12     : 0xa77bcc2b  A13     : 0x3ffc2538  
A14     : 0x3ffca20c  A15     : 0x3ffca20c  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000134  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400e3624:0x3ffca260 0x400d103a:0x3ffca290 0x400d46c7:0x3ffca2c0 0x4008e105:0x3ffca2e0

PC: 0x400e3624
EXCVADDR: 0x00000134

Decoding stack results
0x400d103a: setup() at /Users/david/Documents/Arduino/ExternalWakeUpTestlowpower/ExternalWakeUpTestlowpower.ino line 90
0x400d46c7: loopTask(void*) at /Users/david/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/main.cpp line 14
0x4008e105: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

lbernstone, 2019-12-09:
Please follow the [ISSUE TEMPLATE](https://github.com/espressif/arduino-esp32/blob/master/docs/ISSUE_TEMPLATE.md).  Not providing code is a surefire way to not get any answers.
The correct method in Arduino is `WiFi.mode(WIFI_OFF);`

LDavidFr, 2019-12-09:
> Please follow the [ISSUE TEMPLATE](https://github.com/espressif/arduino-esp32/blob/master/docs/ISSUE_TEMPLATE.md). Not providing code is a surefire way to not get any answers.
> The correct method in Arduino is `WiFi.mode(WIFI_OFF);`

Humm 
I tested WiFi.mode(WIFI_OFF);
But i have same problem : https://github.com/espressif/arduino-esp32/issues/3551

I have MacOS, ArduinoIDE 1.8.8., ESP32 Pico D4, i use ESP PICO Kit

lbernstone, 2019-12-09:
The radio uses way more than 800uA. You will know when it is on.  There is almost certainly some other device on your board using several 100s of uA.  Note that the low power measurements that you reference are for JUST the esp32.  It doesn't even include the power for a reference oscillator, flash, or other fairly necessary components.  Many dev boards power the regulator & uart adapter even if you are using a battery.

LDavidFr, 2019-12-09:
I understand for the radio.
But il have only esp32 D4 pico. Flash is embedded.
Simply PW > ESP32 i have not another composent (whitout LDO, UART etc ..).

closed 2020-02-21 by stale[bot]