espressif/arduino-esp32 #3554: Support for ethernet PHY IP101

dslc, 2019-12-09:
### Hardware:
Board:							ESP32-Ethernet-Kit V1.1
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						Arduino IDE

### Description:

I know that the IP101 ethernet phy is supported in the ESP IDF - but am hoping to use the Arduino library for a particular project. When testing the [ESP32 ethernet kit](https://docs.espressif.com/projects/esp-idf/en/latest/hw-reference/get-started-ethernet-kit.html I noticed that the IP101 is not officially supported in Arduino ESP32.

_However_ ... After making some trivial modifications to the ETH.h and [ETH.cpp](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/ETH.cpp) files, it seems that the SDK actually supports it. The sketch below - adapted from [this example](https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFi/examples/ETH_TLK110) works after the afore-mentioned tweaks.

Please see the patch attached.

I have only tested this change very briefly though. Is there a reason that `ETH_PHY_IP101` is not offered in `ETH.h`?

### Sketch:

Here is the sketch. It seems to work after applying the patch attached. (I have only tested it very briefly.)

The phy address for the IP101 on the ethernet kit is `1`.

```cpp

#include <ETH.h>

#define ETH_ADDR        1
#define ETH_POWER_PIN   5
#define ETH_TYPE        ETH_PHY_IP101

// snip

void setup()
{
  Serial.begin(115200);
  WiFi.onEvent(WiFiEvent);
  ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_PHY_MDC, ETH_PHY_MDIO, ETH_TYPE);
}
```
[phy_ip101.patch.txt](https://github.com/espressif/arduino-esp32/files/3941395/phy_ip101.patch.txt)

Blom1974, 2020-01-13:
Tested this and it works great! Saved me lots of headache.

DheeranYuvaraj, 2020-01-31:
Hi @dslc 

it works fine with Espressif Ethernet kit_V1.1

lathoub, 2020-03-09:
I made the changes to both ETH.cpp and ETH.h, but got a error: `'phy_ip101_default_ethernet_config' was not declared in this scope` 

`extern const eth_config_t phy_ip101_default_ethernet_config;` is declared in thy_ip101.h, but that it about it.

I'm using the Latest stable release espressif/arduino-esp32 v1.0.4 from October 2019

What am I missing?

Thank you

DheeranYuvaraj, 2020-03-10:
Hi @lathoub 
I'm attaching ETH(cpp) ETH(h) files and EthernetLAN_IP101(ino) Arduino example code. 

Try to use this. 

[ETH(cpp).txt](https://github.com/espressif/arduino-esp32/files/4310321/ETH.cpp.txt)
[ETH(h).txt](https://github.com/espressif/arduino-esp32/files/4310322/ETH.h.txt)
[EthernetLAN_IP101(ino).txt](https://github.com/espressif/arduino-esp32/files/4310323/EthernetLAN_IP101.ino.txt)

lathoub, 2020-03-12:
putting `randomSeed(analogRead(0));` before `ETH.begin` results in a `E (2086) emac: Reset EMAC Timeout` error

lathoub, 2020-03-14:
Assume analog pin 0 is **not free**! Don't use analog 0 as seed for randomiser.

Blom1974, 2020-05-13:
Thinking this should be merged into the SDK !

atanisoft, 2020-05-13:
@Blom1974 for it to be considered it needs to be submitted as a PR

closed 2020-07-26 by stale[bot]