espressif/arduino-esp32 #3557: How to properly send LOW/HIGH signals through pins

GeorgeFlorian, 2019-12-10:
Board: ESP32-EVB

Hello !

I am trying to emulate a 26-Bit Wiegand Protocol data transfer. I am basically trying to send a card information through the two data wires, W0 and W1, from the ESP32 to a station card reader to emulate a card being read.
I have hooked up a card reader that acts a station to an ESP32-EVB:
- GPIO13 is W0;
- GPIO17 is W1.

Both the ESP32 and the card reader have a common GND.

**Note:** I do not know if "station" is the right word for it, but this particular card reader can act as a Standalone Card Reader or as a Central for other card readers. One device will act as a card reader, sending the data about the card that has been read to a second device that acts as a station/central/system and determines if the card has access or not.

Using a [Wiegand Calculator](https://pripd.com/tools/cardCalculator.php) and a decimal number from a real access card ( 12679548 ), we obtain the 26-Bit binary number ( 01100000101111001011111000 ) that has to be sent to the W0 and W1 data wires.

Reading [from here](https://docs.tibbo.com/taiko/ser_wiegand_mode) we find out that between the sent bits there has to be a delay, that's in the vicinity of 100uS (microseconds) and that between card IDs there has to be a pause of 20-100ms (milliseconds).
```
#define W0 13
#define W1 17

change = false;

void setup() {
  Serial.begin(115200);
  pinMode(W0, OUTPUT);
  pinMode(W1, OUTPUT);
  digitalWrite(W0, HIGH);
  digitalWrite(W1, HIGH);
  // decimal number 12679548 transforms into Wiegand 26-Bit binary number: 
01100000101111001011111000
  // store the binary number into 'bool* wiegandArray = new bool[26];'
  // if a new card ID has come then change = true;
  int pulseWidth = 90;
}

void loop() {
  if(change) {
    for(int i = 0; i<26; i++) {
      if(wiegandArray[i] == 0) {
        digitalWrite(W0, LOW);
        delayMicroseconds(pulseWidth);
        digitalWrite(W0, HIGH);
      } else {
        digitalWrite(W1, LOW);
        delayMicroseconds(pulseWidth);
        digitalWrite(W1, HIGH);
      }
    }
  }
}
```

I have put the card reader device in Station mode and the wires are set as W0 and W1 inputs.

From my point of view the code is good and should be working but it's not.

What do you think ?

lbernstone, 2019-12-10:
The [RMT](https://docs.espressif.com/projects/esp-idf/en/v3.2/api-reference/peripherals/rmt.html) peripheral does just this.

GeorgeFlorian, 2019-12-10:
> The [RMT](https://docs.espressif.com/projects/esp-idf/en/v3.2/api-reference/peripherals/rmt.html) peripheral does just this.

You mean something [like this](http://www.buildlog.net/blog/2017/11/esp32-sending-short-pulses-with-the-rmt/) ?

atanisoft, 2019-12-10:
@GeorgeFlorian re examples in Arduion: https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/RMT

The RMT will only output to one pin either a high or low signal based on the timing you give it. If you need to split the signal on the output you can do that via NPN/PNP transistors to create an inverse signal or direct the output to two outputs based on the high/low state of the single output.

GeorgeFlorian, 2019-12-10:
> @GeorgeFlorian re examples in Arduion: https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/RMT
>
> The RMT will only output to one pin either a high or low signal based on [...]

Isn't that a bit too hard ?

Shouldn't my code work just fine ?

atanisoft, 2019-12-10:
The problem with your current code is that the timing is not guaranteed. It can and very likely will end up not sending a stable signal across the two pins due to the way that FreeRTOS tasks are scheduled. By using the RMT to generate the signal you are guaranteed that it will be consistent (within 1-2uS typically) and you won't have any impact by the FreeRTOS scheduler.

beegee-tokyo, 2019-12-10:
Your code is not working because in your loop after `digitalWrite(W0, HIGH);` you write immediately `digitalWrite(W0, LOW);` without `delayMicroseconds(pulseWidth);`

GeorgeFlorian, 2019-12-12:
> Your code is not working because in your loop after `digitalWrite(W0, HIGH);` you write immediately `digitalWrite(W0, LOW);` without `delayMicroseconds(pulseWidth);`
> 
> Try
> 
> ```c++
> for(int i = 0; i<26; i++) {
>       if(wiegandArray[i] == 0) {
>         digitalWrite(W0, LOW);
>         delayMicroseconds(pulseWidth);
>         digitalWrite(W0, HIGH);
>         delayMicroseconds(pulseWidth);
>       } else {
>         digitalWrite(W1, LOW);
>         delayMicroseconds(pulseWidth);
>         digitalWrite(W1, HIGH);
>         delayMicroseconds(pulseWidth);
>       }
> ```

Yes, that was a slip-up on my part. But even after fixing it it still doesn't work

GeorgeFlorian, 2019-12-12:
I have encountered something strange: the execution time of the [b]for[/b] loop should be 26*90*2 >= 4680us. But more than half of executions last way longer than that. The longest execution time was more than 11000us.

```
    start = micros();
    for(int i = 0; i<26; i++) {
      if(wiegandArray[i] == 0) {
        digitalWrite(W0, LOW);
        delayMicroseconds(pulse);
        digitalWrite(W0, HIGH);
      } else {
        digitalWrite(W1, LOW);
        delayMicroseconds(pulse);
        digitalWrite(W1, HIGH);
      }
      delayMicroseconds(pulse);
    }
    finish = micros();
```

GeorgeFlorian, 2019-12-12:
@atanisoft  Isn' there a way of suspending the FreeRTOS scheduler before the `for loop` and resume it after ?

I'm trying this:
```
  if(change) {
    Serial.println("Sending ID...");
    pulse = pulseWidth.toInt();

    vTaskSuspendAll();   
 
    for(int i = 0; i<26; i++) {
      if(wiegandArray[i] == 0) {
        digitalWrite(W0, LOW);
        delayMicroseconds(pulse);
        digitalWrite(W0, HIGH);
      } else {
        digitalWrite(W1, LOW);
        delayMicroseconds(pulse);
        digitalWrite(W1, HIGH);
      }
      delayMicroseconds(pulse);
    }

    xTaskResumeAll();

  }
```

But the card doesn't receive the signal each time.

atanisoft, 2019-12-12:
@GeorgeFlorian it is not possible to suspend the FreeRTOS scheduler.

GeorgeFlorian, 2019-12-12:
> @GeorgeFlorian it is not possible to suspend the FreeRTOS scheduler.

Ok. Then my only option is to learn to use RMT. Great...
The examples from here https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/RMT don't even include GPIOs.

And the example from here: http://www.buildlog.net/blog/2017/11/esp32-sending-short-pulses-with-the-rmt/ includes just one pin.

How do I make it for two pins ? How do I send my information `in one go` on two pins ?
Wiegand Protocol starts with both pins HIGH. When I send a 0, W0 pin is pulled LOW for usually 20-100us and W1 stays HIGH. When a 1 is being send W1 is pulled LOW for usually 20-100us and W0 stays HIGH.

atanisoft, 2019-12-12:
If you need two pins you may need to add some hardware to split the signal or use a different option.

beegee-tokyo, 2019-12-12:
You can try to stop task switching by using ```portENTER_CRITICAL``` and  ```portEXIT_CRITICAL```. But if your loop is too long you might cause a WDT reset.

I am doing this to get data from a DHT11 sensor which is a one wire protocol and you have to measure time between highs and lows to see if the sensor sends a 1 or a 0.

```cpp
// ESP32 is a multi core / multi processing chip
// It is necessary to disable task switches during the readings
portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED;
portENTER_CRITICAL(&mux);

for(int i = 0; i<26; i++) {
   if(wiegandArray[i] == 0) {
      digitalWrite(W0, LOW);
      delayMicroseconds(pulse);
      digitalWrite(W0, HIGH);
   } else {
      digitalWrite(W1, LOW);
      delayMicroseconds(pulse);
      digitalWrite(W1, HIGH);
   }
}
portEXIT_CRITICAL(&mux);
```

GeorgeFlorian, 2019-12-12:
> You can try to stop task switching by using `portENTER_CRITICAL` and `portEXIT_CRITICAL`. But if your loop is too long you might cause a WDT reset.
> 
> I am doing this to get data from a DHT11 sensor which is a one wire protocol and you have to measure time between highs and lows to see if the sensor sends a 1 or a 0.
> 
> ```c++
> // ESP32 is a multi core / multi processing chip
> // It is necessary to disable task switches during the readings
> portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED;
> portENTER_CRITICAL(&mux);
> 
> for(int i = 0; i<26; i++) {
>    if(wiegandArray[i] == 0) {
>       digitalWrite(W0, LOW);
>       delayMicroseconds(pulse);
>       digitalWrite(W0, HIGH);
>    } else {
>       digitalWrite(W1, LOW);
>       delayMicroseconds(pulse);
>       digitalWrite(W1, HIGH);
>    }
> }
> portEXIT_CRITICAL(&mux);
> ```

My `for loop` isn't running forever. Will the WDT be triggered in this case ?
```
void loop() {
  delay(100);
  if(change) {
    Serial.println("Sending ID...");
    pulse = pulseWidth.toInt();
    portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED;
    portENTER_CRITICAL(&mux);
    for(int i = 0; i<26; i++) {
      if(wiegandArray[i] == 0) {
        digitalWrite(W0, LOW);
        delayMicroseconds(pulse);
        digitalWrite(W0, HIGH);
      } else {
        digitalWrite(W1, LOW);
        delayMicroseconds(pulse);
        digitalWrite(W1, HIGH);
      }
      delayMicroseconds(pulse);
    }
    portEXIT_CRITICAL(&mux);
  }

  if(change) {
    change = false;
    for(int i = 0; i<26; i++) {
      Serial.print(wiegandArray[i]);
    }
    Serial.println();
  }
}
```

GeorgeFlorian, 2019-12-12:
> You can try to stop task switching by using `portENTER_CRITICAL` and `portEXIT_CRITICAL`. But if your loop is too long you might cause a WDT reset.
>
> I am doing this to get data from a DHT11 sensor [...]

@beegee-tokyo 
My `for loop` isn't running forever. Will the WDT be triggered in this case ?
```
void loop() {
  delay(100);
  if(change) {
    Serial.println("Sending ID...");
    pulse = pulseWidth.toInt();
    portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED;
    portENTER_CRITICAL(&mux);
    for(int i = 0; i<26; i++) {
      if(wiegandArray[i] == 0) {
        digitalWrite(W0, LOW);
        delayMicroseconds(pulse);
        digitalWrite(W0, HIGH);
      } else {
        digitalWrite(W1, LOW);
        delayMicroseconds(pulse);
        digitalWrite(W1, HIGH);
      }
      delayMicroseconds(pulse);
    }
    portEXIT_CRITICAL(&mux);
  }

  if(change) {
    change = false;
    for(int i = 0; i<26; i++) {
      Serial.print(wiegandArray[i]);
    }
    Serial.println();
  }
}
```

beegee-tokyo, 2019-12-12:
@GeorgeFlorian 
This kind of software timed protocol is always difficult in a multi-tasking environment. 

Last thing I can think of is to try to run it in a separate task. But that doesn't guarantee either that your timing is interrupted.

I never used the RMT library, but isn't it possible to use 2 channels at the same time?

GeorgeFlorian, 2019-12-12:
> @GeorgeFlorian
> This kind of software timed protocol is always difficult in a multi-tasking environment.
>
> Last thing I can think of is to try to run it in a separate task. But that doesn't guarantee [...]

From what I've read you can't use `if` or `for` inside tasks. I''ve tried doing this and it just crashes every time.

I have also never used RMT. The only examples I could find don't help me at all, because I hardly understand them.

everslick, 2019-12-12:
> From what I've read you can't use if or for inside tasks.

You definitely can!

 > I've tried doing this and it just crashes every time.

Must have been something else.

atanisoft, 2019-12-12:
> From what I've read you can't use `if` or `for` inside tasks. I''ve tried doing this and it just crashes every time.

If this was true you wouldn't be able to use it inside loop() either since it is running in a task...

> I never used the RMT library, but isn't it possible to use 2 channels at the same time?

You can definitely use two channels of the RMT simultaneously but there is no guarantee that they will be in sync with each other.

> The only examples I could find don't help me at all, because I hardly understand them.

I'd suggest starting with reading the documentation for the RMT which will help you understand what the examples are doing. But in short, the RMT will generate a high/low signal on a single pin per channel using the provided periods for the high and low segments.

If you look at [these](https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/RMT/RMTWriteNeoPixel/RMTWriteNeoPixel.ino#L65-L75) lines from the RMTWriteNeoPixel example, it is building up the data to be transmitted. The first half of the conditional block is for a "1" and the second for a "0", the difference between them is duration of the high/low segments of the signal being generated.

GeorgeFlorian, 2019-12-12:
I have tried using this task:
```
void taskOne(void * parameter ) { 
  if(change) {
    int pulse = pulseWidth.toInt();
    vTaskSuspendAll();
    for(int i = 0; i<26; i++) {
      if(wiegandArray[i] == 0) {
        digitalWrite(W0, LOW);
        delayMicroseconds(pulse);
        digitalWrite(W0, HIGH);
      } else {
        digitalWrite(W1, LOW);
        delayMicroseconds(pulse);
        digitalWrite(W1, HIGH);
      }
      delayMicroseconds(pulse);
    }
    xTaskResumeAll();
    if(change) {
      change = false;
      for(int i = 0; i<26; i++) {
        Serial.print(wiegandArray[i]);
      }
      Serial.println();
    }
    uxTaskGetStackHighWaterMark(NULL);
  }
}

void setup() {
xTaskCreate(
                    taskOne,          /* Task function. */
                    "TaskOne",        /* String with name of task. */
                    10000,            /* Stack size in bytes. */
                    NULL,             /* Parameter passed as input of the task */
                    1,                /* Priority of the task. */
                    NULL);            /* Task handle. */
}
}
void loop() {
delay(1000);
}
```

and this returns:
```
E (5644) FreeRTOS: FreeRTOS Task "TaskOne" should not return, Aborting now!
abort() was called at PC 0x40088b9b on core 0

Backtrace: 0x4008c46c:0x3ffd3400 0x4008c69d:0x3ffd3420 0x40088b9b:0x3ffd3440
Decoding stack results
0x40088b9b: xTaskResumeAll at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/tasks.c line 2226

```

atanisoft, 2019-12-12:
That is because the task *MUST* remain running using a for loop or similar.... But even that task is not going to give you what you are after since the timing is not guaranteed to be consistent.

GeorgeFlorian, 2019-12-12:
> That is because the task _MUST_ remain running using a for loop or similar.... But even that task is not going to give you what you are after since the timing is not guaranteed to be consistent.

So my only option is to send the signal using RMT, do the pnp/npn Voodoo and split it by the value that I want to send, 0 goes to pin W0 and 1 goes to pin W1 ?

atanisoft, 2019-12-12:
I'd suggest moving this to esp32.com forum and discuss the implementation there.

GeorgeFlorian, 2019-12-12:
> I'd suggest moving this to esp32.com forum and discuss the implementation there.

Done: 
https://www.esp32.com/viewtopic.php?f=19&t=13503&p=53201#p53201

closed 2019-12-12 by GeorgeFlorian

atanisoft, 2019-12-12:
@GeorgeFlorian something to look at: https://github.com/MakerLabsVan/makerlabs-acm/tree/master/firmware/components/wiegand_reader

They are using a hardware timer to generate the bit signal, that could work but discussing it on the forum will likely lead to a better solution.

atanisoft, 2019-12-12:
https://github.com/monkeyboard/Wiegand-Protocol-Library-for-Arduino

atanisoft, 2019-12-12:
https://github.com/FresnoIdeaworks/ESP32_Weigand/blob/master/RFID_Weigand_MQTT_FIW_v2.ino

a couple other options...

GeorgeFlorian, 2019-12-12:
Thank you.
These appear to be Wiegand Readers, but what I need is a Wiegand "Sender". I will have to take a better look.

>but discussing it on the forum will likely lead to a better solution.

I don't see that happening.