espressif/arduino-esp32 #3559: BLEScan crashing when running for long time

roy-zahor, 2019-12-10:
### Hardware:
Board:							ESP-WROOM-32D
Core Installation version:			?
IDE name:						Platform.io
Flash Frequency:					40Mhz
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
When running a BLE scan for a long time (timeout=0), or in an environment full of BLE transmitters, memory gets full and eventually leading to a system crash.

### Sketch:  
```cpp
    BLEDevice::init("");
    pBLEScan = BLEDevice::getScan(); //create new scan
    pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
    pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster

    BLEScanResults foundDevices = pBLEScan->start(0);
```
In BLEScan.c, the following line (106):
BLEAdvertisedDevice *advertisedDevice = new BLEAdvertisedDevice();
is not safe, because there is no check that new() operator returned a valid address, leading to system crash in case of low memory.

closed 2020-02-22 by stale[bot]

Ben79543, 2020-10-14:
I would like this issue to be reopened:

I have a device doing BLE Scans and it works fine in general.
Yesterday I tried it in a fair where there were at least 200 beacons around me.
My scan time is 5 sec.
The device crashes and I suppose it is because of memory issue too.

@roy-zahor did you work around this on your side?

roy-zahor, 2020-10-15:
@Ben79543 Solved for me by calling clearResults() once in a while. I'm not sure you have the same problem.