espressif/arduino-esp32 #3565: WIFI_POWER_20dBm = 80 doesn't work

AddisAir, 2019-12-12:
In libraries/WiFi/src/WiFiGeneric.h it allows to set the tx power up to 19.5db. 

But according to [espressif docs](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/network/esp_wifi.html#_CPPv425esp_wifi_set_max_tx_power6int8_t) it can go higher: 
`Maximum WiFi transmitting power, unit is 0.25dBm, range is [40, 82] corresponding to 10dBm - 20.5dBm`
so we should be able to set it higher, up to 20.5db or a value of 82. This is considerable boost: from 89mW to 112mW.

When i add a `WIFI_POWER_20dBm = 80` to WifiGeneric.h, and use `WiFi.setTxPower(WIFI_POWER_20dBm);` in my code, it doesn't change it though. the value i get returned from **`WiFi.getTxPower()` stays 78.**

Anyone knows why this is? 

PS: i know about legal limits TX power limits, but there are cases where this doesn't apply.

chegewara, 2019-12-12:
https://github.com/espressif/esp-idf/issues/4223

torntrousers, 2019-12-13:
Could it just be a bug in getTxPower where its returning WIFI_POWER_19_5dBm instead of power?
https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiGeneric.cpp#L632-L634

AddisAir, 2019-12-13:
good thinking! so for testing i changed the code to 
```
wifi_power_t WiFiGenericClass::getTxPower(){
    int8_t power;
    esp_wifi_get_max_tx_power(&power);
    return (wifi_power_t)power;
}
```
it gives 78 still... bizarre

closed 2020-02-26 by stale[bot]

bodhibrata, 2021-05-01:
Use the following code.

wifi_power_t tx_power;
tx_power=WiFi.getTxPower();
Serial.println(tx_power);