espressif/arduino-esp32 #3566: BLE client has Problem disconnecting 1 of 2 connected servers

davejel, 2019-12-12:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.1
IDE name:						Arduino IDE
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
disconnecting 1 of 2 connected ITags causes both ITags to be disconnected

I have written a client sketch for connecting 2 BLE ITags to one client ESP32. 
connecting to the ITags works well and i can pass information to the client.

My problem comes when i disconnect one of the ITags by switching it off, Instead of just disconnecting the tag i have switched off, onDisconnect is called twice and both tags are disconnected, the debug messages shows this happening. 
As expected in this scenario the second ITag starts beeping because it has been disconnected.
debug messages show that both times onDisconnect is called the peer address is the same, my 2 ITag addresses are ff:ff:c5:14:e3:c7 and ff:ff:c5:14:e3:c4

can anyone please explain to me why this is happening and / or how i can resolve the problem
my only guess is that as both tags have the same service uuid this is causing some confusion somewhere
thanks in advance
Dave

relevant part of sketch, I can show whole sketch if necessary
```
class MyClientCallback : public BLEClientCallbacks {
  void onConnect(BLEClient* pclient) {
    numConnected ++;
    Serial.print(numConnected);
    Serial.println(" ITags Connected\n");
    if(numConnected == 2) {
      BLEDevice::getScan()->stop();
    }
  }

  void onDisconnect(BLEClient* pclient) {
    Serial.print(pClient->getPeerAddress().toString().c_str());
    Serial.println("  Disconnected");
    numConnected --;
    Serial.print(numConnected);
    Serial.println(" ITags Connectedxxz");
  }
};
````

### Debug Messages: while disconnecting
```
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 41
[D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 41
[V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 41
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
ff:ff:c5:14:e3:c7  Disconnected
1 ITags Connectedxxz
[I][BLEDevice.cpp:608] removePeerDevice(): remove: 0, GATT role client
[V][FreeRTOS.cpp:120] give(): Semaphore giving: name: RssiCmplEvt (0x3ffde974), owner: <N/A>
[V][FreeRTOS.cpp:120] give(): Semaphore giving: name: SearchCmplEvt (0x3ffde914), owner: <N/A>
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 41
[D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 41
[V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 41
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
ff:ff:c5:14:e3:c7  Disconnected
0 ITags Connectedxxz
[I][BLEDevice.cpp:608] removePeerDevice(): remove: 1, GATT role client
[V][FreeRTOS.cpp:120] give(): Semaphore giving: name: RssiCmplEvt (0x3ffe0e14), owner: <N/A>
[V][FreeRTOS.cpp:120] give(): Semaphore giving: name: SearchCmplEvt (0x3ffe0db4), owner: <N/A>
[... 427 lines cut ...]
[V][BLEUtils.cpp:1818] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1050] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1265] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEClient.cpp:464] handleGAPEvent(): BLEClient ... handling GAP event!
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring ff:ff:c5:15:cd:c4, already seen it.
[V][BLEUtils.cpp:1818] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1050] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1265] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEClient.cpp:464] handleGAPEvent(): BLEClient ... handling GAP event!
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring ff:ff:c5:15:cd:c4, already seen it.
[V][BLEUtils.cpp:1818] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1050] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1265] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEClient.cpp:464] handleGAPEvent(): BLEClient ... handling GAP event!
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring ff:ff:c5:15:cd:c4, already seen it.
[V][BLEUtils.cpp:1818] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1050] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1265] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEClient.cpp:464] handleGAPEvent(): BLEClient ... handling GAP event!
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring ff:ff:c5:15:cd:c4, already seen it.
[V][BLEUtils.cpp:1818] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1050] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1265] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEClient.cpp:464] handleGAPEvent(): BLEClient ... handling GAP event!
```

chegewara, 2019-12-12:
> Core Installation version: 1.0.1

Why not switching to 1.0.4?

davejel, 2019-12-12:
sorry i think i misunderstood what core installation version was, in 
C:\Users\Dave\AppData\Local\Arduino15\packages\esp32\hardware\esp32
the version is 1.0.4, is that the correct one ?

chegewara, 2019-12-13:
Could you try this to get more info?
https://github.com/nkolban/esp32-snippets/issues/735#issuecomment-444007131

davejel, 2019-12-13:
@chegewara thanks for your help AGAIN!!!
i added the lines as you said:
``static void my_gap_event_handler(esp_gap_ble_cb_event_t  event, esp_ble_gap_cb_param_t* param) {
  ESP_LOGW(LOG_TAG, "custom gap event handler, event: %d", (uint8_t)event);
}

static void my_gattc_event_handler(esp_gattc_cb_event_t event, esp_gatt_if_t gattc_if, esp_ble_gattc_cb_param_t* param) {
  ESP_LOGW(LOG_TAG, "custom gattc event handler, event: %d", (uint8_t)event);
}

static void my_gatts_event_handler(esp_gatts_cb_event_t event, esp_gatt_if_t gattc_if, esp_ble_gatts_cb_param_t* param) {
  ESP_LOGW(LOG_TAG, "custom gatts event handler, event: %d", (uint8_t)event);
}

void setup() {
  Serial.begin(115200);
  Serial.println("Multi Connect\n");
  doScan = true;
    
    BLEDevice::setCustomGapHandler(my_gap_event_handler);
    BLEDevice::setCustomGattsHandler(my_gatts_event_handler);
    BLEDevice::setCustomGattcHandler(my_gattc_event_handler);  
  
  BLEDevice::init("");
  pBLEScan = BLEDevice::getScan();
//  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setInterval(600);
  pBLEScan->setWindow(500);
  pBLEScan->setActiveScan(true);
}
``
this caused a crash with corrupt heap when i connected the first ITag
``
[D][FreeRTOS.cpp:165] take(): Semaphore taking: name: ReadCharEvt (0x3ffe0938), owner: <N/A> for readValue
[D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: ReadCharEvt (0x3ffe0938), owner: readValue
[V][FreeRTOS.cpp:70] wait(): >> wait: Semaphore waiting: name: ReadCharEvt (0x3ffe0938), owner: readValue for readValue
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 3
[D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 3
[V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 3
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
CORRUPT HEAP: Bad head at 0x3ffe10b5. Expected 0xabba1234 got 0x00000000
assertion "head != NULL" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
abort() was called at PC 0x400e9583 on core 0

Backtrace: 0x40091448:0x3ffcff20 0x40091679:0x3ffcff40 0x400e9583:0x3ffcff60 0x400910d5:0x3ffcff90 0x40084b26:0x3ffcffb0 0x400850e1:0x3ffcffd0 0x4000bec7:0x3ffcfff0 0x400d5992:0x3ffd0010 0x400d625d:0x3ffd0050 0x400d4771:0x3ffd0070 0x400d54f1:0x3ffd0120 0x40106c2d:0x3ffd0190 0x400ffda2:0x3ffd01e0 0x4008e089:0x3ffd0210

Rebooting...
ets Jun  8 2016 00:22:57
``
i decoded the backtrace and got:

``Decoding stack results
0x40091448: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x40091679: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x400e9583: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 63
0x400910d5: multi_heap_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c line 214
0x40084b26: heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 268
0x400850e1: _free_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c line 42
0x400d5992: BLERemoteCharacteristic::gattClientEventHandler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at D:\Arduino\libraries\ESP32_BLE_Arduino\src\BLERemoteCharacteristic.cpp line 192
0x400d625d: BLERemoteService::gattClientEventHandler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at D:\Arduino\libraries\ESP32_BLE_Arduino\src\BLERemoteService.cpp line 120
0x400d4771: BLEClient::gattClientEventHandler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at D:\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp line 314
0x400d54f1: BLEDevice::gattClientEventHandler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at D:\Arduino\libraries\ESP32_BLE_Arduino\src\BLEDevice.cpp line 169
0x40106c2d: btc_gattc_cb_handler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/profile/std/gatt/btc_gattc.c line 31
0x400ffda2: btc_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/core/btc_task.c line 110
0x4008e089: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
``
which i am afraid means nothing to me but hopefully you will understand it 

after a few reboots the first ITag did connect ok without crashing, i tried connecting the second one but it did not connect, this gave a repeating output on the serial monitor as follows
``[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring ff:ff:c5:15:cd:c4, already seen it.
[W][ClientMultiConnect.ino:65] my_gap_event_handler(): custom gap event handler, event: 3
[V][BLEUtils.cpp:1818] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1050] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1265] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring ff:ff:c5:15:cd:c4, already seen it.
``

i have now reached a point where i do not understand what i am doing so i need to learn
my feeling is that the corrupt heap is probably the most relevant thing to sort out but this only appeared after i added the lines from #735 so it might be those lines causing the crash
please teach me guru

chegewara, 2019-12-13:
Im not sure, but this may be related
https://github.com/espressif/arduino-esp32/issues/3402

wakwak-koba, 2019-12-17:
Would you like to try my Fork?
https://github.com/wakwak-koba/arduino-esp32/tree/master/libraries/BLE

The sample I published uses the newly prepared method, but keeps forward compatibility as much as possible.
I think that even if you don't use these new methods, they will work.

I know it should ideally be PR, but I modified too many.
I have lost confidence to explain the effect.

I need a tester because we are still worried about whether it is worth the PR.

davejel, 2019-12-17:
@wakwak-koba thank you yes i will try, i had given up but i can resurrect the project again and test in the next few days and let you know how i get on

davejel, 2019-12-17:
i have one other question if anyone can answer please
in the BLE example sketches, in the loop() function there is usually a delay(1000) or (2000)
what is the purpose of the delay, is it to allow ble stack housekeeping? i have found a few times that problems i cannot track go away with this delay in place but i am not certain if i am treating a symptom or a cause
thanks

wakwak-koba, 2019-12-17:
I don't care much, but I think there is another reason why behavior changes depending on delay ().

wakwak-koba, 2019-12-18:
I asked an acquaintance whether delay () was necessary.
In his answer, if the loop () is left empty, the current consumption increases, it's better to include 'delay (1);'

davejel, 2019-12-18:
@wakwak-koba thank you for that but it still confuses me a little as to why the functionality should change with / without it unless something takes time to recover from the extra power useage

davejel, 2019-12-18:
@wakwak-koba 
thats definitely better, i have just been testing it with my sketch above. Now onDisconnect only gets called once but there is still the problem that if i disconnect both tags the line:
    Serial.print(pClient->getPeerAddress().toString().c_str());
returns the same address on each disconnect so it looks as though an index in the lib is not getting changed...... somewhere

wakwak-koba, 2019-12-19:
It is possible that removePeerDevice() has not been called when disconnecting/closing.
I will check a little more.

davejel, 2019-12-19:
ok, no problem i'm in no rush if you let me know i'll test again after its done

closed 2020-03-02 by stale[bot]