espressif/arduino-esp32 #3572: Using HSPI pins on ESP32 for interfacing SD card module

PrateekGoyal18, 2019-12-14:
Hello everyone, 
I am trying to interface SD card module on the HSPI pis of ESP32 since I need to use to VSPI pins as GPIO for some other task.  My codes is,

`#include "FS.h"
#include "SD.h"
#include "SPI.h"
SPIClass spiSD(HSPI);
#define SD_CS 15

void setup()
{
  Serial.begin(9600);
  Serial.println();
  SPI.end(); 
  spiSD.begin(14, 12, 13, SD_CS); //CLK,MISO,MOIS,SS
  if (!SD.begin(SD_CS, spiSD))
  {
    Serial.println("Card Mount Failed");
    return;
  }
  else
  {
    Serial.println("Card Mount Successful");
  }
}

void loop()
{
}`

First of all if I upload the code with all the pins of sd card module connected to esp32, it does not upload due to I think GPIO12 property "boot fail if pulled high". So when I take out the jumper form GPIO12 and upload the code, some errors are thrown on the Serial monitor.
Can anyone help me out here?

lbernstone, 2019-12-14:
You are too far away for my mind reading to work.  Please turn on verbose logging, and <b>post</b> the log. Please read the [issue template](https://github.com/espressif/arduino-esp32/blob/master/docs/ISSUE_TEMPLATE.md), and provide the information requested there, along with your device connection information- particularly what sort of pullup resistors you are using.
For GPIO12, see https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card#note-about-gpio12

closed 2020-01-08 by PrateekGoyal18

davemaster, 2020-09-24:
Finally, I would be able to use HSPI for smartmatrix-teensylc ESP32 compatible library, with this pins arrangement in MatrixHardware_ESP32_V0.h smartmatrix file, 

    **#define R1_PIN  GPIO_NUM_2
    #define G1_PIN  GPIO_NUM_23//15
    #define B1_PIN  GPIO_NUM_4
    #define R2_PIN  GPIO_NUM_16
    #define G2_PIN  GPIO_NUM_27
    #define B2_PIN  GPIO_NUM_17

    #define A_PIN   GPIO_NUM_5
    #define B_PIN   GPIO_NUM_18
    #define C_PIN   GPIO_NUM_19
    #define D_PIN   GPIO_NUM_21
    #define E_PIN   GPIO_NUM_33//12
    #define LAT_PIN GPIO_NUM_26
    #define OE_PIN  GPIO_NUM_25**

And this in the smartmatrix_player.ino

SPIClass spiSD(HSPI);

// Chip select for SD card
#if defined(ESP32)
    #define SD_CS 15     // 5 **
#elif defined (ARDUINO)
    #define SD_CS BUILTIN_SDCARD
    //#define SD_CS 15
#endif
.......................
.......................
.......................

Serial.print("Initializing SD card...");

spiSD.begin(14, 32, 13, 15); //SCK,MISO,MOSI,SS //HSPI1
     
if (!SD.begin( SD_CS, spiSD )) 
{
   Serial.println("Card Mount Failed");
   return;
}

Then, it run perfectly, loading gifs from /gifs directory

Best Regards

iharshadec, 2023-11-01:
> MatrixHardware_ESP32_V0.h

Hi I am facing something similar issue where with latest ESP32 Board manager my ICM-20948 is not working on HSPI as it was working earlier nicely with 1.0.6 version.
where do I find the file you mentioned? MatrixHardware_ESP32_V0.h?