espressif/arduino-esp32 #3576: SPI_HAS_TRANSACTION has issues with being defined

Avamander, 2019-12-16:
I'm one of the maintainers of github.com/nRF24/RF24 and for some reason our previously working code for detecting SPI transaction support does not work for ESP32:
```c++
#if defined (SPI_HAS_TRANSACTION) && !defined (SPI_UART) && !defined (SOFTSPI)
  #define RF24_SPI_TRANSACTIONS
#endif // defined (SPI_HAS_TRANSACTION) && !defined (SPI_UART) && !defined (SOFTSPI)
```

For some reason ESP32 users are [saying that transactions don't get enabled](https://github.com/nRF24/RF24/issues/508) and they need frankly weird [patches](https://github.com/nRF24/RF24/pull/549/files) to fix those issues. There's a bit of debugging done in the linked issue, it's... weird.

It would be very appreciated if someone on the core team could take a look at what's missing in ESP32 core that it doesn't act like an Arduino board for those people. 

### Hardware:
Core Installation version:			1.0.3
IDE name:						Arduino IDE and Platform.io
Computer OS:                                         Unspecified

Avamander, 2020-02-14:
Ping

Avamander, 2020-04-14:
Ping

Avamander, 2020-06-13:
Ping

atanisoft, 2020-06-13:
@Avamander if you know of a fix for this, feel free to submit a PR with it for review.

Avamander, 2020-06-13:
@atanisoft https://github.com/espressif/arduino-esp32/pull/3730

Though by now the suspicion is that platform.io "eats" the flag somewhere, but technically the PR should fix it.

atanisoft, 2020-06-13:
@Avamander there is a build-platformio.py file under tools that will need to add that default flag. That should fix it for PIO as well.

Avamander, 2020-06-13:
@atanisoft 
> there is a build-platformio.py file under tools that will need to add that default flag. That should fix it for PIO as well.

I made that change too.

closed 2020-08-27 by stale[bot]