espressif/arduino-esp32 #3580: WebServer and ASyncWebServer do not work with Ethernet

ivoh95, 2019-12-18:
### Hardware:
Board:							ESP32 Dev Module + w5500 chip
Core Installation version:			1.0.4
IDE name:						Arduino IDE and Platform.io
Flash Frequency:					80Mhz
PSRAM enabled:                                    no 
Upload Speed:					921600
Computer OS:                                        Windows 10

### Description:
I have been trying to get the arduino webserver library or asyncwebserver library working over Ethernet with the built in Ethernet library for a few days now and failing miserably. I am using a w5500 Ethernet chip and it works fine in other projects using udp or tcp messages with no issue, using the exact same hardware setup. The issue arises as soon as i use either webserver library and call the begin function the esp32 crashes and reboots. Ive tried platformio and the arduino ide with the latest core, and 1.0.3 and results are the same.  

Ive used the webserver examples for both libraries and both work perfectly fine over wifi, but as soon as i switch it to using Ethernet, crashes happen when begin is called. 

I believe the issue stems from the esp32 core and some nonstandard implementations of the server begin function? Despite my attempts i have not been able to figure out how to fix this issue.  

Any suggestions on what to try next? Any more information or data i could provide?

The sketch will crash in the same way even without the w5500 hardware attached so that to test and confirm all that is needed is a esp32 dev board. Ive attached one of the webserver examples modified for Ethernet instead of wifi. 

Thanks for any assistance!

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Ethernet.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>

IPAddress ip(192, 168, 10, 13); //failsafe ip 
byte mac[6] = {0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED}; //failsafe mac

AsyncWebServer server(80);

const char* PARAM_MESSAGE = "message";

void notFound(AsyncWebServerRequest *request) {
    request->send(404, "text/plain", "Not found");
}
void setup(void) {
  Serial.begin(115200);
  setupEthernet();
  Serial.print("IP address: ");
  Serial.println(Ethernet.localIP());
  delay(1000);
  /*return index page which is stored in serverIndex */
     server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
        request->send(200, "text/plain", "Hello, world");
    });

    // Send a GET request to <IP>/get?message=<message>
    server.on("/get", HTTP_GET, [] (AsyncWebServerRequest *request) {
        String message;
        if (request->hasParam(PARAM_MESSAGE)) {
            message = request->getParam(PARAM_MESSAGE)->value();
        } else {
            message = "No message sent";
        }
        request->send(200, "text/plain", "Hello, GET: " + message);
    });

    // Send a POST request to <IP>/post with a form field message set to <message>
    server.on("/post", HTTP_POST, [](AsyncWebServerRequest *request){
        String message;
        if (request->hasParam(PARAM_MESSAGE, true)) {
            message = request->getParam(PARAM_MESSAGE, true)->value();
        } else {
            message = "No message sent";
        }
        request->send(200, "text/plain", "Hello, POST: " + message);
    });

    server.onNotFound(notFound);

    server.begin();
}

void loop(void) {

}

void setupEthernet(){
  pinMode(4, OUTPUT);
  digitalWrite(4, LOW);
  delay(25);
  digitalWrite(4, HIGH); //enable the ethernet chip after resetting
  Ethernet.init(5);  //set our CS pin for ethernet

  Ethernet.begin(mac, ip); 
  
 if (Ethernet.hardwareStatus() == EthernetNoHardware) {
    Serial.println(F("Ethernet hardware not found"));
  }
  }
```

### Debug Messages:
```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
IP address: 192.168.10.13
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1442 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40088071 on core 1

Backtrace: 0x4008b744:0x3ffb1da0 0x4008b975:0x3ffb1dc0 0x40088071:0x3ffb1de0 0x400f9556:0x3ffb1e20 0x400f9622:0x3ffb1e40 0x400ee000:0x3ffb1e60 0x400ee069:0x3ffb1e80 0x400ef5c0:0x3ffb1ea0 0x400ef6ac:0x3ffb1ed0 0x400d294d:0x3ffb1ef0 0x400d69ba:0x3ffb1f40 0x400d19e2:0x3ffb1f60 0x400d92cb:0x3ffb1fb0 0x40088385:0x3ffb1fd0

Rebooting...

Decoding stack results
0x4008b744: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x4008b975: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x40088071: xQueueGenericReceive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 1442
0x400f9556: sys_mutex_lock at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c line 78
0x400f9622: sys_arch_protect at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c line 469
0x400ee000: do_memp_malloc_pool at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/memp.c line 302
0x400ee069: memp_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/memp.c line 398
0x400ef5c0: tcp_alloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c line 1661
0x400ef6ac: tcp_new_ip_type at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c line 1782
0x400d294d: AsyncServer::begin() at C:\Users\Ivo\Documents\Arduino\libraries\AsyncTCP\src\AsyncTCP.cpp line 1274
0x400d69ba: AsyncWebServer::begin() at C:\Users\Ivo\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebServer.cpp line 84
0x400d19e2: setup() at C:\Users\Ivo\Documents\Arduino\OTAWebUpdaterESPTEST/OTAWebUpdaterESPTEST.ino line 52
0x400d92cb: loopTask(void*) at C:\Users\Ivo\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\main.cpp line 14
0x40088385: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

```

ivoh95, 2019-12-20:
Here is a backtrace using the webserver library

### Debug Messages:
```
0x4008bbf8: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x4008be29: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x400880a1: xQueueGenericReceive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 1442
0x40107a82: sys_mutex_lock at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c line 78
0x40107c82: sys_arch_protect at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c line 469
0x400f9bb0: do_memp_malloc_pool at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/memp.c line 302
0x400f9c19: memp_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/memp.c line 398
0x400f6042: netconn_alloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/api_msg.c line 742
0x400f4c94: netconn_new_with_proto_and_callback at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/api_lib.c line 133
0x400f9154: lwip_socket at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/sockets.c line 1587
0x400d625f: WiFiServer::begin(unsigned short) at C:\Users\Ivo\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/lwip/lwip/sockets.h line 593
0x400d3f60: WebServer::begin() at C:\Users\Ivo\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer\src\WebServer.cpp line 97
0x400d1db2: setup() at C:\Users\Ivo\AppData\Local\Temp\arduino_modified_sketch_533200/HelloServer.ino line 61
0x400d906f: loopTask(void*) at C:\Users\Ivo\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\main.cpp line 14
0x400883b5: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

ivoh95, 2019-12-20:
https://github.com/arduino-libraries/Ethernet/issues/88#issuecomment-455498941

Additionally this may be relevant to the issue? Not sure what the right direction to go towards for fixing this is?

orvisevans, 2019-12-31:
It seems a similar problem was seen earlier in ESP32's client.h and was fixed by bringing the ESP32 core in line with the Arduino core: https://github.com/espressif/arduino-esp32/issues/2755

ivoh95, 2020-01-13:
So i believe then the issue comes from the Sever.h file? 

Esp32 has  virtual void begin(uint16_t port=0) =0;

and Arduino has   virtual void begin() =0;

That can be changed, but will likely break other things in the esp32 core, which should also then be brought in line with the arduino core.

Im not super familiar but what is the difference between the two in practice when calling the being function?

closed 2020-03-27 by stale[bot]

CircuitSetup, 2020-07-06:
I'm also having this issue. Was a resolution ever figured out?

aaronse, 2020-09-01:
@ivoh95 and others, thanks for starting this issue thread.

**problem:**
server.begin() executes on ESP8266, but fails on ESP32 with same stack @ivoh95 posted on Dec 18, 2019.  Mem allocation for pcb struct fails.

**cause**
I suspect...  that tcp stack not initialized before WebServer begin was called.  Speculating based on behavior observed after spraying code with traces, reading code referenced by the stack trace, and trying various changes.  

Ideally, would confirm using an inline debugger.  This issue caused me to discover existence of JTAG inline debugging, am currently awaiting ESP-PROG debug board from digikey.  Am new to ESP32, starting tinkering last week.

**workaround/fix**
Ensure WiFi.mode is called and configured before calling .begin on WebServer instance.

Backtrace: 0x4008c434:0x3ffb1d70 0x4008c665:0x3ffb1d90 0x40088869:0x3ffb1db0 0x40123442:0x3ffb1df0 0x4012371e:0x3ffb1e10 0x401124d4:0x3ffb1e30 0x4011253d:0x3ffb1e50 0x40113b08:0x3ffb1e70 0x40113bf4:0x3ffb1ea0 0x401688c5:0x3ffb1ec0 0x40134b39:0x3ffb1f10 0x400d1d6a:0x3ffb1f30 0x400d1075:0x3ffb1f80 0x400d443f:0x3ffb1fb0 0x40088b7d:0x3ffb1fd0

	Decoded to…

	0x4008c434: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
	0x4008c665: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
	0x40088869: xQueueGenericReceive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 1442
	0x40123442: sys_mutex_lock at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c line 78
	0x4012371e: sys_arch_protect at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c line 469
	0x401124d4: do_memp_malloc_pool at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/memp.c line 302
	0x4011253d: memp_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/memp.c line 398
	0x40113b08: tcp_alloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c line 1661
	0x40113bf4: tcp_new_ip_type at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c line 1782
	0x401688c5: AsyncServer::begin() at .pio\libdeps\heltec_wifi_kit_32\AsyncTCP\src\AsyncTCP.cpp line 1262
	0x40134b39: AsyncWebServer::begin() at .pio\libdeps\heltec_wifi_kit_32\ESP Async WebServer\src\WebServer.cpp line 87
	0x400d1d6a: webServer::begin() at lib\esp8266-iot-framework\src\webServer.cpp line 47
	0x400d1075: setup() at src\main.cpp line 85
	0x400d443f: ...

lbernstone, 2020-09-01:
It probably needs the tcpip adapter loaded.  WiFi.mode will set that, but you should be able to get there without WiFi if you:
```
#include <tcpip_adapter.h>
tcpip_adapter_init();
```
Note that this will be deprecated in IDF4, and the new function will be esp_netif_init() in esp_netif.h.  Perhaps it makes sense to add that somewhere in the ETH initialization.

JayKay135, 2020-11-26:
@ivoh95 

Did you manage to get it working correctly and could share your experience with us?
And could you possibly give us the information how exactly you connected the w5500 module with the esp32 dev board.

Many thanks in advance ;D

ArminPP, 2020-12-15:
Hello,
I got the same issue.

Unfortunately this does not work for me:
> #include <tcpip_adapter.h>
> tcpip_adapter_init();

It would be really nice, if there is a working solution out there :-)

ArminPP, 2020-12-25:
> **workaround/fix**
> Ensure WiFi.mode is called and configured before calling .begin on WebServer instance.

Hello aaronse,
I tried to put WiFi.mode (WIFI_STA) before server.begin ().
Unfortunately, I still get the "ERR_CONNECTION_REFUSED" error in my web browser.
Could you please give me a little example of how you solved it?
Many Thanks,
Armin

alainkovacs, 2020-12-29:
Hi All,

@ivoh95 did you ever make any progress on this? 

For me the below does the trick, meaning that the ESP stops restarting, I manage to get an IP but the webserver just isn't there.. 

> It probably needs the tcpip adapter loaded. WiFi.mode will set that, but you should be able to get there without WiFi if you:
> 
> ```
> #include <tcpip_adapter.h>
> tcpip_adapter_init();
> ```
> 

Wishing everybody a Happy New Year! 
-Alain

ranelolesk, 2020-12-30:
Hello!

@ivoh95 Would you mind sharing your experience on this topic?

Take care and wish you a good ending for 2020!

EmitMark, 2021-02-08:
Just as @alainkovacs 
when I include the tcpip_adapter and call the init() it stops crashing and an IP address is assigned (I can find it perfectly fine when I scan my local network. Other services making use of the Ethernet module work perfectly fine as well but the Async Webserver just is no where to be found.

dj-fiorex, 2021-02-09:
Hi all, same issue for me, i can build the project, the ESP32 doesn't reboot, i got an ip address but it's impossible to comunicate with the server. Seems like LWIP doesn't know anything of the ethernet library and inboud packets.
Anyone have any idea?

kamil2234, 2021-03-28:
I have exactly the same problem

EmitMark, 2021-03-29:
I currently got a webserver functioning with the Ethernet module using the AWOT library. Maybe it might help you.
https://awot.net/

argo9, 2021-04-13:
same problem here: we would like to serve a web page on wifi _and_ ethernet, same time, or at least setichable at run time with w5500

ramsesgarciad, 2021-05-20:
Hi, i have a same promblem i trying to sever a one api from async web server with ethernet, but want reuse my code wrote with same lib.

thanks

geraldcells18, 2024-07-21:
any solutions?

me-no-dev, 2024-07-21:
With Arduino 3.x you can use Ethernet and PPP the same as WiFi (Our Ethernet library, not external ones)

JAndrassy, 2024-07-21:
>  not external ones

with the exception: my EthernetESP32 library

me-no-dev, 2024-07-21:
> with the exception: my EthernetESP32 library

It is still based on our Ethernet library. I meant ones that talk to the ETH chips directly through SPI

geraldcells18, 2024-07-21:
I managed to handle using awot, the framework similar to express but for esp boards.

liamcharmer, 2024-11-03:
> With Arduino 3.x you can use Ethernet and PPP the same as WiFi (Our Ethernet library, not external ones)

@geraldcells18  @me-no-dev 

Is this possible to do this and have this Library ESPAsyncWebServer work as a Web Socket Async server?