espressif/arduino-esp32 #3592: SPIFFS vs Preferences vs NVS storage for periodic data write / delete cycles

DeltaVetal26, 2019-12-22:
Hi,

I need to write / delete data up to 300 bytes in size about 5-10 times a day.

Data structure:

`uidTag: 123456789123
 startTime: 2019-12-22 12:00:00
 stopTime: 2019-12-22 12:10:00
 id: 120
 name: Alex`

Initially, I planned to use SPIFFS storage, but its service life is about 10000-20000 write / delete cycles. I am not sure if it is intended for constant data changes.

NVS storage also works via flash (if I understand correctly), which introduces about the same service life limits.

What about Preferences? Is it the best option for a frequent write / delete cycle?

If yes, then the questions are:
1) What are the Preferences based on? NVS and SPIFFS on flash.
2) Is it possible to add data to Preferences without deleting existing ones?
In the library, I saw only putData and getData.

The option using an SD card does not suit me at the moment.

Regards,
Alex

lbernstone, 2019-12-22:
Preferences is based on NVS.  Please post general questions on esp32.com.  This forum is for issues with this codebase.

DeltaVetal26, 2019-12-22:
@lbernstone ,

Ok, thank you for the answer!

closed 2019-12-22 by DeltaVetal26