espressif/arduino-esp32 #3593: Serial Monitor is not showing IP Address of ESP32-CAM

Argentum238, 2019-12-22:
After uploading sketch 'CameraWebServer' to the ESP32-CAM using Arudino IDE, I try using the serial monitor to show the IP address of the ESP32, however It just continues to load infinitely with no error code. I've attached an image. The dots just keep on printing and printing. 
[image]

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Wrover Module
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                    yes
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

closed 2020-03-05 by stale[bot]

YXGuan, 2020-10-13:
Hey buddy@Argentum238,  did you solve the issue? I am facing exactly the same issue with my ESP32

Argentum238, 2020-10-13:
Hi,

Sorry I was not able to solve this problem. But good luck!

YXGuan, 2020-10-14:
I figured it out. Let me know if you or someone are still working on this

csmithy1999, 2020-10-14:
Hi, just got the problem, mind telling me how you fixed it

rmk-homeassistant, 2020-10-24:
@YXGuan how did you solve this issue as i have the same problem.

slaveofcode, 2020-10-30:
I'm facing the same problem, please anybody help

slaveofcode, 2020-10-30:
Hi, got the problem because the ESP32 cannot connect to the wifi with the provided SSID & Password on the script code. After change / fix the value I've got it working so well

YXGuan, 2020-10-31:
@slaveofcode is correct, definitely double-check to make sure that SSID & password matches. I got the error because of wrong wiring. GPIO 0 and  GND should be connected by a jumper wire when uploading the code. When you finish uploading, remove the jumper wire, open serial monitor and push flash button, voila here you go.
Here is a more extensive explanation: [https://randomnerdtutorials.com/esp32-cam-troubleshooting-guide/#:~:text=Most%20common%20errors%3A,error%20%E2%80%93%20Wrong%20partition%20scheme%20selected]

ateeq666, 2021-03-30:
> Hi, got the problem because the ESP32 cannot connect to the wifi with the provided SSID & Password on the script code. After change / fix the value I've got it working so well

can u like emphasize on what you mean by the wifi not being able to connect with the ssid and password

T0b1e, 2021-06-26:
I got same issue how can I fixing it I can't get IP after  press reset button and remove jumper GPIO

y2knbb, 2021-07-09:
> Hi, got the problem because the ESP32 cannot connect to the wifi with the provided SSID & Password on the script code. After change / fix the value I've got it working so well

I also had same issue. I have put correct SSID and password in the code and try it. It worked...

y2knbb, 2021-07-10:
> I got same issue how can I fixing it I can't get IP after press reset button and remove jumper GPIO

Double check your SSID and password that you mentioned in the code.

himakarbavikaty, 2021-07-29:
Is it our router's SSID and Password need to update in ESP32 webserver?

anvinder, 2021-08-15:
I've tried to change the baud rate in the webserver file to 74800/ 115200, also changed the CPU frequency, flash frequency, PSRAM config, uploading speed and the serial monitor baud rate. None of this has resulted in any resolution. The SSID and password are correct. Can't think of  a reason why the serial monitor starts printing .........................................................
after a few lines and never shows the IP. 
i checked on the router and there is no IP assigned to ESP. 
Any thoughts ? The ESP itself is good coz i've run multiple other programs and it is working well.

sansillusion, 2021-08-16:
Try a wifi scanner sample code to see if you can see your router. If so make sure the resulting ssid is the same you input in you camera code. Try not to use special caracters in your wifi passkey. Make sure there is no spaces in both ssid and passkey.

ammiredd, 2022-04-15:
> After uploading sketch 'CameraWebServer' to the ESP32-CAM using Arudino IDE, I try using the serial monitor to show the IP address of the ESP32, however It just continues to load infinitely with no error code. I've attached an image. The dots just keep on printing and printing. [image]
> 
> ----------------------------- Remove above -----------------------------
> 
> ### Hardware:
> Board: ?ESP32 Wrover Module Core Installation version: ?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3? IDE name: Arduino IDE Flash Frequency: 40Mhz PSRAM enabled: yes Upload Speed: 115200 Computer OS: Windows 10
> 
> ### Description:
> Describe your problem here
> 
> ### Sketch: (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
> ```c++
> //Change the code below by your sketch
> #include "esp_camera.h"
> #include <WiFi.h>
> 
> //
> // WARNING!!! Make sure that you have either selected ESP32 Wrover Module,
> //            or another board which has PSRAM enabled
> //
> 
> // Select camera model
> //#define CAMERA_MODEL_WROVER_KIT
> //#define CAMERA_MODEL_ESP_EYE
> //#define CAMERA_MODEL_M5STACK_PSRAM
> //#define CAMERA_MODEL_M5STACK_WIDE
> #define CAMERA_MODEL_AI_THINKER
> 
> #include "camera_pins.h"
> 
> const char* ssid = "Thunder";
> const char* password = "Thunder";
> 
> void startCameraServer();
> 
> void setup() {
>   Serial.begin(115200);
>   Serial.setDebugOutput(true);
>   Serial.println();
> 
>   camera_config_t config;
>   config.ledc_channel = LEDC_CHANNEL_0;
>   config.ledc_timer = LEDC_TIMER_0;
>   config.pin_d0 = Y2_GPIO_NUM;
>   config.pin_d1 = Y3_GPIO_NUM;
>   config.pin_d2 = Y4_GPIO_NUM;
>   config.pin_d3 = Y5_GPIO_NUM;
>   config.pin_d4 = Y6_GPIO_NUM;
>   config.pin_d5 = Y7_GPIO_NUM;
>   config.pin_d6 = Y8_GPIO_NUM;
>   config.pin_d7 = Y9_GPIO_NUM;
>   config.pin_xclk = XCLK_GPIO_NUM;
>   config.pin_pclk = PCLK_GPIO_NUM;
>   config.pin_vsync = VSYNC_GPIO_NUM;
>   config.pin_href = HREF_GPIO_NUM;
>   config.pin_sscb_sda = SIOD_GPIO_NUM;
>   config.pin_sscb_scl = SIOC_GPIO_NUM;
>   config.pin_pwdn = PWDN_GPIO_NUM;
>   config.pin_reset = RESET_GPIO_NUM;
>   config.xclk_freq_hz = 20000000;
>   config.pixel_format = PIXFORMAT_JPEG;
>   //init with high specs to pre-allocate larger buffers
>   if(psramFound()){
>     config.frame_size = FRAMESIZE_UXGA;
>     config.jpeg_quality = 10;
>     config.fb_count = 2;
>   } else {
>     config.frame_size = FRAMESIZE_SVGA;
>     config.jpeg_quality = 12;
>     config.fb_count = 1;
>   }
> 
> #if defined(CAMERA_MODEL_ESP_EYE)
>   pinMode(13, INPUT_PULLUP);
>   pinMode(14, INPUT_PULLUP);
> #endif
> 
>   // camera init
>   esp_err_t err = esp_camera_init(&config);
>   if (err != ESP_OK) {
>     Serial.printf("Camera init failed with error 0x%x", err);
>     return;
>   }
> 
>   sensor_t * s = esp_camera_sensor_get();
>   //initial sensors are flipped vertically and colors are a bit saturated
>   if (s->id.PID == OV3660_PID) {
>     s->set_vflip(s, 1);//flip it back
>     s->set_brightness(s, 1);//up the blightness just a bit
>     s->set_saturation(s, -2);//lower the saturation
>   }
>   //drop down frame size for higher initial frame rate
>   s->set_framesize(s, FRAMESIZE_QVGA);
> 
> #if defined(CAMERA_MODEL_M5STACK_WIDE)
>   s->set_vflip(s, 1);
>   s->set_hmirror(s, 1);
> #endif
> 
>   WiFi.begin(ssid, password);
> 
>   while (WiFi.status() != WL_CONNECTED) {
>     delay(500);
>     Serial.print(".");
>   }
>   Serial.println("");
>   Serial.println("WiFi connected");
> 
>   startCameraServer();
> 
>   Serial.print("Camera Ready! Use 'http://");
>   Serial.print(WiFi.localIP());
>   Serial.println("' to connect");
> }
> 
> void loop() {
>   // put your main code here, to run repeatedly:
>   delay(10000);
> }
> ```
> 
> ### Debug Messages:
> ```
> Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
> ```

same problem for me also, if you resolved this error.. will you please help me  resolve this error

ammiredd, 2022-04-15:
> Hi, got the problem because the ESP32 cannot connect to the wifi with the provided SSID & Password on the script code. After change / fix the value I've got it working so well

how you fix/change the value for connecting esp32-cam with wifi

visalvyasan, 2023-04-02:
> I figured it out. Let me know if you or someone are still working on this

Im having this issue

Duswanth, 2023-04-14:
> I figured it out. Let me know if you or someone are still working on this

its working out but not showing "wifi connected", and ip address in the serial monitor

morebray, 2023-08-13:
sss

morebray, 2023-08-13:
Hello, hbe also and still the problem like everyone else. Got me at AIEXPRESS several of these ESP-CAM modules with the USB board ESP32-CAM-MB, all do not work !
These boards are just bad !  Have contact problems with the camera, and the classic, the WIFI thing doesn't work, that has nothing to do with the SID or Passwd either ! 
There seem to be too many plagiarism manufacturers that do not deliver quality. 
I have programmed many of this series ESP32 (no CAM's) etc. that always worked reliably, but this CAm module is terrible ! 
Despite this I ask if anyone can get this CAM module to run properly and how he did it, please report. 
But please do not report without detailed instructions and indication of the source.... like ... "it works for me", that helps no one.  Thanks to all

6ReasoN2, 2024-01-08:
I know I'm very late to this topic but, just in case someone else made the same "OMG I'm a dumb@ss" mistake I did (and continues to think they've got their SSID and password correct)... there is no 5g capability on the esp32 boards afik, so if you see, for example, "TP-Link_5A80_5G", drop the "_5g" and you'll get an ip instead of .........................................................................................................................................