espressif/arduino-esp32 #3594: Enhancement request - add support for FS OTA updates.

mike-s123, 2019-12-23:
@Androbin

You added FS OTA upload support to ESP8266. Please do it for ESP32, it would make (my, at least) life much easier. :)

Androbin, 2019-12-23:
As far as I can tell, is already implemented for the ESP32:
[`espressif/arduino-esp32 : libraries/HTTPUpdate/src/HTTPUpdate.cpp`](https://github.com/espressif/arduino-esp32/blob/master/libraries/HTTPUpdate/src/HTTPUpdate.cpp)
It is based on (an older version of) the code for the ESP8266:
[`esp8266/Arduino : libraries/ESP8266httpUpdate/src/ESP8266httpUpdate.cpp`](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266httpUpdate/src/ESP8266httpUpdate.cpp)

I merely improved some things about the OTA process, but I'm willing to help nonetheless.
Do you want some updates carried over or are you just trying to figure out how to use it?

mike-s123, 2019-12-23:
What you pointed to lets the ESP pull a file from a web server. What I was looking for was the ability to update from a PC through a web page served by the ESP.

I've been using ESP8266HTTPUpdateServer.h on the ESP8266, and Update.h on the ESP32. 

I did a git pull a while back, and for the ESP8266 the option to upload a filesystem just magically appeared on the update page, I didn't change a thing in my code, it was a change in a library I was using. I found a commit you made related to FS upload on ESP8266, which is why I tagged you. Digging deeper, it looks like @caverna did a commit to ESP8266HTTPUpdateServer a couple of months ago which provided a front end change, while you're pointing out that support already exists at a deeper level.

ESP8266 is certainly much more refined - I think there were 3 lines of code I needed to add. ESP32, not so much, there's a bunch because there's no simple front end library available. Perhaps @karan6190 could update the ESP32 OTAWebUpdater example to include support for FileSystem uploads on the ESP32. 

This stuff isn't my day job, I struggle when things get too deep here, or I'd dive in deeper myself. I rely heavily on copy/paste from examples.

Androbin, 2019-12-23:
This should be relatively easy to do as the `Update` library is already there and you just need to pass it a stream of some kind. Both ESP8266 and ESP32 already have the code for initiating a download via a `HTTPClient` and streaming it to the `Update` library. The ESP8266 also has code for accepting uploads via a `WebServer`, which can be found here:
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPUpdateServer/src/ESP8266HTTPUpdateServer-impl.h
It should be possible to copy most of this code verbatim and use it on the ESP32.

Androbin, 2019-12-23:
The changeset by @caverna adapted the code to handle FS uploads as well as Sketch uploads.
The original code actually stems from a piece of example code by "John Doe":
[`esp8266/Arduino : commit/0d969e97600c76b83bb0f8f41fe981ae629c1d57`](https://github.com/esp8266/Arduino/commit/0d969e97600c76b83bb0f8f41fe981ae629c1d57#diff-2e37b12a29e5fb0dcb7695941ce6aca8)
It was later extracted and reworked into a library by @mangelajo:
[`esp8266/Arduino : commit/065e1ccb8ce7adbc3de1d0c9d0bf3c1e01bae8ca`](https://github.com/esp8266/Arduino/commit/065e1ccb8ce7adbc3de1d0c9d0bf3c1e01bae8ca#diff-7e33012bb077f9e6d203d394a13f2515)

lbernstone, 2019-12-25:
[This example](https://github.com/lbernstone/asyncUpdate/blob/master/AsyncUpdate.ino) uses ESPAsyncWebServer, but that may be an advantage for you, since it should work on both esp32 and 8266.

Androbin, 2019-12-26:
@lbernstone Great clean and tidy example code. Just why `UPDATE_SIZE_UNKNOWN` for ESP32?

I have found some example WebServer Update code that is maintained in this repository:
[`espressif/arduino-esp32 : libraries/ArduinoOTA/examples/OTAWebUpdater/OTAWebUpdater.ino`](https://github.com/espressif/arduino-esp32/blob/master/libraries/ArduinoOTA/examples/OTAWebUpdater/OTAWebUpdater.ino)
The frontend is horribly bloated, but other than that you just gotta change the `Update.begin` line.

Androbin, 2019-12-26:
Didn't notice it at first because `ArduinoOTA` is its own library. At least it is in the ESP8266 repository.

lbernstone, 2019-12-28:
IIRC, esp32 will accept compressed files, so the file size as presented in the header cannot be trusted as the sketch size.  So, you must use UPDATE_SIZE_UNKNOWN.

closed 2020-03-11 by stale[bot]