espressif/arduino-esp32 #3595: documentation missing about WebServer example "HelloServer.ino"

dsyleixa, 2019-12-24:
a detailed documentation is missing about WebServer example "HelloServer.ino":
what is #include <ESPmDNS.h> for?
what is void handleRoot() for?
what is void handleNotFound() for?
what is   server.on("/", handleRoot) for?
what is  server.on("/inline", []() for?
what is server.onNotFound(handleNotFound) for?
what is server.handleClient() for?

if a documentation is provided though, please insert a link to thet description into the source code.
Thank you in advance!

```
#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>
#include <ESPmDNS.h>

const char* ssid = "........";
const char* password = "........";

WebServer server(80);

const int led = 13;

void handleRoot() {
  digitalWrite(led, 1);
  server.send(200, "text/plain", "hello from esp8266!");
  digitalWrite(led, 0);
}

void handleNotFound() {
  digitalWrite(led, 1);
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET) ? "GET" : "POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i = 0; i < server.args(); i++) {
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
  digitalWrite(led, 0);
}

void setup(void) {
  pinMode(led, OUTPUT);
  digitalWrite(led, 0);
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  if (MDNS.begin("esp32")) {
    Serial.println("MDNS responder started");
  }

  server.on("/", handleRoot);

  server.on("/inline", []() {
    server.send(200, "text/plain", "this works as well");
  });

  server.onNotFound(handleNotFound);

  server.begin();
  Serial.println("HTTP server started");
}

void loop(void) {
  server.handleClient();
}
```

dsyleixa, 2020-02-23:
please provide a detailed example documentation and elaborated code commentation!

dsyleixa, 2020-04-23:
as to stale_set:
this issue is not solved yet, please provide multiple detailed examples, step by step based on each other, feat. documentations and elaborated code commentation!

noorhaq, 2020-04-29:
I have tried to answer all of your questions by adding comments to the code you have poster above. Hopefully, this will be of some help.

`#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>
#include <ESPmDNS.h>

const char* ssid = "........";
const char* password = "........";

WebServer server(80);

const int led = 13;

void handleRoot() { //It is been used to blink the ESP32 LED to show some transfer of data and the data being sent here is "hello from esp 8266"
  digitalWrite(led, 1);
  server.send(200, "text/plain", "hello from esp8266!");  // ESP32 is the succesor of esp8266. And both boards run on the same artitecture with esp 32 having faster CPU,more ram and GPIOs.
  digitalWrite(led, 0);
}

void handleNotFound() { //In case you are trying to connect to a server. Or to open a file and the file is not accessble or does not exist. This error msg will be displayed. If function not included the whole project will stop working .
  digitalWrite(led, 1);
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET) ? "GET" : "POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i = 0; i < server.args(); i++) {
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
  digitalWrite(led, 0);
}

void setup(void) {
  pinMode(led, OUTPUT);
  digitalWrite(led, 0);
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  if (MDNS.begin("esp32")) {
    Serial.println("MDNS responder started");
  }

  server.on("/", handleRoot); // Associate the handler function to the path.

  server.on("/inline", []() {
    server.send(200, "text/plain", "this works as well");
  });

  server.onNotFound(handleNotFound); //if Server is not found it will call the above handleNotFound function.

  server.begin();
  Serial.println("HTTP server started");
}

void loop(void) {
  server.handleClient(); //server. handleClient() then looks for pending requests, matches the requested address and executes the code that was registered in setup. Where the variable httpout has been filled by a previous function that measures things and generates the resulting HTML. You can also read more on ot on https://www.reddit.com/r/esp8266/comments/942579/what_does_serverhandleclient_do/
}`

dsyleixa, 2020-04-29:
thank you very much, there are really helpful comments!
e.g. 1 is still not clear: what is this for, and can it perhaps be dropped completely? Or is it inevitible?

server.on("/inline", {
server.send(200, "text/plain", "this works as well");
});

noorhaq, 2020-04-29:
This can be dropped if you want. it just sending a message to the server just like the above one. You can read more about inline functions which on http://www.cplusplus.com/articles/2LywvCM9/ for a better understanding.

dsyleixa, 2020-04-29:
thank you, ok, so IIUC:

for server.on("/", handleRoot)  there must be an extra function void handleRoot(),
but for server.on("/inline",...) all what shall happen is written directly behind and inside this.

Nonetheless, for both of them somewhere  "/" or alternatively "/inline" must be called by someone somehow sometimes?
But were and how exactly?

lbernstone, 2020-04-29:
What do you think a WebServer does?

dsyleixa, 2020-07-01:
still not solved.

dsyleixa, 2020-08-30:
still not solved.

goliat58, 2020-08-30:
super !

dsyleixa, 2020-10-31:
still no solution yet!

dsyleixa, 2020-12-31:
still no solution yet AFAIK

dsyleixa, 2021-06-21:
still no solution yet, especially explanations about wha the webserver and client codes actualy do and how they work and communicate.

closed 2021-08-21 by dsyleixa