espressif/arduino-esp32 #3598: TLS session resumption

smdjeff, 2019-12-24:
It seems most ESP32 users are setting up and tearing down the TLS session for each and every transaction.  Unfortunately, "Using TLS on the ESP32 requires a significant amount of energy. Compared to an unencrypted transmission, approximately 14 times more energy is required to establish a full TLS session." via
http://www2.inf.h-brs.de/~mradem2m/files/myPaperVersionsForWebsite/2019-ESP32-TLS-Power.pdf

Has anyone attempted TLS session resumption?  Perhaps something like esp_tls_conn_http_new() would return a resumption ticket id that can be used as input into the esp_tls_cfg_t parameter in the next call?

smdjeff, 2020-02-22:
No one?

closed 2020-05-06 by stale[bot]

jbbr, 2020-10-02:
I'm also interested in this feature 
(hope adding a comment will reopen the issue)