espressif/arduino-esp32 #3599: strange error:

dsyleixa, 2019-12-24:
Arduino IDE: 1.8.9
ESP32 board core: 1.0.4
(automatic default settings)

strange error aftr upload:  
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d467a  PS      : 0x00060630  A0      : 0x800d4b1f  A1      : 0x3ffb1ea0  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x3ffc138c  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000020  
A14     : 0x3ffc1a8c  A15     : 0x3ffc1a4c  SAR     : 0x00000011  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc  

Backtrace: 0x400d467a:0x3ffb1ea0 0x400d4b1c:0x3ffb1ee0 0x400d13d0:0x3ffb1f10 0x400d2440:0x3ffb1f30 0x400dd7ef:0x3ffb1fb0 0x4008954d:0x3ffb1fd0

Rebooting...

setup code:
```
void setup() {
   Serial.begin(500000);
   delay(500);

   //LED_pin=LED_BUILTIN;  // default: LED_pin=LED_BUILTIN
   pinMode(LED_pin, OUTPUT);

   //---------------------------------------------------------
   Adafruit_HX8357_ini(1);  // init function in lib <display_HX3857.h>

   COLOR_BGND = BLACK;
   COLOR_TEXT = WHITE;
   display.fillScreen(COLOR_BGND);
   display.setTextColor(WHITE);
   display.setTextSize(2);
   //display.setFont(&FreeMono9pt7b);
   Serial.println("setup(): display setup done!");
   Serial.println();
   
   //---------------------------------------------------------
   //TS buttons
   TSbutton1.initButton(&display, display.width()-30, 20,  60,30,  CYAN, BLUE, YELLOW, "Btn1", 2);
   TSbutton2.initButton(&display, display.width()-30,100,  60,30,  CYAN, BLUE, YELLOW, "Btn2", 2);
   TSbutton3.initButton(&display, display.width()-30,180,  60,30,  CYAN, BLUE, YELLOW, "Btn3", 2);
   TSbutton4.initButton(&display, display.width()-30,260,  60,30,  CYAN, BLUE, YELLOW, "Btn4", 2);

   Serial.println("setup(): ts buttons setup done!");
   Serial.println();
   

   //---------------------------------------------------------
   // SD
   if( !SDioerr ) {
      Serial.println("SD.begin(SD_CS) failed!");
      Serial.println();      
      delay(2000); // delay here
   }
   else {
      SdPath = SD.open("/");
      Serial.println("setup(): SD setup done!\n");      
   }

   //---------------------------------------------------------
   //i2c Wire

   Wire.begin();
   Wire.setClock(400000);
   ads0.begin();
   Serial.println("setup(): i2c+ads1115 setup done!\n");
   
   init_MPU6050();

   //-----------------------------------------------------
   // connecting to router
   //-----------------------------------------------------
   Serial.print("try WiFi, localIP "); Serial.println(local_ip);
   Serial.println();
   display.setTextColor(YELLOW);   
   delay(1);

   //WiFi.mode(WIFI_STA);   delay(1);
   WiFi.config(local_ip, gateway, subnet);
   WiFi.begin(ssid, password);
   delay(1);

   // Wait for connection
   while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
   }
   delay(1);
   Serial.println();
   Serial.print("  Connected to ");   Serial.print(ssid);
   Serial.print("  IP address: ");
   Serial.println(WiFi.localIP());
   Serial.println();

   delay(1);

   //---------------------------------------------------------
   //init and get the time
   delay(1);
   configTime(gmtOffset_sec, daylightOffset_sec, ntpServer);
   delay(1);
   
   Serial.println("\nWaiting for time");
   unsigned start = millis();
   while (!time(nullptr))
   {
      Serial.print(".");
      delay(1000);
   }
   delay(1000);

   Serial.println("Time...");
   currenttime = time(nullptr);
   String StrBuf=ctime(&currenttime);
   
   Serial.print(StrBuf);
   Serial.println();
   
   delay(100);

   //---------------------------------------------------------
   // WebServer

   server.on("/", handleRoot);
   server.on("/inline", []() {
      server.send(200, "text/plain", "this works as well");
   });
   server.onNotFound(handleNotFound);

   server.begin();
   Serial.println("HTTP server started!");
   delay(1);

   //---------------------------------------------------------
   // end of setup
   Serial.println();
   Serial.println("setup(): done!\n");
   
   delay(3000);   

}

```

atanisoft, 2019-12-24:
Decode the backtrace and it will tell you where/why it is crashing. Most likely it is an uninitialized variable or other allocation failure

dsyleixa, 2019-12-25:
Merry Christmas!
Unfortunately, I honestlydon't understand what you mean.

atanisoft, 2019-12-25:
Merry Christmas to you as well.

Install this plug-in for the Arduino IDE: https://github.com/me-no-dev/EspExceptionDecoder. A new menu item should show up under Tools (I believe). Copy and paste the crash output (Guru Mediation lines until the Rebooting line) and it should print out the source line(s) which caused the failure.

dsyleixa, 2019-12-25:
I do not understand how to do that, tbh

atanisoft, 2019-12-25:
The readme at the link above should have instructions on how to install it.

dsyleixa, 2019-12-25:
I certainly read it, but as stated, I don't understand it.

VincentJGeisler, 2020-01-23:
I have exactly the same problem.

here is the decoder output.  It looks like dhcp offer is craping the bed

PC: 0x00000000
EXCVADDR: 0x00000000

Decoding stack results
0x40113ec6: handle_dhcp at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/apps/dhcpserver/dhcpserver.c line 1031
0x40120d4d: udp_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/udp.c line 401
0x40125d55: ip4_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c line 740
0x4012aff2: ethernet_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/netif/ethernet.c line 184
0x401148cf: tcpip_thread at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c line 135
0x4008b439: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

dsyleixa, 2020-01-23:
hi, 
thank you! Not sure evidentially, but as to dhcp/dns  it seems I could fix that FTM (?) by using
WiFi.config(local_ip, gateway, subnet, gateway, gateway); // gateway = dns

VincentJGeisler, 2020-01-23:
I've been hacking around at it, and i think it may be some sort of race condition between the DHCP and SoftAP. 

    WiFi.disconnect(); //added to start with the wifi off, avoid crashing
    WiFi.mode(WIFI_OFF); //added to start with the wifi off, avoid crashing
    WiFi.mode(WIFI_AP);
    delay(2000); //settle in time to prevent crashing, yea the RTOS is really sensitive
    WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));//configure the AP
    WiFi.softAP("test1", "", 6);//configure soft ao mode, ssid, password, channel`

closed 2020-04-06 by stale[bot]