espressif/arduino-esp32 #3600: bug: WiFi.config(local_ip, gateway, subnet); corrupts <time.h> functions configTime(), time(nullptr);

dsyleixa, 2019-12-25:
issue:
WiFi.config(local_ip, gateway, subnet); corrupts <time.h> functions configTime(), time(nullptr);
it always shows 1970 date/time:
Thu Jan  1 01:00:02 1970
Thu Jan  1 01:00:02 1970
Thu Jan  1 01:00:03 1970
Thu Jan  1 01:00:03 1970

board: ESP32 core 1.0.4 (Adafruit Feather ESP32)
Arduino IDE 1.8.9
default settings

OTOH, when out-commenting WiFi.config(local_ip, gateway, subnet);
it works correctly!
Wed Dec 25 11:11:59 2019
Wed Dec 25 11:12:00 2019
Wed Dec 25 11:12:00 2019
Wed Dec 25 11:12:01 2019
Wed Dec 25 11:12:01 2019

code:
```

#include "time.h"
#include <ESP8266WiFi.h>

const char* ssid     = "*****";                    // your network SSID (name)
const char* password = "**********";       // your network password

IPAddress local_ip(192, 168,  2, 222);
IPAddress gateway (192, 168,  2,  1);
IPAddress subnet  (255, 255, 255, 0);

const char* ntpServer = "pool.ntp.org";
const long  gmtOffset_sec = 0;
const int   daylightOffset_sec = 3600;

void setup()
{
  Serial.begin(115200);
  delay(10);
 
  Serial.print("\n\nConnecting to ");
  Serial.println(ssid);
 
  WiFi.mode(WIFI_STA);
  WiFi.config(local_ip, gateway, subnet);  // <<<<<<<<<<  !!!! works only if out-commented
  WiFi.begin(ssid, password);
 
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
 
  Serial.println();
  Serial.println("WiFi connected");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
 
 configTime(gmtOffset_sec, daylightOffset_sec, ntpServer);
 
  Serial.println("\nWaiting for time");
  unsigned timeout = 5000;
  unsigned start = millis();
  while (!time(nullptr))
  {
    Serial.print(".");
    delay(1000);
  }
  delay(1000);
 
  Serial.println("Time...");
}

void loop()
{
  time_t current = time(nullptr);
  Serial.print(ctime(&current));
  delay(500);
}
```

lbernstone, 2019-12-25:
You haven't set a dns server, and are using a name for the ntp server.

dsyleixa, 2019-12-25:
why is that indispensible?
how to set a dns server?

lbernstone, 2019-12-25:
https://en.wikipedia.org/wiki/Domain_Name_System
https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiSTA.h#L43

dsyleixa, 2019-12-25:
thank you,I'll try it!
do I need to set 2 dns servers?

rel1ct, 2019-12-25:
**dsyleixa**, try this sketch.
```cpp
#include "time.h"
//#include <ESP8266WiFi.h>
#include <WiFi.h>

const char* ssid     = "*****";           // your network SSID (name)
const char* password = "**********";       // your network password

IPAddress local_ip(192, 168,  2, 222);
IPAddress gateway (192, 168,  2,  1);
IPAddress subnet  (255, 255, 255, 0);

const char* ntpServer = "pool.ntp.org";
const long  gmtOffset_sec = 3600;
const int   daylightOffset_sec = 0;

void WiFiEvent(WiFiEvent_t event){
  switch (event) {
    case SYSTEM_EVENT_STA_START:
      WiFi.config(local_ip, gateway, subnet);  // <<<<<<<<<<  !!!! works only if out-commented
    break;
  }
}

void setup()
{
  Serial.begin(115200);
  delay(10);
 
  Serial.print("\n\nConnecting to ");
  Serial.println(ssid);
 
  WiFi.mode(WIFI_STA);
  WiFi.onEvent(WiFiEvent);
  WiFi.begin(ssid, password);
 
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
 
  Serial.println();
  Serial.println("WiFi connected");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
 
 configTime(gmtOffset_sec, daylightOffset_sec, ntpServer);
 
  Serial.println("\nWaiting for time");
  unsigned start = millis();
  while (!time(nullptr))
  {
    Serial.print(".");
    delay(1000);
  }
  delay(1000);
 
  Serial.println("Time...");
}

void loop()
{
  time_t current = time(nullptr);
  Serial.println(ctime(&current));
  delay(500);
}
```

dsyleixa, 2019-12-25:
@rel1ct 
would you perhaps comment on this a little more?
(cmment parameter -v  8) )

rel1ct, 2019-12-25:
I don’t see the point.
Your previous post:

WiFi.config(local_ip, **gateway**, subnet, **gateway**, dns1, dns2)
WiFi.config(local_ip, **gateway**, subnet, **gateway**)

Comment:
I see this nonsense not the first time.

dsyleixa, 2019-12-25:
@rel1ct
I rate your post as senseless and accusing. You altered my TOP code, in my context the issue report is valid.

dsyleixa, 2019-12-25:
@rel1ct
I rate your post as senseless, accusing and insulting. You changed my TOP code, in my reported context the issue report is valid, as also stated here: https://github.com/espressif/arduino-esp32/issues/3600#issuecomment-568907808

atanisoft, 2019-12-25:
@dsyleixa you should only need to add one DNS server and it should be the fourth parameter and not gateway as seen in @rel1ct post above.

Calling config from the event handler is an alternative way to configure it whereas you code set the mode and then config before starting the SSID connection process.

dsyleixa, 2019-12-26:
@atanisoft,
thank you for your reply, and your advice works now, too.
My point actually was:
1) having not used WiFi.config() at all, the time.h functions work well,
2) using WiFi.config() with a constant local_IP *router+subnet it fails, 
3) it then needed
an extra dns-server.
If Wifi.begin works witout any WiFi.config() , wha does it fail then for (2)?
Obviously the dnssettings of (1)  are made automatically correct by default (probaly using the gateway IP) ,
so why doesn't it do it for (2)?
Expectedly, if no dns-server is set, then the gateway IP is supposed to be taken for the dns-server as well.

OTOH, using an event handler is actually totally weird, no common Arduino user (i.e., hobby programmers like pupils, craftsmen,  or artists) would ever do that for that purpose.

atanisoft, 2019-12-26:
By not calling WiFi.config() it defaults to DHCP which configures all parameters based on what the AP provides as a response to the DHCP request, it does not use the gateway as a default.

By calling WiFi.config() without a DNS server address the WiFi driver will not be able to resolve hostnames. There is no default value used when not configured, it will not use the gateway for DNS. I don't know where that might be coming from as the code does not do that today. The API should require at least one DNS address to be provided or it should provide a default (8.8.8.8/1.1.1.1 as example)

As for event handlers, there are a few Arduino libraries that use them. But in general, it is a bit of an advanced topic that is not common in most AVR Arduino examples. That is mostly due to limitations of the platform (flash and ram size mostly).

dsyleixa, 2019-12-26:
ok, I see, thanks for clarification.

rel1ct, 2019-12-26:
> OTOH, using an event handler is actually totally weird, no common Arduino user (i.e., hobby programmers like pupils, craftsmen, or artists) would ever do that for that purpose.

You have been offered a solution and you are unhappy.
Good luck finding non-existent bugs.

dsyleixa, 2019-12-26:
is there a forum's limitation to arrogance and smugness...? ;)

dsyleixa, 2019-12-26:
is there a forum's limitation to arrogance, smugness, or stubbornness ...? ;)
edit: 
fyi, the issue is meanwhile  resolverd...

lbernstone, 2019-12-26:
Please just close the issue if it is resolved.

closed 2019-12-28 by dsyleixa