espressif/arduino-esp32 #3603: SD.exists("/") works faulty

dsyleixa, 2019-12-31:
in case the following issue is caused by dedicated  ESP libs, not by the original Arduino libs:

SD.exists(filename) works faulty!
If the SD has been started correctly by SD.begin(SD_CS) and the SD should be checked if intermediately mistakenly removed, it always shows "OK" and never shows an error:

   if( SD.exists("/") ) {
      Serial.println(" SD path '/' OK!  ");  // either if SD inserted or removed !!!
      // debug
      delay(2000);
   }
   else {
      Serial.println( "error! no SD file path found!  "); 
      // debug
      delay(2000);
   }

Arduino IDE 1.8.9
ESP32 board 1.0.4

dsyleixa, 2019-12-31:
PS, update, just checked:
on an original Arduino board  (Arduino Due)  this fault is not happening, the code above works fine as expected.
Appearently the Arduino SD libs are fine.
So the fauly behaviour must be due to ESP SD libs.

dsyleixa, 2019-12-31:
@rel1ct: what do you mean by your effing emojies?

lbernstone, 2019-12-31:
Your expectations really could use some work.  This is a community project- nobody here works for you.  If you think the behavior is wrong, work to fix it.  Saying "this is broken" is lazy and boorish.
From a posix standpoint, checking whether a mount point exists should always return true.  Instead of using the bad workaround that some Duo example used, think about a better way to do this.  Checking the size of the filesystem should be more portable and reliable.

dsyleixa, 2019-12-31:
I have no idean, I am a simple Arduino user, a hobby programmer, no CS, technician,  or programming education, learning just from example sketches.
I reported the bug in the hope that it can be fixed by pros and experts.
As stated, the original Arduino API functions for SD on an Arduino Due (same SD used for testing) works well though.

dsyleixa, 2019-12-31:
I have no idea about driver code or technical failures, I am a simple Arduino user, a hobby programmer (like almost all common arduino users, for whom Arduino/Wiring had been developed),
I have no CS, no technician or programming education, learning just from example sketches.
I reported the bug in the hope that it will be of interest and that it can be fixed by pros and/or experts.
As stated, the original Arduino API functions for SD on an Arduino Due (same SD used for testing) works well though.
Happy new year!

rel1ct, 2020-01-01:
Where did you read that the libraries and functionality for ESP32 should be completely identical to Arduino Due or ESP8266?

dsyleixa, 2020-01-01:
the way it's currently working for Arduino is fine, and the way how it's currently working for ESP is faulty - that's the crucial point which I had reported above.

rel1ct, 2020-01-03:
What is your understanding of the "original" Arduino?

lbernstone, 2020-01-03:
@dsyleixa : I will need to get my sd card rigged up so I can fix this.  Expect a PR soon.

dsyleixa, 2020-02-05:
will there be a fix of the broken SD lib functions 
SD.exists("/")
SD.begin() 
to actually "really" check or restart anew?

dsyleixa, 2020-04-23:
will there soon be some fixes available?

dsyleixa, 2020-06-22:
issue is still unresolved

dsyleixa, 2021-06-23:
still no report of being  solved

dsyleixa, 2021-08-22:
it still is not resolved, still works faulty.

P-R-O-C-H-Y, 2021-09-16:
Hi @dsyleixa,

When using SD.exist() the file you looking for may be found in a cached directory block. The only sure way to detect removal or inserting of the SD is to use the SD detect switch (CD pin) on the SD socket. 

Set pull up input on a CD pin and then just call digitalRead.
If pin is LOW = card is detected, if HIGH = no card inserted

Try this:

```
void setup() 
{
  // Init serial, SD, etc..

  // Set pull up input on a CD pin
  pinMode(SD_CD,INPUT_PULLUP);
}

void loop() 
{
  
  bool SD_detection = digitalRead(SD_CD); // 1 = no card inserted; 0 = card detected
  
  if (!SD_detection)
  {
      Serial.println("card detected");
  }
  else
  {    
      Serial.println("card removed");
   }
  
  // rest of code
}
```

dsyleixa, 2021-09-17:
thank you for your suggestion! I can't try it currently anymore because my device is defective but will try it out ASAP.
OTOH, this is a functionality whicxh is supposed to be supported by  SD.exist() internally and should work correctly out of the box.

dsyleixa, 2021-09-17:
thank you for your suggestion! I can't try it currently anymore because my device is defective but will try it out ASAP.
OTOH, this is a functionality which is supposed to be processed by SD.exist() internally and should work correctly out of the box.

dsyleixa, 2021-09-17:
thank you for your suggestion! I can't try it currently anymore because my device is defective but will try it out ASAP.

Additionally, I have just these pins provided, no more than that:
   SD card attached to SPI bus as follows:
```

 ** MOSI - pin 11
 ** MISO - pin 12
 ** CLK - pin 13
 ** CS - pin 4 

void setup() {
  SD.begin(SD_CS);
  //...
}
```

OTOH, the SD-check is a functionality which is supposed to be processed by SD.exist() internally and should work correctly out of the box.

dsyleixa, 2021-09-17:
thank you for your suggestion! I can't try it currently anymore because my device is defective but will try it out ASAP.

Additionally, I have just these pins provided, no more than that:
```
/*
 SD card attached to SPI bus as follows:
 ** MOSI - pin 11
 ** MISO - pin 12
 ** CLK - pin 13
 ** CS - pin 4 
*/

void setup() {
  SD.begin(SD_CS);
  //...
}
```

OTOH, the SD-check is a functionality which is supposed to be processed by SD.exist() internally and should work correctly out of the box.

dsyleixa, 2021-09-17:
thank you for your suggestion! I can't try it currently anymore because my device is defective but will try it out ASAP.

Additionally, I have just these pins provided, no more than that (SD slot is soldered to an Adafruit Feather TFT shield):
```
/*
 SD card attached to SPI bus as follows:
// *SNIP*: TFT & TS shield pin settings
// SD pin:
#define SD_CS    14

*/

void setup() {
  SD.begin(SD_CS);
  //...
}
```

OTOH, the SD-check is a functionality which is supposed to be processed by SD.exist() internally and should work correctly out of the box.

dsyleixa, 2021-09-17:
thank you for your suggestion! I can't try it currently anymore because my device is defective but will try it out ASAP.

Additionally, I have just these pins provided, no more than that (SD slot is soldered to an Adafruit Feather TFT shield):
```

// MOSI - pin 11
// MISO - pin 12
// CLK - pin 13

// *SNIP*: TFT & TS shield pin settings
// SD pin:
#define SD_CS    14

void setup() {
  SD.begin(SD_CS);
  //...
}
```

OTOH, the SD-check is a functionality which is supposed to be processed by SD.exist() internally and should work correctly out of the box.

dsyleixa, 2021-09-17:
thank you for your suggestion! I can't try it currently anymore because my device is defective.

Additionally, I have just these pins provided, no more than that (SD slot is soldered to an Adafruit Feather TFT shield), so IIUC I'm afraid such a SD_CD pin is not available:
```

// MOSI - pin 11
// MISO - pin 12
// CLK - pin 13

// *SNIP*: TFT & TS shield pin settings
// SD pin:
#define SD_CS    14

void setup() {
  SD.begin(SD_CS);
  //...
}
```

OTOH, the SD-check is a functionality which is supposed to be processed by SD.exist() internally and should work correctly out of the box.

P-R-O-C-H-Y, 2021-09-17:
Can you please share what exact Adafruit Feather TFT board you have? In the SD socket there is always the CD pin but some boards use it and some not.

For best performance the FileSystem is cached so even if you call SD.exist("/") or SD.open("/foo.txt",FILE_READ). they will always return true until you call SD.end().

dsyleixa, 2021-09-17:
I have the 480x320  3,5" Feather-Wing https://learn.adafruit.com/adafruit-3-5-tft-featherwing 
anaway, as stated, I actually expect the ESP32 SD lib to handle that issue, perhaps by disabling internally any status caching.

dsyleixa, 2021-09-17:
I have a 3-years old 480x320  3,5" Feather-Wing https://learn.adafruit.com/adafruit-3-5-tft-featherwing  plus a 3-years old version of Adafruit ESP32 Huzzah https://www.adafruit.com/product/3405 
anyway, as stated, I actually expect the ESP32 SD lib to handle that issue, perhaps by disabling internally any status caching.
Just to mention that I use SD.h, not SdFat.h or any other SD lib.

P-R-O-C-H-Y, 2021-10-05:
We internally use the FS in our SD-library. So every time you call SD.Exist("foo.txt") and file was existing on the card before removing and calling SD.end, it will find the file in cache. 
The Sd.exist should not be used to determine if SD card is removed.
If you don't have the CD pin, this workaround will work for you to detect if the card is removed.

Sample code:
```
bool SdExist(fs::FS &fs){
    Serial.println("Creating foo.txt file to detect the SD card");
    if(SD.open("/fooo.txt","w")) 
    {
        SD.remove("/fooo.txt");
        Serial.println("Succesfuly created and removed.");
        Serial.println("SD card detected");
        return true;
    }
    else 
    {
        Serial.println("Error in creating file");
        Serial.println("SD card not detected");
        return false;
    }
}
```
In loop just call the function:
```
if(!SdExist(SD)){
        //SD card removed
}

```

dsyleixa, 2021-10-05:
thank you, which lib is it using? SD.h?

P-R-O-C-H-Y, 2021-10-05:
Yes the SD lib.
```#include <SD.h>```

dsyleixa, 2021-10-05:
that's great, thank you very much!
I have to admit that  I don't understand the meaning of the passed variant
SdExist(fs::FS &fs)
but IIUC I can be sure that the result of file create/remove will not be stored in any cache when called several times repeatedly?

P-R-O-C-H-Y, 2021-10-05:
You are passing SD (FS) class you are working with to the function SdExist(fs::FS &fs) .I assume you work with default SD class already initialized in SD.h file:
```extern fs::SDFS SD;```
But if you want to use own class instead the function is ready to get your class passed to it.
Like functions in examples of SD library are made exactly like this function.

But if you use the default SD class you can change the function SdExist just to this:
```
bool SdExist(void){
    Serial.println("Creating foo.txt file to detect the SD card");
    if(SD.open("/fooo.txt","w")) 
    {
        SD.remove("/fooo.txt");
        Serial.println("Succesfuly created and removed.");
        Serial.println("SD card detected");
        return true;
    }
    else 
    {
        Serial.println("Error in creating file");
        Serial.println("SD card not detected");
        return false;
    }
}
```

Yes you can be sure with that result, if the **file does not exist before calling the function**. It needs to create the file. If the file is existing, it will be found in cache. Thats why there is the remove after creating the file, so next time it needs to be created again.

dsyleixa, 2021-10-05:
I do/did not declare, initialize, or instantiate "SD" in my program by myself explicitely (IIRC), so I guess SD is something which is provided by SD.h by default just in the moment when I include this lib.
Again, thank you very much for your very helpful contributions!

dsyleixa, 2021-10-05:
I do/did not declare or instantiate "SD" in my program by myself explicitely (IIRC), so I guess SD is something which is provided by SD.h by default just in the moment when I include this lib when calling SD.begin(SD_CS).
Again, thank you very much for your very helpful contributions!

P-R-O-C-H-Y, 2021-10-06:
Yes exactly, you use the default one decleared in SD.h file.
So I can consider this issue as closed ?

P-R-O-C-H-Y, 2021-12-07:
Hi, made an PR to fix the problem for SD library. The SD_MMC already contains fix on master branch (updated esp-idf already merged).So now when you remove card from slot, the SD.exist will return false.
Once merged, both SD and SD_MMC will work the same as expected :)

closed 2021-12-14 by me-no-dev

dsyleixa, 2021-12-14:
hi,
what do I have to do if I want to use the new fix?

P-R-O-C-H-Y, 2021-12-15:
You can download the master branch from github and replace the library in Arduino IDE.
Link: https://github.com/espressif/arduino-esp32/archive/refs/heads/master.zip

dsyleixa, 2021-12-15:
ah, no, thanks, I would prefer to wait until it is available in the Arduino IDE out of the box, e.g. by updating the ESP core version
(currently:
ESP32 board 1.0.6
default settings)

dsyleixa, 2021-12-15:
in my Arduino IDE 1.8.9 I can only choose one of 1.0.0. to 1.0.6,
Board Manager URL in Arduino preferences: https://dl.espressif.com/dl/package_esp32_index.json

P-R-O-C-H-Y, 2021-12-15:
There are new URLs for board manager.
Refer to [Installing Arduino-esp32](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html)
For stable releases use this: 
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
For development releases use this: 
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json

dsyleixa, 2021-12-15:
thank you, never got a notification that my old url had to be changed, now I also have 2.0.1.

dsyleixa, 2022-03-12:
still not solved. versions 2.0.x are working faulty with SD access.

P-R-O-C-H-Y, 2022-03-23:
@dsyleixa 
Everything is fixed in master branch now. You can use that or wait for release 2.0.3 for fixes. Even read speed will be improved.
Thanks

dsyleixa, 2022-03-24:
in the Arduino Board manager? which version?

P-R-O-C-H-Y, 2022-03-25:
Not yet in the Arduino Board manager. All fixes are planned for version 2.0.3 but you will have to wait for release to be able to update in Arduinos board manager.
BUT you can clone Arduino-esp32 master branch to your Arduino. Instructions can be found [here](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#windows-manual-installation)

dsyleixa, 2022-03-25:
thank you, then I will have to wait!

P-R-O-C-H-Y, 2022-03-30:
@dsyleixa You can now update from Arduino Board Manager to 2.0.3-RC1 version :) So you can give it a shot.

dsyleixa, 2022-04-01:
thanks, but unfortunately only 2.0.1 and 2.0.2 are available!

VojtechBartoska, 2022-04-04:
@dsyleixa Take a look on this page: https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html

You need to change to development release link in your Arduino IDE preferences.

dsyleixa, 2022-04-07:
I do not understand, tbh: 
I always can select either  board release version via the Arduino IDE board manager, for ESP32 everything up to 2.0.2 is available:
so why not 2.0.3?
what exactly do I have to change additionally?

VojtechBartoska, 2022-04-07:
@dsyleixa It's because we released only Release candidate. It's not a stable release version, because of that, link needs to be changed to Development release link.

dsyleixa, 2022-04-07:
thanks, but as already stated I will only  install stable versions via the Arduino board manager.

dsyleixa, 2022-04-07:
thanks, but as already stated I will only  install stable versions via the Arduino board manager and I won't change any links which perhaps may mess up everything for all my other programs and projects..

dsyleixa, 2022-12-08:
2.0.4 and 2.0.5 do not compile anymore at all1

dsyleixa, 2022-12-08:
2.0.4 and 2.0.5 do not compile anymore at all!

P-R-O-C-H-Y, 2022-12-12:
@dsyleixa can you post the output you are getting?

dsyleixa, 2022-12-12:
for 1.0.6 it's
```
"D:\\Arduino\\portable\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-size" -A "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_16622/ESP32_Box031i.ino.elf"
Der Sketch verwendet 861538 Bytes (65%) des Programmspeicherplatzes. Das Maximum sind 1310720 Bytes.
Globale Variablen verwenden 124572 Bytes (38%) des dynamischen Speichers, 203108 Bytes für lokale Variablen verbleiben. Das Maximum sind 327680 Bytes.
```
for 2.0.5 it's far too big, excerpt:
```
Detecting libraries used...
(...)
"-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/esp_netif/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/esp_eth/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/tcpip_adapter/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/esp_phy/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/esp_phy/esp32/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/esp_ipc/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/app_trace/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/esp_timer/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/mbedtls/port/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/mbedtls/mbedtls/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/mbedtls/esp_crt_bundle/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/app_update/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/spi_flash/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/bootloader_support/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/nvs_flash/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/pthread/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/esp_gdbstub/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/esp_gdbstub/xtensa" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/esp_gdbstub/esp32" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/espcoredump/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/espcoredump/include/port/xtensa" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/wpa_supplicant/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/wpa_supplicant/port/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/wpa_supplicant/esp_supplicant/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/ieee802154/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/console" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/asio/asio/asio/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/asio/port/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/bt/common/osi/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/bt/include/esp32/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/bt/common/api/include/api" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/bt/common/btc/profile/esp/blufi/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/bt/common/btc/profile/esp/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/bt/host/bluedroid/api/include/api" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/bt/esp_ble_mesh/mesh_common/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/bt/esp_ble_mesh/mesh_common/tinycrypt/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/bt/esp_ble_mesh/mesh_core" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/bt/esp_ble_mesh/mesh_core/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/bt/esp_ble_mesh/mesh_core/storage" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/bt/esp_ble_mesh/btc/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/bt/esp_ble_mesh/mesh_models/common/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/include/bt/esp_ble_mesh/mesh_models/client/include" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\2. [... cut ...] \\portable\\packages\\esp32\\hardware\\esp32\\2.0.5/tools/sdk/esp32/dio_qspi/include" -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -ggdb -Os -freorder-blocks -Wwrite-strings -fstack-protector -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -std=gnu++11 -fexceptions -fno-rtti -MMD -c -w -D

dsyleixa, 2022-12-12:
as reported in the other linkk:
Update:
I added  `#include <vector>`
but now still a compile error persisting, not mentioning std::vector anymore though

```
Arduino: 1.8.9 (Windows 10), Board: "Adafruit ESP32 Feather, 80MHz, 921600, None, Default"

D:\Arduino\arduino-builder -dump-prefs -logger=machine -hardware D:\Arduino\hardware -hardware D:\Arduino\portable\packages -tools D:\Arduino\tools-builder -tools D:\Arduino\hardware\tools\avr -tools D:\Arduino\portable\packages -built-in-libraries D:\Arduino\libraries -libraries D:\Arduino\portable\sketchbook\libraries -fqbn=esp32:esp32:featheresp32:FlashFreq=80,UploadSpeed=921600,DebugLevel=none,PartitionScheme=default -ide-version=10809 -build-path C:\Users\helmu\AppData\Local\Temp\arduino_build_141749 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=D:\Arduino\portable\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=D:\Arduino\portable\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.esptool_py.path=D:\Arduino\portable\packages\esp32\tools\esptool_py\3.3.0 -prefs=runtime.tools.esptool_py-3.3.0.path=D:\Arduino\portable\packages\esp32\tools\esptool_py\3.3.0 -prefs=runtime.tools.mkspiffs.path=D:\Arduino\portable\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=D:\Arduino\portable\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc.path=D:\Arduino\portable\packages\esp32\tools\xtensa-esp32s3-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=D:\Arduino\portable\packages\esp32\tools\xtensa-esp32s3-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.riscv32-esp-elf-gcc.path=D:\Arduino\portable\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.riscv32-esp-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=D:\Arduino\portable\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.mklittlefs.path=D:\Arduino\portable\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=D:\Arduino\portable\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc.path=D:\Arduino\portable\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=D:\Arduino\portable\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -verbose D:\Akten\Programmierung\ArduinoProgs\ESP32_GBox\ESP32_Box031i\ESP32_Box031i.ino
D:\Arduino\arduino-builder -compile -logger=machine -hardware D:\Arduino\hardware -hardware D:\Arduino\portable\packages -tools D:\Arduino\tools-builder -tools D:\Arduino\hardware\tools\avr -tools D:\Arduino\portable\packages -built-in-libraries D:\Arduino\libraries -libraries D:\Arduino\portable\sketchbook\libraries -fqbn=esp32:esp32:featheresp32:FlashFreq=80,UploadSpeed=921600,DebugLevel=none,PartitionScheme=default -ide-version=10809 -build-path C:\Users\helmu\AppData\Local\Temp\arduino_build_141749 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=D:\Arduino\portable\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=D:\Arduino\portable\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.esptool_py.path=D:\Arduino\portable\packages\esp32\tools\esptool_py\3.3.0 -prefs=runtime.tools.esptool_py-3.3.0.path=D:\Arduino\portable\packages\esp32\tools\esptool_py\3.3.0 -prefs=runtime.tools.mkspiffs.path=D:\Arduino\portable\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=D:\Arduino\portable\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc.path=D:\Arduino\portable\packages\esp32\tools\xtensa-esp32s3-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=D:\Arduino\portable\packages\esp32\tools\xtensa-esp32s3-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.riscv32-esp-elf-gcc.path=D:\Arduino\portable\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.riscv32-esp-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=D:\Arduino\portable\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.mklittlefs.path=D:\Arduino\portable\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=D:\Arduino\portable\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc.path=D:\Arduino\portable\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=D:\Arduino\portable\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -verbose D:\Akten\Programmierung\ArduinoProgs\ESP32_GBox\ESP32_Box031i\ESP32_Box031i.ino
Using board 'featheresp32' from platform in folder: D:\Arduino\portable\packages\esp32\hardware\esp32\2.0.4
Using core 'esp32' from platform in folder: D:\Arduino\portable\packages\esp32\hardware\esp32\2.0.4
[... cut ...]
Bibliothek I2Cdev im Ordner: D:\Arduino\portable\sketchbook\libraries\I2Cdev (legacy) wird verwendet
Bibliothek MPU6050 im Ordner: D:\Arduino\portable\sketchbook\libraries\MPU6050 (legacy) wird verwendet
Bibliothek WiFi in Version 2.0.0 im Ordner: D:\Arduino\portable\packages\esp32\hardware\esp32\2.0.4\libraries\WiFi  wird verwendet
Bibliothek WebServer in Version 2.0.0 im Ordner: D:\Arduino\portable\packages\esp32\hardware\esp32\2.0.4\libraries\WebServer  wird verwendet
exit status 1
Fehler beim Kompilieren für das Board Adafruit ESP32 Feather.

```

P-R-O-C-H-Y, 2022-12-12:
No issue on my side after adding `#include <vector>` + some code using std::vector.
Can you try to restart IDE / reinstall Arduino Core.

dsyleixa, 2022-12-12:
my code which does not compile is this one:
https://github.com/dsyleixa/Arduino/tree/master/ESP32_GBox/ESP32_Box031i

P-R-O-C-H-Y, 2022-12-12:
But in the output I don't see what is exact issue.
I can tell for sure that SD lib is working for sure in 2.0.5, was doing some tests last week.