espressif/arduino-esp32 #3604: Why ESP32 (BLE scan in arduino IDE) sometimes doesn't scan the BLE devices for about 12 to 25 minutes?

Gopal123456789, 2019-12-31:
I am working on ESP32 in Arduino IDE to scan for some BLE devices which advertise few data all the time. I am only scanning Eddystone data and printing on serial continuously in 5sec of time gap. The BLE scan part works fine but, sometimes I have seen in my serial output logs, it doesn't scan the data at all. This happens for 12 to 25 minutes as of now. What could be the problem? 

Below is the code I use. It's not the complete code which I use for my project, but the scan mechanism is the same as shown in this code. Here I am printing the temperature value in every scan. I also take battery voltage and mac ID of the BLE device as my data.

Thank you.

// Here is the code
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>
#include "BLEBeacon.h"
#include "BLEEddystoneTLM.h"
#include "BLEEddystoneURL.h"
#include <WiFi.h>

BLEScan *pBLEScan;
int scanTime = 5; //In seconds
uint16_t beconUUID = 0xFEAA;

class MyAdvertisedDeviceCallbacks : public BLEAdvertisedDeviceCallbacks
{
  void onResult(BLEAdvertisedDevice advertisedDevice)
  {
    std::string strServiceData = advertisedDevice.getServiceData();
    uint8_t cServiceData[100];
    strServiceData.copy((char *)cServiceData, strServiceData.length(), 0);

    if (advertisedDevice.getServiceDataUUID().equals(BLEUUID(beconUUID)) == true)
    { 
      if (cServiceData[0] == 0x20)
      {
        BLEEddystoneTLM oBeacon = BLEEddystoneTLM();
        oBeacon.setData(strServiceData);
        uint16_t temp = ENDIAN_CHANGE_U16_1((uint16_t)oBeacon.getTemp());
        Serial.println("tempurature is: %d", temp);
      }
    }
  }
};
setup_wifi()
{
  WiFi.begin(myssid, mypskd);
  while (WiFi.status() != WL_CONNECTED)
  {
    WiFi.begin(myssid, mypskd);
    delay(2000);
  }

  Serial.println("WiFi connected to Local IP address:");
  Serial.println(WiFi.localIP());
}
void setup()
{
  Serial.begin(115200);
  setup_wifi();
 
  //BLE Initializations
  BLEDevice::init("");
  pBLEScan = BLEDevice::getScan(); //create new scan
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
 
}

void loop()
{
  
    Serial.println("\nScanning...");
    BLEScanResults foundDevices = pBLEScan->start(scanTime);
    Serial.printf("\nScan done! Devices found: %d\n", foundDevices.getCount());
    delay(2000);
}

h2zero, 2020-01-05:
I had a similar problem, I believe it's caused by calling pBLEScan->start() before the previous scan started. Since your scanTime variable is 5 seconds but you call scan start every 2 seconds in your loop it creates issues in the library. 

I modified the library in my code to ignore the call until the scan is complete but for your purposes you could just make your loop delay 5500 instead of 2000 and it should solve it.

Gopal123456789, 2020-01-07:
Thanks h2zero for responding on it. 
The delay of 2 seconds was to give the scan part a little rest before it continues again. I tried running a timer for 5 seconds by replacing 2 seconds of delay. But the problem is still occurring. The "foundDevices.getCount()" also show 0 after each scan. After about 15 to 20 minuets it repeats, when it goes for scanning, it gets stuck and due to my WDT ESP32 restarts.
Thank you.

h2zero, 2020-01-20:
Sorry for the delay in response. There is a semaphore in the library that holds things up. You have a scan time of 5 seconds and you are calling it to start every 2 seconds in your posted code so you're asking it to start again before the first scan is completed and that causes issues with the semaphore in my experience. Adjusting it to 5 seconds before the call to start scanning isn't long enough either as the scan might not have yet completed, as a test make it 8 seconds and see what happens. 

I have modified the library for my purposes to not allow the call until the scan stopped message arrives. I'll see if I can put it up in my repo.

Gopal123456789, 2020-01-20:
So are you saying that, after the call back of scanning process is completed, and the print statement says "scan done!", still the process is running? And after about 8 seconds it finally stops?

h2zero, 2020-01-20:
My apologies, seems I misread your code at first. The call you’re using is the blocking call which should work fine, I thought you were manually stopping the scan, my bad. 

Anyway, does it get to the point that it prints “scan done” and just not start again or does it just hang when it starts scanning?

Gopal123456789, 2020-01-21:
No problem h2zero.

This problem of scanning occurs sometimes. The scan works fine and scanned devices are shown regularly. But suddenly it prints "Scan done! Devices found: 0" and nothing. Not even the print statement inside the callback. This continues till 12 to 25 minutes and the code stucks and due to WDT, it reboots and starts scanning normally again.

Thank you.

h2zero, 2020-01-21:
Ok, well that’s good info. My guess since I can’t see your full code is when you get to delay in loop the scheduler switches to another task somewhere that doesn’t have a delay call often enough for the idle task to reset the watchdog.

Can you post the error message when it triggers?

Gopal123456789, 2020-01-22:
I have not got any error for this issue yet. Because, while scanning it just sucks or hangs and after WDT is done (40 seconds), it just restarts.

h2zero, 2020-01-23:
Yes, that's the message I'm interested it. When the watchdog resets it will state what process didn't feed the watchdog and what was running on each core at the time. My bet is Core 0 running wifi. Unless you can provide more information all I can suggest is commenting out all the wifi stuff in your code and see if you still have issues.

Gopal123456789, 2020-01-23:
Ok. Let me check that.
Thanks for the info.

Gopal123456789, 2020-01-30:
Sorry for coming late. 
Below is the serial print I have got with the error message.

[2020-01-29 21:36:35.399] Scanning...
[2020-01-29 21:36:40.398] 
[2020-01-29 21:36:40.398] Scan done! Devices found: 0

[2020-01-29 21:36:43.430] Scanning...
[2020-01-29 21:36:48.415] 
[2020-01-29 21:36:48.415] Scan done! Devices found: 0

[2020-01-29 21:36:51.444] Scanning...
[2020-01-29 21:36:56.449] 
[2020-01-29 21:36:56.449] Scan done! Devices found: 0

[2020-01-29 21:36:59.479] Scanning...
[2020-01-29 21:37:04.479] 
[2020-01-29 21:37:04.479] Scan done! Devices found: 0

and so...... on till 18 minutes..
[2020-01-29 21:53:36.071] Scanning...
[2020-01-29 21:53:41.055] 
[2020-01-29 21:53:41.055] Scan done! Devices found: 0

[2020-01-29 21:53:44.090] Scanning...
[2020-01-29 21:53:49.087] 
[2020-01-29 21:53:49.087] Scan done! Devices found: 0

[2020-01-29 21:53:52.118] Scanning...
[2020-01-29 21:54:32.112] reboot
[2020-01-29 21:54:32.402] Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
[2020-01-29 21:54:32.417] Core 0 register dump:
[2020-01-29 21:54:32.417] PC      : 0x4008f725  PS      : 0x00060f34  A0      : 0x800908f1  A1      : 0x3ffd5e40  
[2020-01-29 21:54:32.433] A2      : 0x3ffbf5e8  A3      : 0x0000cdcd  A4      : 0xb33fffff  A5      : 0x00000001  
[2020-01-29 21:54:32.433] A6      : 0x00060d23  A7      : 0x0000abab  A8      : 0x0000abab  A9      : 0x3ffd5f50  
[2020-01-29 21:54:32.448] A10     : 0x3fffd308  A11     : 0x401053e4  A12     : 0x40081814  A13     : 0x4010f064  
[2020-01-29 21:54:32.448] A14     : 0x000000f5  A15     : 0x00000d10  SAR     : 0x00000018  EXCCAUSE: 0x00000005  
[2020-01-29 21:54:32.448] EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
[2020-01-29 21:54:32.464] 
[2020-01-29 21:54:32.464] Backtrace: 0x4008f725:0x3ffd5e40 0x400908ee:0x3ffd5e70 0x4008e433:0x3ffd5e90 0x4008e53d:0x3ffd5ed0 0x40081801:0x3ffd5ef0 0x40107cf7:0x3ffd5f10 0x40105402:0x3ffd5f30 0x4010262e:0x3ffd5f50 0x4009512b:0x3ffd5f70 0x4008e725:0x3ffd5fb0
[2020-01-29 21:54:32.480] 
[2020-01-29 21:54:32.480] Core 1 register dump:
[2020-01-29 21:54:32.480] PC      : 0x400910b2  PS      : 0x00060734  A0      : 0x80090293  A1      : 0x3ffbe830  
[2020-01-29 21:54:32.495] A2      : 0x3fffd3f0  A3      : 0x3ffbc6c0  A4      : 0x00000001  A5      : 0x00000001  
[2020-01-29 21:54:32.495] A6      : 0x00060723  A7      : 0x00000000  A8      : 0x3ffbc6c0  A9      : 0x3ffbc6c0  
[2020-01-29 21:54:32.511] A10     : 0x00000019  A11     : 0x00000019  A12     : 0x00000001  A13     : 0x00000001  
[2020-01-29 21:54:32.511] A14     : 0x00060721  A15     : 0x00000000  SAR     : 0x00000014  EXCCAUSE: 0x00000005  
[2020-01-29 21:54:32.527] EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
[2020-01-29 21:54:32.527] 
[2020-01-29 21:54:32.527] Backtrace: 0x400910b2:0x3ffbe830 0x40090290:0x3ffbe850 0x4008e4d3:0x3ffbe870 0x400e73a5:0x3ffbe8b0 0x4010662a:0x3ffbe8d0 0x400faef1:0x3ffbe8f0 0x400816d3:0x3ffbe920 0x40080fc4:0x3ffbe940 0x40081369:0x3ffbe960 0x40084881:0x3ffbe980 0x401f5c93:0x3ffbc5b0 0x400e825e:0x3ffbc5d0 0x4008ff09:0x3ffbc5f0 0x4008e725:0x3ffbc610
[2020-01-29 21:54:32.558] 
[2020-01-29 21:54:32.558] Rebooting...
[2020-01-29 21:54:32.558] ets Jun  8 2016 00:22:57

As you suspected, its a Core 0 issue.
I have not tested with WiFi disabled. I will test that also and let you know.

I made the scan type to passive scan by making 
" pBLEScan->setActiveScan(false); //active scan uses more power, but get results faster".
And I have not seen this problem yet. Can you tell me what is the difference between Active and passive scan?

Gopal123456789, 2020-01-30:
Also during BLE scan process, I am not using WiFi. I send my data using MQTT, after BLE scan is done and before the scan starts again.

h2zero, 2020-01-30:
Good, now you have a backtrace, can you decode it and post the results. If you don’t have the decoder already, it’s here.

https://github.com/me-no-dev/EspExceptionDecoder

Where in your code are you sending the MQTT message? I’m guessing it’s in the onResult() callback, if so don’t do that, probably the cause of your problem.

Gopal123456789, 2020-01-31:
My MQTT message is being sent after a complete scan of BLE. There is no disturbance to the BLE scan process caused by any other process. 
And the exception decoder result can be seen in the screenshot.
[image]

h2zero, 2020-02-01:
Seems to be getting stuck on taking the semaphore [here](https://github.com/espressif/arduino-esp32/blob/9b75c65fc76ea67b7b8420dbc2c8ccc917182765/libraries/BLE/src/BLEScan.cpp#L239)
or [here](https://github.com/espressif/arduino-esp32/blob/9b75c65fc76ea67b7b8420dbc2c8ccc917182765/libraries/BLE/src/BLEScan.cpp#L197). This happens if something doesn't give the semaphore back, which should happen in the event callback. You could try turning on verbose logging to help find out where. [](url)

closed 2020-04-16 by stale[bot]