espressif/arduino-esp32 #3605: Can't upload a sketch from the Arduino IDE running on Raspberry Pi4 to ESP32 (re-opened)

roblatour, 2019-12-31:
please see: [](https://github.com/arduino/Arduino/issues/9568)

As suggested at the post above, I tried 
sudo apt-get --purge remove modemmanager 
and a shutdown and restart but that did not help.

Also I tried adding my user name (pi) to the dailout group with the command:
sudo usermod -a -G dialout pi

and confirmed it using the groups command
and a shutdown and restart but that did not help.

The issue remains the same as described in the link above.

lbernstone, 2019-12-31:
Close all apps and unplug the esp32.  Plug it back in and run `dmesg`.  Look for errors near the bottom regarding the cp210x connection, or else you should see something along the lines of `usb 1-6: cp210x converter now attached to ttyUSB0`
If you are getting a tty device, then you have a general connection issue.  Search through the issues for "failed to connect".

roblatour, 2019-12-31:
that seems to look ok

[ 5658.308299] usbserial: USB Serial support registered for generic
[ 5658.313732] usbcore: registered new interface driver cp210x
[ 5658.314021] usbserial: USB Serial support registered for cp210x
[ 5658.314334] cp210x 1-1.1:1.0: cp210x converter detected
[ 5658.320991] usb 1-1.1: cp210x converter now attached to ttyUSB0

but the original problem remains.

roblatour, 2020-01-04:
I was able to upload a sketch using a third board.  will close the issue,  Thanks for your help.

closed 2020-01-04 by roblatour