espressif/arduino-esp32 #3612: ZE08-CH20 UART problem

Listian, 2020-01-04:
### Hardware:
Board:							ESP32 Lolin32
Core Installation version:			1.0.4
IDE name:						?Arduino IDE?
Flash Frequency:					?40Mhz?
Upload Speed:					?921600?
Computer OS:                                        ?Windows 10? 

### Description:

Hello!
I have a ZE08-CH20 sensor. I tried connecting to the ESP32-lolin32 via UART, but commands are not accepted. I connected the sensor via external and internal USB-TTL (reset on gnd and tx-tx, rx-rx) and the commands are accepted.

I guess the problem is in the software. Installed earlier versions, but the result too. 
What else can I do or what other information should I send to understand the problem in the software or not?

lbernstone, 2020-01-04:
Please explain more completely how you connected the device- what pins did you use, what code did you run.  Does it work if you connect the DAC pin to a ESP32 ADC?

Listian, 2020-01-04:
I forgot to add that nextion, pms7003 work fine on uart. problem with only with ZE08-CH2.

I connect as usual rx-tx and tx-rx. 
The sensor has 3v at rx / tx outputs. tx - green, rx blue.
DSO 150 shows data transfer.

DAC (pin 2) works. But the problem is that adc esp32 is not linear and the result is incorrect compared to the multimeter. But this is an ESP32 problem already.

https://imgur.com/MimhAoM
https://imgur.com/xJHrGlz
https://imgur.com/K30S9ry

lbernstone, 2020-01-04:
Use a different uart and set of pins for the device than rx/tx and Serial.  Unless you turn it off, debug logging will be on Serial.
`Serial1.begin(9600, SERIAL_8N1, 16, 17); //TX=16 RX=17

stickbreaker, 2020-01-04:
@Listian Also, you could be having a problem sharing tx/rx with the USB->Serial bridge.  Depending on how the circuit is designed between the bridge chip and the ESP32, the Bridge chip could be overpowering your sensor, as @lbernstone suggests, try using other pins that you know are not also connected to another TTL UART. 

Chuck.

Listian, 2020-01-05:
Hmm, through serial1 it worked.

I connected again nextion and pms7003 to serial and they work fine, but ZE08-CH20 does not. 

There is a feature, when loading the firmware, it does not see the ZE08-CH20 commands sent by rx / tx and easily loads it.

All this is strange.

closed 2020-03-19 by stale[bot]