espressif/arduino-esp32 #3614: Preferences getBytes function is malfunctioning

mntolia, 2020-01-05:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Ubuntu

### Description:
I am storing std::vector<int> array in the flash memory using the preferences putBytes() function. When I am adding a new std::vector<int>, the getBytes() function works fine. But when I restart the ESP32, the getbytes() function returns a std::vector<int> of a very large size although getBytesLength() returns the correct length of the std::vector<int>.

### Sketch:  
```
Preferences fmem;
//preference key store
std::vector<int> key_store;
void loop()
{
  //Get Current Time
  fmem.begin("timer");
  size_t key_size = fmem.getBytesLength("keys");
  fmem.getBytes("keys", &key_store, key_size);
  Serial.println(fmem.getBytesLength("keys"));
  Serial.println(key_store.size());
}
void add_to_keystore(epoch_time)
{
fmem.begin("timer");
  Serial.println("Keystore Length Before Adding New Key");
  Serial.println(fmem.getBytesLength("keys"));
  fmem.getBytes("keys", &key_store, fmem.getBytesLength("keys"));
  key_store.push_back(atoi(epoch_time));
  //Storing all the keys
  fmem.putBytes("keys", &key_store, key_store.size());
  Serial.println("Keystore Length After Adding Key");
  Serial.println(fmem.getBytesLength("keys"));
  fmem.end();
}
```

### Debug Messages:
```
1
4294967254
```
1 is the byte length of the key and is what is printing as the keystore size. I get a guru meditation error once I put the std::vector<int> into a loop.

stickbreaker, 2020-01-05:
@mntolia a std::vector is a dynamic object. Its' data storage is resized when needed. You can't just overwrite it to fill it.

Check out this discussion on StackOverload [std::vector internals](https://stackoverflow.com/questions/11893251/stdvector-internals)

Chuck.

lbernstone, 2020-01-05:
The guru error you are getting when doing this in a push loop (the correct method) are probably allocation errors.  You must reserve space in a dynamic array if you plan to fill it with a lot of data.

stickbreaker, 2020-01-05:
@mntolia If you want to use std::vector, you have to do more work:
```c++

Preferences fmem;
//preference key store
std::vector<int> key_store;
void loop()
{
  //Get Current Time
  fmem.begin("timer");
  size_t key_size = fmem.getBytesLength("keys"); // length in bytes, not ELEMENTS

  key_store.resize(key_size / sizeof(int),0); // allocate array for number of possible elements, default to zero value
  key_size = key_store.size() * sizeof(int); // limit it to fit in allocated, in case odd bytes exist     
    // memory pointer pointing to the 
    // first element 
  int* pos = key_store.data(); 
     
  fmem.getBytes("keys", pos, key_size); // fill data array with bytes from pref
  Serial.printf("pref Size = %d, actual bytes used %d, number of elements %d\n",fmem.getBytesLength("keys"),key_size,key_store.size());
  
}
```

Chuck.

mntolia, 2020-01-07:
@stickbreaker Thanks Chuck for going the extra mile and showing me how to use the vector. I assumed it got resized in putBytes!

closed 2020-01-07 by mntolia