espressif/arduino-esp32 #3615: BasicOTA and "ESP32 sketch data upload" tool

JPMJPM, 2020-01-06:
Hi,

Board: ESP32 Dev module
Core Installation: esp32 v1.0.2
IDE name: Arduino IDE v1.8.10
CPU Frequency: 240MHz
Flash Frequency: 80MHz
PSRAM enabled: Disabled
Upload Speed: 115200
Computer OS: Windows 10

I am testing BasicOTA and uploading the sketch works fine using
the port "hostname at 192.168.1.28 (ESP32 Dev Module)" but the
"ESP32 sketch data upload" tool does no work.

Regards.

lbernstone, 2020-01-06:
It works as designed, which is that it doesn't do that.  The sketch upload tool in Arduino IDE is for directly connected devices.  However, the [espota.py](https://github.com/esp8266/Arduino/blob/master/tools/espota.py) script can upload spiffs partition images, and the [update library](https://github.com/espressif/arduino-esp32/tree/master/libraries/Update) can be used to write flexible updaters to meet your needs (push/pull/http/etc).
There are lots of examples that google knows about. 
https://github.com/lbernstone/asyncUpdate/blob/master/AsyncUpdate.ino

JPMJPM, 2020-01-06:
I forgot to say that I use OTA password "ArduinoOTA.setPassword("password");".
With "ESP32 sketch data upload" there is not any dialog box pops up asking for
password that's way  the data upload is not possible.

Timmmy001, 2020-03-01:
Hello together.
I can confirm the problem:
Without a OTA-Password (Hash in my case) it works.

With a password I get this error message in Arduino IDE after password-windows

> Sending invitation to 192.168.188.36 
Traceback (most recent call last):
  File "espota.py", line 345, in <module>
  File "espota.py", line 340, in main
  File "espota.py", line 127, in serve
UnicodeDecodeError: 'ascii' codec can't decode byte 0xdc in position 38: ordinal not in range(128)
Failed to execute script espota

(Win 10 - German // ArduinoIDE 1.8.13 hourly)

Timmmy001, 2020-03-15:
Hello again.
I found the problem. It is a incompatible with special character like "ü" "ö" "ß" (German).
Normal characters like "abc 123 ?!$"  are ok.

Timmmy001, 2020-05-16:
I'am not really a python expert.
But it works - need a compile to .exe (#4005)

Timmmy001, 2020-07-12:
The espota.pl file also works without modifications. :-|
But the compiled exe differs from the behaviour.
I have compiled the file myself as .exe and it now works with special characters.

Can someone trigger the recompilation in GitHub?

closed 2020-09-26 by stale[bot]