espressif/arduino-esp32 #3616: what would be the correct capacitance on EN for serial upload to work?

ZinggJM, 2020-01-06:
I have just received several ESP32 WROOM and ESP32 WROVER boards that time out on program upload over USB. Hitting the Boot button during upload helps.
I see in esptool.py near line 400 that this issue is addressed, but don't really understand it.

I am quite sure that I am not the only one suffering from low cost Chinese ESP32 boards with this issue. So I dare to ask if there is a workaround, or what capacitance should be added to EN.

Thank you for any help!

lbernstone, 2020-01-06:
Espressif has reference designs at https://www.espressif.com/en/support/download/documents/development-board
This forum is for issues with the arduino code base.  General questions belong at esp32.com

ZinggJM, 2020-01-06:
@lbernstone,

Thank you for your answer. It doesn't help me much, but I will try adding a capacitor according to the reference design.

**But I still think this is the correct place to raise this issue**, as the fix for issue #136 doesn't seem to cope with recent ESP32 boards. I don't think so many boards have defective transistors in the RTS DTR reset/boot circuit. I believe you will hear more about ESP32 boards with this issue being thrown on the market actually.

Jean-Marc

ZinggJM, 2020-03-06:
Maybe the title of the issue is wrong. But I think it is still an issue, if I have to push the Boot button for successful program upload from the Arduino IDE.
One of the ESP32 board that shows the issue is **not a nonname**, it is a **ESP32 DEVKIT V1** from www.doit.am, available e.g. from: https://www.aliexpress.com/item/4000165144198.html
I assume the board has a newer SILABS CP2102 that doesn't work with the tricks in the phyton code for upload?
And this is not the only ESP32 board that has this issue.

karls0, 2020-03-18:
I recently bought two ESP32-boards from OLIMEX and had problems uploading. It was very strange, since it worked the first time, but never again ??? After googling around for quite some time, I found your solution.
DanKoloff from Olimex gave me some tips too, but it was a capacitor at EN that helped. From curiosity I tested different caps, diffent resistors and the two Olimex-boards plus a AZ-delivery ESP32-Dev which I had bought the week before.
You can find the table of my tests at
https://github.com/OLIMEX/ESP32-POE/issues/11
Regards, Karl

closed 2020-05-31 by stale[bot]