espressif/arduino-esp32 #3617: ESP32, frequent calls to ledcWriteTone function will lead to inaccurate output.

SuhaylZhao, 2020-01-07:
Hi,

I find that frequent calls to ledcWriteTone function will lead to inaccurate output. 

The alert() function can execute correctly in Arduino uno, but not in esp32.

Increasing the delay time can improve this situation, but it can't achieve the effect I want.

Hoping to get help.

Thanks.

This is the complete code.

`#define PIN_BUZZER 12
#define CHN       0  

void setup() {
  pinMode(PIN_BUZZER, OUTPUT);
  ledcSetup(CHN, 0, 10);
  ledcAttachPin(PIN_BUZZER, CHN);
}

void loop() {
  alert();
}

void alert() {
  float sinVal;       
  int toneVal;          
  for (int x = 0; x < 360; x += 10) {    
    sinVal = sin(x * (PI / 180));       
    toneVal = 2000 + sinVal * 500;     
    ledcWriteTone(CHN, toneVal); 
    delay(1);
  }
}`

stickbreaker, 2020-01-07:
@SuhaylZhao The PWM change only occurs at the end of a PWM period. So if your dwell(base Frequency) on a PMW value is longer than the period between PWM changes, you will skip update values.

Why did you specify the PWM frequency as `0`?
```c++
  ledcSetup(CHN, 0, 10);
```
You can report the actual achieved PWM freq:
```c++
double dd= ledcSetup(CHN, 0, 10);
Serial.printf("actual PWM freq =%f\n",dd);
```

Chuck.

SuhaylZhao, 2020-01-07:
@stickbreaker 
Hi，

Because the pin is connected to the buzzer, the initial value of 0Hz can turn off the buzzer.

I printed the return value of the ```ledcWriteTone()``` functions in ```alert ()```. They are correct, but the actual output is not correct.  

I'll try the return value of the ```ledcSetup()``` function.

Thanks.

stickbreaker, 2020-01-07:
@SuhaylZhao 

Which version of Arduino-ESP32 are you using?
* 1.0.4 installed with Boards Manager?
* Development branch directly from GitHub?

1.0.4 has a memory leak in `ledcWriteTone()`, this leak has been fixed in the development branch.

Can you describe the *"inaccurate output"* in more detail? What is the problem you are seeing?

`ledcWriteTone()` outputs a square wave signal at the specified frequency. What are you seeing?

Chuck.

SuhaylZhao, 2020-01-08:
@stickbreaker 

1. Yes, the version is 1.0.4, I've tried both the stable version and the development version. Their problems are the same.

2. Yes, I follow this page: https://github.com/express/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md

3. About "inaccurate output", It means that the buzzer sound is not consistent with the expected sound. This code should be able to make the buzzer sound like the alarm of a police car, but in fact, it only makes one note sound. I use an oscilloscope to measure the PWM frequency of the output pin. The result shows that the PWM frequency of the output is always 1. 95khz to 2.05khz. But according to the code, it should change from 1.5kHz to 2.5KHz, right? Maybe the accuracy of the oscilloscope I use is not high enough, but it should not have such a big error. The code is as shown below.
```
  for (int x = 0; x < 360; x += 1) {
    sinVal = sin(x * (PI / 180));
    toneVal = 2000 + sinVal * 500;
    ledcWriteTone(CHN, toneVal);
    delay(1);
  }
```
Then I changed the code, as shown below. This is very effective. Use the oscilloscope to measure the PWM frequency of the output pin again, and the result is consistent with the code, with the range of 1.5kHz to 2.5KHz. The buzzer sounds like a police car alarm. But! But! But, The sound is not clear enough. It seems that there is a little noise.
```
  for (int x = 0; x < 360; x += 10) {
    sinVal = sin(x * (PI / 180));
    toneVal = 2000 + sinVal * 500;
    ledcWriteTone(CHN, toneVal);
    delay(10);
  }
```
So I continue to modify the code, as shown below. The performance of the buzzer is exactly the same as the code. The noise is very clear and  no noise. But, because x adds 20 at a time, the rhythm of the sound is no longer continuous， not like the alarm of a police car
```
  for (int x = 0; x < 360; x += 20) {
    sinVal = sin(x * (PI / 180));
    toneVal = 2000 + sinVal * 500;
    ledcWriteTone(CHN, toneVal);
    delay(20);
  }
```

4. At last, I conclude that the function ```ledcWriteTone()``` cannot be called frequently, and there should be a time interval of at least 20ms for each call. So it should be a bug.

Thanks.

Suhayl.

stickbreaker, 2020-01-08:
@SuhaylZhao 
```c++
double sinVal, toneVal,rad;
 for (int x = 0; x < 360; x += 20) {
    // I am converting to double because I have been bitten by compiler optimization.
    rad = x;
    rad = (x/180.0)*PI;
    sinVal = sin(rad);
    toneVal = 2000.0 + sinVal * 500.0;
    Serial.printf("%4.4f\n",toneVal); // should range 2000 > 2500 > 2000 > 1500 >2000
    ledcWriteTone(CHN, toneVal);
    delayMicroseconds(750); // a delay() > 1 will cause a task switch if another task is ready.  delayMicroseconds() will not switch.
  }
```
Non of these changes should affect your freq range, your code should produce 1500 to 2500 hz outputs.

The Boards manager installation will only install the static 1.0.1 .. 1.0.4 version, to use the current Dev version you have to follow these instruction:
[dev installation instructions](https://github.com/espressif/arduino-esp32#installation-instructions) and select one of the platform options under the Development Repository header.

Chuck.

lbernstone, 2020-01-08:
@SuhaylZhao ,
Try Chuck's new code, but you are asking a lot from the hardware to make this change in 10us.  However, there is a way to get the output you want without so much processing, if you use the [i2s peripheral](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/i2s.html).  It is, in fact, designed for managing sound waves :smile:.  There is not a C++ wrapper, but the C code has a decent [example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/i2s).  The compiler may complain about missing identifiers in the struct- you can see them [here](https://github.com/espressif/esp-idf/blob/release/v3.2/components/driver/include/driver/i2s.h#L132-L144)

SuhaylZhao, 2020-01-08:
@stickbreaker @lbernstone 
Hi,

Thanks for your help.

I tried chuck's code and it still doesn't work. As for lberstone's proposal, it needs new hardware, which changes my plan.

I looked at the waveform of the oscilloscope again and found some clues. When the function ```ledcWriteTone()``` is called frequently, some strange waveforms are output.  These are three oscillograms. They were taken under the same code.

[image]
[image]
[image]

So I think when calling ```ledcWriteTone()``` function, it will not only change the frequency of PWM, but also the pulse width of PWM, at least the pulse width of the first cycle. This effect is fatal. When the interval time of ```ledcWriteTone()``` call is 1ms, the frequency of call is 1kHz, while the PWM frequency is 1.5khz-2.5khz. They are too close to each other, so that the frequency and pulse width of each cycle are affected. When the interval time of ```ledcWriteTone()``` call is adjusted to 20ms, PWM has enough time to produce complete square wave, so the effect is relatively small at this time.  

I don't think there is any good way to solve this problem until it is solved officially. I might try using timers to create a software tone.

Thanks again.

Suhayl.

stickbreaker, 2020-01-08:
@SuhaylZhao 
Try this code:
```c++

#define chan 0
#define pin 12

ledcSetup(chan,2000.0,10);
ledcAttachPin(pin,chan);

double rad,tone;
for( int b=0; b<1024; b++){
	for(int a=0;a<360;a+=20){
	   rad = a;
	   rad = (rad/180.0)*PI;
	   tone = sin( rad );
	   int duration = 512 + (tone * 128);
	   ledcWrite(chan,duration);
	   delayMicrosecond(750);
	}
}
```

Chuck.

SuhaylZhao, 2020-01-09:
@stickbreaker 
This code only changes the PWM pulse width, not the frequency. Observe the PWM pin with oscilloscope, and the waveform is consistent with the code. 

I have succeeded by using software PWM, the code is as follows. 
```
#define BUZZER_PIN     12

hw_timer_t * timer = NULL;

void IRAM_ATTR onTimer() {
  digitalWrite(BUZZER_PIN, !digitalRead(BUZZER_PIN));
}

void setup() {
  Serial.begin(115200);
  pinMode(BUZZER_PIN, OUTPUT);
  timer = timerBegin(0, 80, true);
  timerAttachInterrupt(timer, &onTimer, true);
  timerAlarmWrite(timer, 1000000, true);
  timerAlarmEnable(timer);
}
void loop() {
  alert();
}

void alert() {
  float sinVal;
  int toneVal;
  for (int x = 0; x < 360; x += 1) {
    sinVal = sin(x * (PI / 180));
    toneVal = 2000 + sinVal * 500;
    timerAlarmWrite(timer, 500000 / toneVal, true);
    delay(1);
  }
}
```

As for ```ledcWriteTone()```, I'm very sure it's a bug. So I don't want to pay attention to it until it's officially solved. It took me too much time.

Thanks for your help!

Suhayl.

closed 2020-03-23 by stale[bot]

solocatz, 2021-06-19:
I have same issue.
In my application, a requirement of changing tone frequency is called ever loop, the goal is smoothly shifting tone. 

Code works fine in classic Arduino with tune() function, but same logic cause a lot of noise with ESP32 chip. Sounds like a broken PWM wave on each call.

**loop()
{
 //some logic here to shift frequency every loop
  ledcWriteTone(Tone_Pin_Channel, frequency);
  ledcWrite(Tone_Pin_Channel, duty);
}**

davidanderle, 2023-01-25:
Same issue here unfortunately

VojtechBartoska, 2023-01-25:
@P-R-O-C-H-Y can you please take a quick look on this?