espressif/arduino-esp32 #3618: Help program space

laercionit, 2020-01-07:
Hello, first congratulations to those involved in the project, I have been following from the beginning.

I currently have a project using ESP8266, I am looking to port the code to ESP32.
One item I am seeking to understand is the allocation of program space.
I believe it is the SSL features and other options that can be disabled on ESP8266 but do not yet have this feature on ESP32.

Any guidance or ideas?

ESP8266 
Sketch uses 570884 bytes **(54%)** of program storage space. Maximum is 1044464 bytes.
Global variables use 40032 bytes (48%) of dynamic memory, leaving 41888 bytes for local variables. Maximum is 81920 bytes.

ESP32:
Sketch uses 1274882 bytes **(97%)** of program storage space. Maximum is 1310720 bytes.
Global variables use 50336 bytes (15%) of dynamic memory, leaving 277344 bytes for local variables. Maximum is 327680 bytes.

ashtam55, 2020-01-07:
If you are using PlatformIO than you can change the partition table. By default, the chip saves some space for default OTA firmware, which we don't use until required. You can simply add this into your ``platformio.ini`` file -  ``board_build.partitions = no_ota.csv``
👍

laercionit, 2020-01-07:
I am using Arduino IDE, I need OTA, but I found the program space very strange in esp32, I think something might be better scaling, because esp8266 is a lower hardware and supports the execution of the program.

lbernstone, 2020-01-07:
BLE is extremely large.  There are partition schemes in Arduino IDE preset for large apps.  Set your board to "esp32 dev module" and you will see them.

ashtam55, 2020-01-07:
You are using your default partitions table which looks similar to this 
``nvs, data, nvs, 0x9000, 0x5000,``
``otadata, data, ota, 0xe000, 0x2000,``
``app0, app, ota_0, 0x10000, 0x1C0000,``
``app1, app, ota_1, 0x1D0000,0x1C0000,``
``eeprom, data, 0x99, 0x390000,0x1000,``
``spiffs, data, spiffs, 0x391000,0x6F000,``

If you need to increase your size then you will need to delete one partition (app1 might be), to save some space. And you will be able to do OTA as you saved some space.
For Arduino, you can google it - how to change default partition table for ESP32,
I use PlatformIO, which I already discussed above.

laercionit, 2020-01-07:
Thanks for the directions, but I'm still getting acquainted with ESP32.
What is BLE?

I currently use on esp8266 2mb APP, 1MB OTA and 1MB SPIFFS.
Is there any formatting that would suit this type of space allocation, I didn't find any

RudyFiero, 2020-01-10:
BLE = Bluetooth Low Energy = Bluetooth Smart. Not Bluetooth Classic

closed 2020-03-24 by stale[bot]

dalekirkwood, 2021-04-19:
> If you are using PlatformIO than you can change the partition table. By default, the chip saves some space for default OTA firmware, which we don't use until required. You can simply add this into your `platformio.ini` file - `board_build.partitions = no_ota.csv`

I know it is one year later. But a big thanks for this!