espressif/arduino-esp32 #3621: GPIO12 Operation

PrateekGoyal18, 2020-01-08:
Hello everyone, 
I am connecting an SD card module to the HSPI pins ie, 12, 13, 14 & 15 of the esp32. If I remove the GPIO12 connection to esp32, then only I am able to upload my code. Can anyone tell me how to upload my code without disconnecting the GPIO12 wire.

lbernstone, 2020-01-08:
https://github.com/espressif/esp-idf/tree/release/v3.2/examples/storage/sd_card#note-about-gpio12

PrateekGoyal18, 2020-01-08:
So there is no need of any external resistor circuit for connecting SD card to ESP32 ?

chegewara, 2020-01-08:
GPIO12 is special pin on esp32, here is more info in addition to above link:
https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection

WebDust21, 2020-01-08:
@chegewara probably hit the nail on the head; if GPIO12 is held to GND at startup, most ESP32 modules will fail to boot, due to the "MTDI strapping" check setting the ESP32's external FLASH chip VDD to 1.8v, instead of the required 3.3v.

One solution is to burn several eFuse configuration bits to disable the "MTDI strapping" check, permanently enabling the FLASH voltage regulator, and setting it to 3.3vout.  (NOTE: eFuse is OTP--it can only be changed once.  If these fuse bits get set incorrectly, you will permanently brick your ESP32 module.  Use this suggestion at YOUR own risk.)

Defines go at the top of your file obviously; I put the lines following in "setup()"; it will check the eFuse bits, and burn them if necessary.  You only have to run it once on each ESP32 module; **this is a PERMANENT solution**.

`
// libraries required to work with eFuses
#include "soc/efuse_reg.h"
#include "esp_efuse.h"        // for programming eFuse.

  // INITIAL SETUP...BURN THE EFUSE IF NECESSARY FOR PROPER OPERATION.
  // Force the FLASH voltage regulator to 3.3v, disabling "MTDI" strapping check at startup
  if ((REG_READ(EFUSE_BLK0_RDATA4_REG) & EFUSE_RD_SDIO_TIEH) == 0) {esp_efuse_reset(); REG_WRITE(EFUSE_BLK0_WDATA4_REG, EFUSE_RD_SDIO_TIEH); esp_efuse_burn_new_values();}          //burning SDIO_TIEH  -> sets SDIO voltage regulator to pass-thru 3.3v from VDD
  if ((REG_READ(EFUSE_BLK0_RDATA4_REG) & EFUSE_RD_XPD_SDIO_REG) == 0) {esp_efuse_reset(); REG_WRITE(EFUSE_BLK0_WDATA4_REG, EFUSE_RD_XPD_SDIO_REG); esp_efuse_burn_new_values();}    //burning SDIO_REG   -> enables SDIO voltage regulator (otherwise user must hardwire power to SDIO)
  if ((REG_READ(EFUSE_BLK0_RDATA4_REG) & EFUSE_RD_SDIO_FORCE) == 0) {esp_efuse_reset(); REG_WRITE(EFUSE_BLK0_WDATA4_REG, EFUSE_RD_SDIO_FORCE); esp_efuse_burn_new_values();}        //burning SDIO_FORCE -> enables SDIO_REG and SDIO_TIEH
`

WebDust21, 2020-01-08:
Also in the link that @lbernstone provided, you can also change the FLASH voltage with the "espefuse.py" utility.  Same end result as the code above.

chegewara, 2020-01-08:
This is exactly the same info you have from 1st link:
```
GPIO12 is used as a bootstrapping pin to select output voltage of an internal regulator which powers the flash chip (VDD_SDIO). This pin has an internal pulldown so if left unconnected it will read low at reset (selecting default 3.3V operation). When adding a pullup to this pin for SD card operation, consider the following:

For boards which don't use the internal regulator (VDD_SDIO) to power the flash, GPIO12 can be pulled high.
For boards which use 1.8V flash chip, GPIO12 needs to be pulled high at reset. This is fully compatible with SD card operation.
On boards which use the internal regulator and a 3.3V flash chip, GPIO12 must be low at reset. This is incompatible with SD card operation.
In most cases, external pullup can be omitted and an internal pullup can be enabled using a gpio_pullup_en(GPIO_NUM_12); call. Most SD cards work fine when an internal pullup on GPIO12 line is enabled. Note that if ESP32 experiences a power-on reset while the SD card is sending data, high level on GPIO12 can be latched into the bootstrapping register, and ESP32 will enter a boot loop until external reset with correct GPIO12 level is applied.
Another option is to burn the flash voltage selection efuses. This will permanently select 3.3V output voltage for the internal regulator, and GPIO12 will not be used as a bootstrapping pin. Then it is safe to connect a pullup resistor to GPIO12. This option is suggested for production use.
The following command can be used to program flash voltage selection efuses to 3.3V:

    components/esptool_py/esptool/espefuse.py set_flash_voltage 3.3V
This command will burn the XPD_SDIO_TIEH, XPD_SDIO_FORCE, and XPD_SDIO_REG efuses. With all three burned to value 1, the internal VDD_SDIO flash voltage regulator is permanently enabled at 3.3V. See the technical reference manual for more details.

espefuse.py has a --do-not-confirm option if running from an automated flashing script.
```

closed 2020-03-22 by stale[bot]

tsctrl, 2022-04-29:
hi @lbernstone, what will be the impact on burning fuse as above?