espressif/arduino-esp32 #3625: Unable to fetch content in AWS s3 OTA update

Nakul93, 2020-01-09:
I have set up everything correctly. Here is my output:

Got application/macbinary payload.
Got 744384 bytes from server
contentLength : 744384, isValidContentType : 0
There was no content in the response

why is isValidContentType is 0, I am using HTTPS with 443 as port

DeltaVetal26, 2020-01-12:
Hello @Nakul93 ,
You can try to completely reset the chip with the ESP tool: https://github.com/espressif/esptool

1) Connect ESP and open a command prompt
2) While holding the BOOT button on ESP, enter esptool.py erase_flash
3) ESP must be completely reset

Perhaps the ESP tool will detect the ESP chip.

jeffapp, 2020-02-03:
Check client whether to set  set timeout  time  .

chegewara, 2020-02-04:
Why is payload type:
`application/macbinary`

closed 2020-04-18 by stale[bot]