espressif/arduino-esp32 #3626: i2c stop function

tomtobback, 2020-01-09:
Hi, i'm working on an esp32 low-power sensor node with deep sleep. For some wake ups i use a sensor over i2c (standard Wire library), and i notice that once the i2c has been initiated, the deep sleep takes 3mA extra; i assume because the i2c remains active. How can i switch off the Wire instance? I see in the [library](https://github.com/espressif/arduino-esp32/blob/master/libraries/Wire/src/Wire.h) that there is the function `~TwoWire();` but i am not sure how to call it in my sketch, please advise, thanks.

stickbreaker, 2020-01-09:
@tomtobback just like normal explicit destructor.
~~~c++
Wire.~TwoWire();
// then to restart
Wire = TwoWire(0);
Wire.begin(sdapin,sclpin,freq);
~~~
Chuck

tomtobback, 2020-01-09:
many thanks @stickbreaker that works indeed

closed 2020-01-09 by tomtobback