espressif/arduino-esp32 #3627: avr/io.h in esp32-AI CAM

ronitsrma16, 2020-01-09:
### Hardware:
Board:							ESP32 AI cam
Core Installation/update date:		-
IDE name:						Ardiuino 1.8.5
Flash Frequency:					-
PSRAM enabled:                                     no psram
Upload Speed:				       -
Computer OS:                                        Windows 10

### Description:

I'm trying to store an image in the SD card and sending to server simultaneously. The included libraries are:

#include "esp_camera.h"
//#include <WiFi.h>
#include "Arduino.h"
#include <ArduinoWebsockets.h>
// Time
#include "time.h"
#include "lwip/err.h"
#include "lwip/apps/sntp.h"

// MicroSD
#include "driver/sdmmc_host.h"
#include "driver/sdmmc_defs.h"
#include "sdmmc_cmd.h"
#include "esp_vfs_fat.h"

#include <Arduino_FreeRTOS.h>
#include "camera_pins.h"

I commented wifi.h because it's getting multiple file issue

#### Debug messgaes

In file included from C:\Users\CG-DTE\Documents\Arduino\libraries\Arduino_FreeRTOS\src/Arduino_FreeRTOS.h:104:0,

                 from D:\Codes\esp cam\cam_sd\camsd2\camsd2.ino:16:

C:\Users\CG-DTE\Documents\Arduino\libraries\Arduino_FreeRTOS\src/FreeRTOSVariant.h:12:20: fatal error: avr/io.h: No such file or directory

compilation terminated.

exit status 1
Error compiling for board AI Thinker ESP32-CAM.

How can I resolve this issue

atanisoft, 2020-01-09:
Arduino_FreeRTOS.h is not needed or supported on the esp32 as the functionality for FreeRTOS is already available by default.

ronitsrma16, 2020-01-10:
Yes, you're right.  #include <Arduino_FreeRTOS.h> which is not supported for esp32 board. So we need to include below the library

#include "freertos/FreeRTOS.h"

Adding this library will resolve issue.

closed 2020-01-10 by ronitsrma16