espressif/arduino-esp32 #3630: abort() was called at PC 0x40087ab9 on core 1

slushpuppy, 2020-01-10:
### Hardware:
Board:							ESP32 cam
Core Installation/update date:			
IDE name:							Ardu IDE
Flash Frequency:					?40Mhz?
PSRAM enabled:                                          Yes
Upload Speed:						?115200?
Computer OS:                                            Windows 10

### Description:
I am trying to switch between BLE and Wifi when I encounter this error trying to turn Wifi on using wifi ssd/pass set by the user in BLE characteristics. 

However when I shifted wifi.begin into setup() without turning BLE off, this error occurs:

E (5030) wifi: Expected to init 16 rx buffer, actual is 2

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
... setup wifi and ble services
}

void loop() {
	if (wifi_pass != "" && wifi_name != "") {
		const char * wifi_name_c = wifi_name.c_str();
		const char * wifi_pass_c = wifi_pass.c_str();
		stopBLServices();
		delay(500);
		  WiFi.begin(wifi_name_c, wifi_pass_c); <--- line 186
        }
}
```

### Debug Messages:
```
abort() was called at PC 0x40087ab9 on core 1

Backtrace: 0x400963a8:0x3ffd1e70 0x400965d9:0x3ffd1e90 0x40087ab9:0x3ffd1eb0 0x40087ae4:0x3ffd1ed0 0x40087c35:0x3ffd1f00 0x4011b242:0x3ffd1f20 0x4011e4b1:0x3ffd2230 0x40086bd6:0x3ffd2260 0x401ed2ae:0x3ffd22b0 0x401e83b3:0x3ffd22f0 0x400ece96:0x3ffd2310 0x400ecfea:0x3ffd2330 0x400dc78d:0x3ffd2350 0x400d3291:0x3ffd2370 0x400d3355:0x3ffd2470 0x400d3441:0x3ffd2490 0x400d2c95:0x3ffd25c0 0x400d9ce5:0x3ffd2650 0x40092af1:0x3ffd2670
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:155
0x400965d9
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:170
0x40087ab9
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c:81
0x40087ae4
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c:134
0x40087c35
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c:171
0x4011b242
/Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:860 (discriminator 2)
0x4011e4b1
/Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vprintf.c:39
0x40086bd6
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/log/log.c:215
0x401ed2ae
??:?
0x401e83b3
??:?
0x400ece96
??:?
0x400ecfea
??:?
0x400dc78d
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/wifi_init.c:62
0x400d3291
C:\eclipse\ard\arduinoPlugin\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi\src/WiFiGeneric.cpp:115
0x400d3355
C:\eclipse\ard\arduinoPlugin\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi\src/WiFiGeneric.cpp:533
0x400d3441
C:\eclipse\ard\arduinoPlugin\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi\src/WiFiSTA.cpp:118
0x400d2c95
C:\eclipsepref\RUST_CAM\Release/..\RUST_CAM.ino:186
0x400d9ce5
C:\eclipse\ard\arduinoPlugin\packages\esp32\hardware\esp32\1.0.4\cores\esp32/main.cpp:19
0x40092af1
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:143

```

Pallanez, 2020-01-17:
Try: 
`WiFi.disconnect();
WiFi.begin(wifi_name_c, wifi_pass_c);`

MoveLikePro, 2020-02-14:
> Try:
> `WiFi.disconnect(); WiFi.begin(wifi_name_c, wifi_pass_c);`

Still getting same issue

closed 2020-04-28 by stale[bot]