espressif/arduino-esp32 #3634: Guru Meditation Error: Core 1 panic'ed (Cache disabled but cached memory region accessed) ESP32 ARDUINO IDE VSC Platformio

petervflocke, 2020-01-11:
### Hardware:
Board:							ESP32 Dev Module node32
IDE name:						Platform.io on Visual Studio Code
Computer OS:                                         Ubuntu?

### Description:
Decoding Guru Meditation does not provide a place of the issue.

> Exception Cause: Not found
> 
> 0x400d1d90: loop() at /home/peter/Documents/PlatformIO/Projects/monitorv1/src/main.cpp:150 (discriminator 10)
> 0x40084774: _xt_coproc_exc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1091
> 0x40086a21: spi_flash_restore_cache at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/cache_utils.c:283
> 0x400d1d90: loop() at /home/peter/Documents/PlatformIO/Projects/monitorv1/src/main.cpp:150 (discriminator 10)
> 0x400d1d90: loop() at /home/peter/Documents/PlatformIO/Projects/monitorv1/src/main.cpp:150 (discriminator 10)
> 0x40084771: _xt_coproc_exc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1090
> 0x40086a1e: spi_flash_phys2cache at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_mmap.c:466
> 0x40082b8b: esp_reset_reason_get_hint at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/reset_reason.c:117
> 0x40088b7d: vQueueDelete at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1796 (discriminator 1)
> 

The system reads sensor data over simple 433Mhz ASK routine ( [RFReceiver](https://github.com/zeitgeist87/RFReceiver) or [RadioHead](https://github.com/PaulStoffregen/RadioHead) ) and updates a web page on ESPAsyncWebServer. There is also a routine to read the time from NTP server.

If I don't initialize 433Mhz radio objects, failure of the WiFi does not  influence the system. If WiFi will be available again the system recovers the WiFI connection and works as design.
With an active radio object and failure of the WiFi after few seconds system reboots with guru meditation: 

> Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
> Core 1 register dump:
> PC      : 0x400d1d90  PS      : 0x00060034  A0      : 0x40084774  A1      : 0x3ffbe7b0  
> A2      : 0x00000004  A3      : 0x3ffc192c  A4      : 0x00000000  A5      : 0x00000010  
> A6      : 0x00000000  A7      : 0x1300005c  A8      : 0x80080f80  A9      : 0x00000001  
> A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3ffc1c7c  A13     : 0x00000000  
> A14     : 0x3ffc1c78  A15     : 0xffffffff  SAR     : 0x00000018  EXCCAUSE: 0x00000007  
> EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
> Core 1 was running in ISR context:
> EPC1    : 0x40086a21  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400d1d90
> 
> Backtrace: 0x400d1d90:0x3ffbe7b0 0x40084771:0x3ffbe7d0 0x40086a1e:0x3ffba1f0 0x40082b8b:0x3ffba210 0x40088b7d:0x3ffba230

The reboots happens also even if the transmitter is switched off and actually there is no reason to call any radio method.

I read [here1](https://www.esp32.com/viewtopic.php?t=7684) and [here2](https://github.com/espressif/arduino-esp32/issues/855) that a part of the code within interrupt routine must always be in the ESP32's IRAM - is that correct?

I tried to add IRAM_ATTR to few function handling radio parts without any success. 

### Question1:
Can i somehow disable (taking into consideration a slowdown of the performance) the caching? Is there something similar in Arduino / Platformio config to disable SPI_MASTER_ISR_IN_IRAM in ESP32 config settings?

### Question2:
Or how can I identify a function, which requires to stay in IRAM (if it is the solution of my problem)?

### Sketch:  
```cpp

#include <Arduino.h>
#include "main.h"
#include "index.h"
#include "credentials.h"
#include <NTPClient.h>
#include <WiFiUdp.h>
#include <TimeLib.h>
#include <Timezone.h>
#include "WiFi.h"
#include <ESPmDNS.h>
#include "ESPAsyncWebServer.h"
#include <PinChangeInterruptHandler.h>
#include <RFReceiver.h>

void updateJson();
void blink();
[... 254 lines cut ...]

#if  DEBUG_ON
#define PRINT(s, v) { Serial.print(s); Serial.print(v); }    
#define PRINTX(s, v) { Serial.print(s); Serial.print(v, HEX); }  
#define PRINTS(s) Serial.print(s)   
#define PRINTLN Serial.println()
#else
#define PRINT(s, v)   
#define PRINTX(s, v)  
#define PRINTS(s)     
#define PRINTLN
#endif
```

### Sketch index.h:  
```cpp
const char index_html[] PROGMEM = R"rawliteral(
<!DOCTYPE html>
<html lang=pl>
<head>
....
</body>
</html>
)rawliteral";
```

petervflocke, 2020-01-12:
Ok, it can be closed.
Solution: switch to radiohead from adafruit: https://github.com/adafruit/RadioHead
and add ESP32 handling for RH_PLATFORM_ESP32:
`#define INTERRUPT_ATTR ICACHE_RAM_ATTR`
by changing in RH_ASK.cpp file 
### FROM:
```cpp
#if (RH_PLATFORM == RH_PLATFORM_ESP8266)
    // interrupt handler and related code must be in RAM on ESP8266,
    // according to issue #46.
    #define INTERRUPT_ATTR ICACHE_RAM_ATTR
#else
    #define INTERRUPT_ATTR
#endif
```
### TO:
```cpp
#if (RH_PLATFORM == RH_PLATFORM_ESP8266)
    // interrupt handler and related code must be in RAM on ESP8266,
    // according to issue #46.
    #define INTERRUPT_ATTR ICACHE_RAM_ATTR
#elif (RH_PLATFORM == RH_PLATFORM_ESP32)
    #define INTERRUPT_ATTR IRAM_ATTR 
#else
    #define INTERRUPT_ATTR
#endif
```

closed 2020-01-12 by petervflocke

minhaz005, 2020-10-14:
did you get any solution to this problem... i am suffering the same problem...
i have integrate lua on esp32 project for user scripting ... now when user event occur on mqtt or rf loop... few moments its getting restart...after that work ok then again restart... i used vtaskdelay() and yield() function ....now still the same problem

IQ3/mtbd_haa/cmd/DO1
IOSTAT EVENT=1191
IQ3/mtbd_haa/cmd/DO1
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x40080f92  PS      : 0x00060034  A0      : 0x800810a1  A1      : 0x3ffbe750  
A2      : 0x00000001  A3      : 0x0000000c  A4      : 0x000000ba  A5      : 0x3ffbe760  
A6      : 0x00000008  A7      : 0x00000001  A8      : 0x3f408338  A9      : 0x000000d0  
A10     : 0x116e29a7  A11     : 0x3ffc183c  A12     : 0x000000ad  A13     : 0x3ffc1cfc  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x0000001c  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x4008734f  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40080f92

Backtrace: 0x40080f92:0x3ffbe750 0x4008109e:0x3ffbe780 0x40081125:0x3ffbe7b0 0x40085099:0x3ffbe7d0 0x4008734c:0x3ffba190 0x400834b3:0x3ffba1b0 0x400894a5:0x3ffba1d0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
Contents of file 'if count==nil then count=0;end;count=count+1;delay(1000);print(count);print("\n->Sec Up Lua!");function rfevent() print(RFCODE);delay(1);end;function ioevent()  print("IOSTAT EVENT=");print(IOSTAT);delay(10); end;function mqevent()  print(MQMSG);print("\r\n");print(MQTOPIC);print("\r\n")delay(10);end;function timerevent()print("\nMin Timer Called\r\n"); delay(10);end;function alarmevent() print(ALARMMSG);delay(10);end; function thevent() print(HUMIDITY);print(TEMPERATURE);delay(10);end;
Configurarion Found!

atanisoft, 2020-10-14:
@minhaz005 Perhaps open your own issue and provide actual code. This error typically will only occur if you have an interrupt routine that runs while accessing flash (ie:spiffs)

muhaimin365, 2021-02-02:
did you get any solution ?  I'm also facing same problem.
I want to make mesh communication between the lora device by using ESP32 module. so I want to make routing table for each node.
the Arduino sketch that attempts to talk to all other nodes in the mesh. Each node sends its routing information to every other node. The process of sending data and receiving acknowledgements lets a node determine which nodes it can successfully communicate with directly. This is how each node builds up it's routing table.

`
#include <EEPROM.h>
#include <RHRouter.h>
#include <RHMesh.h>
#include <RH_RF95.h>
#define RH_HAVE_SERIAL
#define LED 9
#define N_NODES 3
#define INTERRUPT_ATTR IRAM_ATTR

uint8_t nodeId;
uint8_t routes[N_NODES]; // full routing table for mesh
int16_t rssi[N_NODES]; // signal strength info

// Singleton instance of the radio driver
RH_RF95 rf95;

// Class to manage message delivery and receipt, using the driver declared above
RHMesh *manager;

// message buffer
char buf[RH_MESH_MAX_MESSAGE_LEN];

int freeMem() {
  return ESP.getFreeHeap();
}

void setup() {
  randomSeed(analogRead(0));
  pinMode(LED, OUTPUT);
  Serial.begin(115200);
  while (!Serial) ; // Wait for serial port to be available

  nodeId = EEPROM.read(0);
  if (nodeId > 10) {
    Serial.print(F("EEPROM nodeId invalid: "));
    Serial.println(nodeId);
    nodeId = 1;
  }
  Serial.print(F("initializing node "));

  manager = new RHMesh(rf95, nodeId);

  if (!manager->init()) {
    Serial.println(F("init failed"));
  } else {
    Serial.println("done");
  }
  rf95.setTxPower(23, false);
  rf95.setFrequency(915.0);
  rf95.setCADTimeout(500);

  // Possible configurations:
  // Bw125Cr45Sf128 (the chip default)
  // Bw500Cr45Sf128
  // Bw31_25Cr48Sf512
  // Bw125Cr48Sf4096

  // long range configuration requires for on-air time
  boolean longRange = false;
  if (longRange) {
    RH_RF95::ModemConfig modem_config = {
      0x78, // Reg 0x1D: BW=125kHz, Coding=4/8, Header=explicit
      0xC4, // Reg 0x1E: Spread=4096chips/symbol, CRC=enable
      0x08  // Reg 0x26: LowDataRate=On, Agc=Off.  0x0C is LowDataRate=ON, ACG=ON
    };
    rf95.setModemRegisters(&modem_config);
    if (!rf95.setModemConfig(RH_RF95::Bw125Cr48Sf4096)) {
      Serial.println(F("set config failed"));
    }
  }

  Serial.println("RF95 ready");

  for(uint8_t n=1;n<=N_NODES;n++) {
    routes[n-1] = 0;
    rssi[n-1] = 0;
  }

  Serial.print(F("mem = "));
  Serial.println(freeMem());
}

const __FlashStringHelper* getErrorString(uint8_t error) {
  switch(error) {
    case 1: return F("invalid length");
    break;
    case 2: return F("no route");
    break;
    case 3: return F("timeout");
    break;
    case 4: return F("no reply");
    break;
    case 5: return F("unable to deliver");
    break;
  }
  return F("unknown");
}

void updateRoutingTable() {
  for(uint8_t n=1;n<=N_NODES;n++) {
    RHRouter::RoutingTableEntry *route = manager->getRouteTo(n);
    if (n == nodeId) {
      routes[n-1] = 255; // self
    } else {
      routes[n-1] = route->next_hop;
      if (routes[n-1] == 0) {
        // if we have no route to the node, reset the received signal strength
        rssi[n-1] = 0;
      }
    }
  }
}

// Create a JSON string with the routing info to each node
void getRouteInfoString(char *p, size_t len) {
  p[0] = '\0';
  strcat(p, "[");
  for(uint8_t n=1;n<=N_NODES;n++) {
    strcat(p, "{\"n\":");
    sprintf(p+strlen(p), "%d", routes[n-1]);
    strcat(p, ",");
    strcat(p, "\"r\":");
    sprintf(p+strlen(p), "%d", rssi[n-1]);
    strcat(p, "}");
    if (n<N_NODES) {
      strcat(p, ",");
    }
  }
  strcat(p, "]");
}

void printNodeInfo(uint8_t node, char *s) {
  Serial.print(F("node: "));
  Serial.print(F("{"));
  Serial.print(F("\""));
  Serial.print(node);
  Serial.print(F("\""));
  Serial.print(F(": "));
  Serial.print(s);
  Serial.println(F("}"));
}

void loop() {

  for(uint8_t n=1;n<=N_NODES;n++) {
    if (n == nodeId) continue; // self

    updateRoutingTable();
    getRouteInfoString(buf, RH_MESH_MAX_MESSAGE_LEN);

    Serial.print(F("->"));
    Serial.print(n);
    Serial.print(F(" :"));
    Serial.print(buf);

    // send an acknowledged message to the target node
    uint8_t error = manager->sendtoWait((uint8_t *)buf, strlen(buf), n);
    if (error != RH_ROUTER_ERROR_NONE) {
      Serial.println();
      Serial.print(F(" ! "));
      Serial.println(getErrorString(error));
    } else {
      Serial.println(F(" OK"));
      // we received an acknowledgement from the next hop for the node we tried to send to.
      RHRouter::RoutingTableEntry *route = manager->getRouteTo(n);
      if (route->next_hop != 0) {
        rssi[route->next_hop-1] = rf95.lastRssi();
      }
    }
    if (nodeId == 1) printNodeInfo(nodeId, buf); // debugging

    // listen for incoming messages. Wait a random amount of time before we transmit
    // again to the next node
    unsigned long nextTransmit = millis() + random(3000, 5000);
    while (nextTransmit > millis()) {
      int waitTime = nextTransmit - millis();
      uint8_t len = sizeof(buf);
      uint8_t from;
      if (manager->recvfromAckTimeout((uint8_t *)buf, &len, waitTime, &from)) {
        buf[len] = '\0'; // null terminate string
        Serial.print(from);
        Serial.print(F("->"));
        Serial.print(F(" :"));
        Serial.println(buf);
        if (nodeId == 1) printNodeInfo(from, buf); // debugging
        // we received data from node 'from', but it may have actually come from an intermediate node
        RHRouter::RoutingTableEntry *route = manager->getRouteTo(from);
        if (route->next_hop != 0) {
          rssi[route->next_hop-1] = rf95.lastRssi();
        }
      }
    }
  }

}`

but when i run the code, i got error as i mentioned in the picture below
[image]

. what is the solution for it?

petervflocke, 2021-02-02:
If you are asking me, then I would suggest to decode the guru mediation, to see where the issue comes from.
For me the solution is presented in previous [post](https://github.com/espressif/arduino-esp32/issues/3634#issuecomment-573423900).

For guru mediation please refer to:
1) https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html
2) https://www.esp32.com/viewtopic.php?t=13615

Best
Peter

muhaimin365, 2021-02-02:
> If you are asking me, then I would suggest to decode the guru mediation, to see where the issue comes from.
> For me the solution is presented in previous [...]

thank you so much sir for you your quick response

kanakamedalasumanth, 2021-08-29:
> @minhaz005 Perhaps open your own issue and provide actual code. This error typically will only occur if you have an interrupt routine that runs while accessing flash (ie:spiffs)

Exactly, I have an interrupt routine and on parallel when I try to write to SPIFF it's getting crashed. What might be the solution? For temporarily I am disabling timerAttachInterrupt while accessing the SPIFF and enabling back after doing the operation. Does anyone suggest something better?

aidanbusby123, 2022-01-04:
One thing I've found is that if you press on the CPU, it gives you that error.

Hamza123-eng, 2022-09-08:
if you are using the interrupt then you have to assign the **IRAM_ATTR**  to your ISR.