espressif/arduino-esp32 #3635: Not getting accurate high speed timer interrupt intervals

nealmartini, 2020-01-12:
### Hardware:
Board:							ESP32-WROOM- Dev Module
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                      no
Upload Speed:					512000
Computer OS:                                         Windows 10

### Description:
I am trying to get esp32 timer 0 to interrupt repeatedly at 10 usec intervals. The sketch below works but there is jitter in the interrupt time intervals of plus or minus 20 nsec. It looks like some other resource is higher priority in the esp32 causing errors in the interrupt intervals.

hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

void IRAM_ATTR onTimer(){
  portENTER_CRITICAL_ISR(&timerMux);
  digitalWrite(23,HIGH);
  digitalWrite(23, LOW);
  portEXIT_CRITICAL_ISR(&timerMux);
  }
void setup() {
  pinMode(23, OUTPUT);
  timer = timerBegin(0, 80, true);// counter#, 80 MHz clock prescaler, count up
  timerAttachInterrupt(timer, &onTimer, true);// true means edge triggered
  timerAlarmWrite(timer, 10, true);// true is auto reload
  timerAlarmEnable(timer);
}
void loop() {
  while(1);
}

Is there an easy way to turn off all other esp32 interrupt sources except for timer 0 to ensure that I get accurate 10 usec interrupt periods? Or am I doing something else wrong in the above sketch?

Thank You
Neal

stickbreaker, 2020-01-12:
@nealmartini The ESP32 is a multiprocessor using a Multitasking operating system (FreeRTOS). The operating system switches task base on priority.  The only way you are going to get microsecond stable interrupt latency is to code for bare metal and roll your own operating system. The base OS tick is 1000us so, your interrupt timer is generating 100 interrupt cycles per task slice.

If you just need a pulse train look at the RMT peripheral [documentation here](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/rmt.html).

Chuck.

nealmartini, 2020-01-12:
I am new to the esp32 world. I thought you had to turn on the FreeRTOS operating system to enable it. I am doing my own scheduling and control, so don't need the FreeRTOS. Can I disable it and still retain the ability to download code from an Arduino IDE?

stickbreaker, 2020-01-12:
Nope Arduino-ESP32 is built upon ESP32-IDF which is based on the FreeRTOS kernel. The current release of Arduino-ESP32 is built against IDF 3.2.  To maximize preformance, you should study IDF 3.2.  The Arduino port is build around the limitations of IDF and FreeRTOS. Most of IDF is public except for the Radio sections WiFi and Bluetooth.  The Radio coding is private due of Regulatory compliance issues. 

Chuck.

nealmartini, 2020-01-13:
That is disappointing. 
Thanks for your help on this.

closed 2020-01-13 by nealmartini

yuwenpeng, 2022-04-29:
You could create a while(1) task on the 2nd core, then in the task use function taskENTER_CRITICAL(&mutex) to block all FreeRT stuff and fully control the cpu. You may have to implement timer in the while(1) loop by yourself because all times and interrupt are blocked too.  It takes around 0.02 um  (5cycle/240M) to get CPU counter by XTHAL_GET_CCOUNT(), and I believe this is the highest resolution we can get..