espressif/arduino-esp32 #3636: Module bricked after using analogSetAttenuation(ADC_11db);

Duncro, 2020-01-12:
Tried resolving analogRead issues and tried using analogSetAttenuation(ADC_11db);
Since then the board doesn't show up on com ports, no device connection sound when connecting it to the pc (win10).
Connected another esp board and it worked fine, so i don't think it's on the PC side.
The board still gets power though, can read the correct voltages on the voltage pins.

Any ideas?
I've even tried GPIO 0 to GND but still didn't work.

DeltaVetal26, 2020-01-12:
Hello @Duncro ,
You can try to completely reset the chip with the ESP tool: https://github.com/espressif/esptool

1) Connect ESP and open a command prompt
2) While holding the BOOT button on ESP, enter esptool.py erase_flash
3) ESP must be completely reset

Perhaps the ESP tool will detect the ESP chip.

Duncro, 2020-01-12:
Thanks, but didn't work. Looks like the FTDI on the module died. Used another FTDI to communicate with it and erase the flash but it still didn't connect with it's own FTDI.
For interest sake this is a LILYGO TTGO T-Call ESP32 Wrover-b module and have had some strange issues. Bought two, the one's FTDI died, and the other has inconsistent pin readings and it's I2C sometimes works then not. I even bought a logic level shifter but that didn't work, had to change the default pins as 21/22 just didn't work on the one module for I2C with identical code and breadboard layout. This module might be trickier to flash but at least it's I2C works :P

Duncro, 2020-01-13:
Noticed some flux residue on the board, cleaned it up with IPA, works fine again :)

closed 2020-01-13 by Duncro