espressif/arduino-esp32 #3639: mqtt_client.h working example!

soheils2, 2020-01-13:
### Hardware:
Board:							ESP32 wroom
Core Installation/update date:		
IDE name:						Arduino IDE
Flash Frequency:					
PSRAM enabled:                                          
Upload Speed:					115200
Computer OS:                                          Windows 10

### Description:

I'm working on something that needs MQTT over websocket (my server limitation is mqtt just over WS )

**On arduino IDE specifically.**

I've looked everywhere for it, but either no one have ability to !ratain or (||) send over websocket packets. :D

I've somehow get to this include in arduino ---> its close src and low language (at least for me!)

`#include "mqtt_client.h"`

and does this on my code:
```

void mqtt_init() {
  esp_mqtt_client_config_t mqttConfig;
  mqttConfig.uri = "ws://test.mosquitto.org";
  mqttConfig.port = 8080;
  mqttConfig.transport = MQTT_TRANSPORT_OVER_WS;
  esp_mqtt_client_start(client);
}

```

and it compiled!
But i dont know anything about callbacks and handle functions 

can you help me stablish a mqtt connection over websocket with arduino ide?
tnx

soheils2, 2020-01-13:
(https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/mqtt/mqtt_client.h)

im talking about this

atanisoft, 2020-01-13:
The only examples for mqtt_client.h are for ESP-IDF, there has not been an arduino-esp32 wrapper created for it. You can find the example for websocket based mqtt here: https://github.com/espressif/esp-idf/tree/release/v3.2/examples/protocols/mqtt/ws.

soheils2, 2020-01-13:
> The only examples for mqtt_client.h are for ESP-IDF, there has not been an arduino-esp32 wrapper created for it. You can find the example for websocket based mqtt here: https://github.com/espressif/esp-idf/tree/release/v3.2/examples/protocols/mqtt/ws.

ok but from what i see the [include ](https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/mqtt/mqtt_client.h))files are available on arduino-esp32 sdk too
 
is there anything i can do for it?  

i need too many libreries in  arduino IDE so i cant migrate to Idf

atanisoft, 2020-01-13:
It might work fine, you would just need to try it.

soheils2, 2020-01-13:
> It might work fine, you would just need to try it.

ok after testing i figure there is no 
`esp_mqtt_client_register_event arduino`
in header file!!! and thats the problem

what this function do is attach event handler for mqtt 

but there is nothing in the header and i get

`esp_mqtt_client_register_event' was not declared in this scope`

this is my code anyway:

```

#include "mqtt_client.h"

static void mqtt_event_handler_cb(esp_mqtt_event_handle_t event)
{
  esp_mqtt_client_handle_t client = event->client;
  int msg_id;
  // your_context_t *context = event->context;
  switch (event->event_id) {
    case MQTT_EVENT_CONNECTED:
      ESP_LOGI(TAG, "MQTT_EVENT_CONNECTED");
      msg_id = esp_mqtt_client_subscribe(client, "/topic/qos0", 0);
      ESP_LOGI(TAG, "sent subscribe successful, msg_id=%d", msg_id);

      msg_id = esp_mqtt_client_subscribe(client, "/topic/qos1", 1);
      ESP_LOGI(TAG, "sent subscribe successful, msg_id=%d", msg_id);

      msg_id = esp_mqtt_client_unsubscribe(client, "/topic/qos1");
      ESP_LOGI(TAG, "sent unsubscribe successful, msg_id=%d", msg_id);
      break;
    case MQTT_EVENT_DISCONNECTED:
      ESP_LOGI(TAG, "MQTT_EVENT_DISCONNECTED");
      break;

    case MQTT_EVENT_SUBSCRIBED:
      ESP_LOGI(TAG, "MQTT_EVENT_SUBSCRIBED, msg_id=%d", event->msg_id);
      msg_id = esp_mqtt_client_publish(client, "/topic/qos0", "data", 0, 0, 0);
      ESP_LOGI(TAG, "sent publish successful, msg_id=%d", msg_id);
      break;
    case MQTT_EVENT_UNSUBSCRIBED:
      ESP_LOGI(TAG, "MQTT_EVENT_UNSUBSCRIBED, msg_id=%d", event->msg_id);
      break;
    case MQTT_EVENT_PUBLISHED:
      ESP_LOGI(TAG, "MQTT_EVENT_PUBLISHED, msg_id=%d", event->msg_id);
      break;
    case MQTT_EVENT_DATA:
      ESP_LOGI(TAG, "MQTT_EVENT_DATA");
      printf("TOPIC=%.*s\r\n", event->topic_len, event->topic);
      printf("DATA=%.*s\r\n", event->data_len, event->data);
      break;
    case MQTT_EVENT_ERROR:
      ESP_LOGI(TAG, "MQTT_EVENT_ERROR");
      break;
    default:
      ESP_LOGI(TAG, "Other event id:%d", event->event_id);
      break;
  }
  return ESP_OK;
}

static void mqtt_event_handler(void *handler_args, esp_event_base_t base, int32_t event_id, esp_mqtt_event_handle_t event_data) {
  ESP_LOGD(TAG, "Event dispatched from event loop base=%s, event_id=%d", base, event_id);
  mqtt_event_handler_cb(event_data);
}

static void mqtt_app_start(void)
{
  esp_mqtt_client_config_t mqttConfig;
  mqttConfig.uri = "ws://test.mosquitto.org";
  mqttConfig.port = 8080;
  mqttConfig.transport = MQTT_TRANSPORT_OVER_WS;

  esp_mqtt_client_handle_t client = esp_mqtt_client_init(&mqttConfig);
  esp_mqtt_client_register_event(client, ESP_EVENT_ANY_ID, mqtt_event_handler, client);
  esp_mqtt_client_start(client);
}

void main (){
mqtt_app_start();
}
```

atanisoft, 2020-01-13:
Very likely it is due to C++ linking, try:
```
extern "C" static void mqtt_event_handler(void *handler_args, esp_event_base_t base, int32_t event_id, esp_mqtt_event_handle_t event_data) {
...
```

soheils2, 2020-01-13:
nope ! nothing to to with that just " esp_mqtt_client_**register**_event " is not implimented! others work correct

atanisoft, 2020-01-13:
looks like the header in this repo is different than the one in the ESP-IDF tree, you should set the callback in the config structure instead: https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/mqtt/mqtt_client.h#L88

soheils2, 2020-01-14:
> looks like the header in this repo is different than the one in the ESP-IDF tree, you should set the callback in the config structure instead: https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/mqtt/mqtt_client.h#L88

Ok but i cant figure it out

How to atrach handler on arduino 

I dont really familiar with callback functions! 

And believe me I've done research every_fuc.king_where

Nestornp, 2020-01-21:
`static int mqtt_event_handler(esp_mqtt_event_handle_t event)
{
  esp_mqtt_client_handle_t client = event->client;
  int msg_id;
  // your_context_t *context = event->context;
  switch (event->event_id) {
    case MQTT_EVENT_CONNECTED:
      ESP_LOGI(TAG, "MQTT_EVENT_CONNECTED");
      msg_id = esp_mqtt_client_subscribe(client, "/topic/qos0", 0);
      ESP_LOGI(TAG, "sent subscribe successful, msg_id=%d", msg_id);
      msg_id = esp_mqtt_client_subscribe(client, "/topic/qos1", 1);
      ESP_LOGI(TAG, "sent subscribe successful, msg_id=%d", msg_id);
      msg_id = esp_mqtt_client_unsubscribe(client, "/topic/qos1");
      ESP_LOGI(TAG, "sent unsubscribe successful, msg_id=%d", msg_id);
      break;
    case MQTT_EVENT_DISCONNECTED:
      ESP_LOGI(TAG, "MQTT_EVENT_DISCONNECTED");
      break;
    case MQTT_EVENT_SUBSCRIBED:
      ESP_LOGI(TAG, "MQTT_EVENT_SUBSCRIBED, msg_id=%d", event->msg_id);
      msg_id = esp_mqtt_client_publish(client, "/topic/qos0", "data", 0, 0, 0);
      ESP_LOGI(TAG, "sent publish successful, msg_id=%d", msg_id);
      break;
    case MQTT_EVENT_UNSUBSCRIBED:
      ESP_LOGI(TAG, "MQTT_EVENT_UNSUBSCRIBED, msg_id=%d", event->msg_id);
      break;
    case MQTT_EVENT_PUBLISHED:
      ESP_LOGI(TAG, "MQTT_EVENT_PUBLISHED, msg_id=%d", event->msg_id);
      break;
    case MQTT_EVENT_DATA:
      ESP_LOGI(TAG, "MQTT_EVENT_DATA");
      printf("TOPIC=%.*s\r\n", event->topic_len, event->topic);
      printf("DATA=%.*s\r\n", event->data_len, event->data);
      break;
    case MQTT_EVENT_ERROR:
      ESP_LOGI(TAG, "MQTT_EVENT_ERROR");
      break;
    default:
      ESP_LOGI(TAG, "Other event id:%d", event->event_id);
      break;
  }
  return msg_id;
}

void mqtt_init() {
  esp_mqtt_client_config_t mqttConfig;
  mqttConfig.uri = "ws://test.mosquitto.org";
  mqttConfig.port = 8080;
  mqttConfig.transport = MQTT_TRANSPORT_OVER_WS;
  mqttConfig.event_handle = mqtt_event_handler;
  esp_mqtt_client_handle_t client = esp_mqtt_client_init(&mqttConfig);
  esp_mqtt_client_start(client);
}
`

Nestornp, 2020-01-21:
`static int mqtt_event_handler(esp_mqtt_event_handle_t event)
{
  esp_mqtt_client_handle_t client = event->client;
  int msg_id;
  // your_context_t *context = event->context;
  switch (event->event_id) {
    case MQTT_EVENT_CONNECTED:
      ESP_LOGI(TAG, "MQTT_EVENT_CONNECTED");
      msg_id = esp_mqtt_client_subscribe(client, "/topic/qos0", 0);
      ESP_LOGI(TAG, "sent subscribe successful, msg_id=%d", msg_id);
      msg_id = esp_mqtt_client_subscribe(client, "/topic/qos1", 1);
      ESP_LOGI(TAG, "sent subscribe successful, msg_id=%d", msg_id);
      msg_id = esp_mqtt_client_unsubscribe(client, "/topic/qos1");
      ESP_LOGI(TAG, "sent unsubscribe successful, msg_id=%d", msg_id);
      break;
    case MQTT_EVENT_DISCONNECTED:
      ESP_LOGI(TAG, "MQTT_EVENT_DISCONNECTED");
      break;
    case MQTT_EVENT_SUBSCRIBED:
      ESP_LOGI(TAG, "MQTT_EVENT_SUBSCRIBED, msg_id=%d", event->msg_id);
      msg_id = esp_mqtt_client_publish(client, "/topic/qos0", "data", 0, 0, 0);
      ESP_LOGI(TAG, "sent publish successful, msg_id=%d", msg_id);
      break;
    case MQTT_EVENT_UNSUBSCRIBED:
      ESP_LOGI(TAG, "MQTT_EVENT_UNSUBSCRIBED, msg_id=%d", event->msg_id);
      break;
    case MQTT_EVENT_PUBLISHED:
      ESP_LOGI(TAG, "MQTT_EVENT_PUBLISHED, msg_id=%d", event->msg_id);
      break;
    case MQTT_EVENT_DATA:
      ESP_LOGI(TAG, "MQTT_EVENT_DATA");
      printf("TOPIC=%.*s\r\n", event->topic_len, event->topic);
      printf("DATA=%.*s\r\n", event->data_len, event->data);
      break;
    case MQTT_EVENT_ERROR:
      ESP_LOGI(TAG, "MQTT_EVENT_ERROR");
      break;
    default:
      ESP_LOGI(TAG, "Other event id:%d", event->event_id);
      break;
  }
  return msg_id;
}

void mqtt_init() {
  esp_mqtt_client_config_t mqttConfig;
  mqttConfig.uri = "ws://test.mosquitto.org";
  mqttConfig.port = 8080;
  mqttConfig.transport = MQTT_TRANSPORT_OVER_WS;
  mqttConfig.event_handle = mqtt_event_handler;
  esp_mqtt_client_handle_t client = esp_mqtt_client_init(&mqttConfig);
  esp_mqtt_client_start(client);
}`

closed 2020-04-04 by stale[bot]

stif, 2021-03-23:
@soheils2: Did you succeed in using the esp-mqtt within the arduino framework?

i tried it as well but got the same problem: `error: 'esp_mqtt_client_register_event' was not declared in this scope`
i found out this is because the declaration of 'esp_mqtt_client_register_event' was introduced in [arduino-esp32 1.0.5](https://github.com/espressif/arduino-esp32/blob/1.0.5/tools/sdk/include/mqtt/mqtt_client.h) and where missing in 1.0.4 - nonetheless i was not able to compile without errors.

i found this [thread](https://community.platformio.org/t/using-esp-idf-library-within-the-arduino-framework-esp32/17226) where they reference some [examples](https://github.com/platformio/platform-espressif32/blob/develop/examples/espidf-arduino-wifiscan/platformio.ini) where they used a mixed build environment and a special branch of the arduino-esp32 (defined in platformio.ini):

```
[env]
platform = espressif32
framework = arduino, espidf
platform_packages =
  ; use a special branch
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#idf-release/v4.0
```
then i tried to compile more or less the matching [mqtt example of the esp-idf@4.0](https://github.com/espressif/esp-idf/blob/release/v4.0/examples/protocols/mqtt/tcp/main/app_main.c) but still got compile errors:
`error: invalid conversion from 'void*' to 'esp_mqtt_event_handle_t' {aka 'esp_mqtt_event_t*'} [-fpermissive]`

according to the [documentation](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md) it should be possible and most likely it is some sort of version mismatch, but my patience and my skills are ending here :(

If someone has a working example how to use esp-mqtt with the arduino framework, i would be happy to know

in the meantime i will start to explore https://github.com/256dpi/arduino-mqtt

r-sherwood, 2022-10-04:
you can find a working arduino sample here:
https://gist.github.com/gmag11/c565a18d361a46993039f8a515e67614