espressif/arduino-esp32 #3643: i2s Problem at V1.0.4

tmsd2001, 2020-01-17:
### Hardware:
Board:							node32s
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Computer OS:                                         Windows 10

### Description:
After a long time, my sketch is running I want to update my sketch.
After uploading the Microphone sketch, the output is only 0 or -1.
At the end of troubleshooting, I switch back to the version 1.0.0 and the sketch works as it should. 

### Sketch
```cpp

/** 
 * ESP32 I2S Serial Plotter Example.
 * 
 * This example is based on the espressif-idf example and Public Domain.
 * @author maspetsberger 
 */

#include <driver/i2s.h>

const i2s_port_t I2S_PORT = I2S_NUM_0;

void setup() {
  Serial.begin(115200);
  Serial.println("Configuring I2S...");
  esp_err_t err;

  // The I2S config as per the example
  const i2s_config_t i2s_config = {
      .mode = i2s_mode_t(I2S_MODE_MASTER | I2S_MODE_RX), // Receive, not transfer
      .sample_rate = 16000,                         // 16KHz
      .bits_per_sample = I2S_BITS_PER_SAMPLE_32BIT, // could only get it to work with 32bits
      .channel_format = I2S_CHANNEL_FMT_ONLY_RIGHT, // although the SEL config should be left, it seems to transmit on right
      .communication_format = i2s_comm_format_t(I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB),
      .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1,     // Interrupt level 1
      .dma_buf_count = 4,                           // number of buffers
      .dma_buf_len = 8                              // 8 samples per buffer (minimum)
  };

  // The pin config as per the setup
  const i2s_pin_config_t pin_config = {
      .bck_io_num = 14,   // BCKL
      .ws_io_num = 15,    // LRCL
      .data_out_num = -1, // not used (only for speakers)
      .data_in_num = 32   // DOUT
  };

  // Configuring the I2S driver and pins.
  // This function must be called before any I2S driver read/write operations.
  err = i2s_driver_install(I2S_PORT, &i2s_config, 0, NULL);
  if (err != ESP_OK) {
    Serial.printf("Failed installing driver: %d\n", err);
    while (true);
  }
  err = i2s_set_pin(I2S_PORT, &pin_config);
  if (err != ESP_OK) {
    Serial.printf("Failed setting pin: %d\n", err);
    while (true);
  }
  Serial.println("I2S driver installed.");
}

void loop() {

  // Read a single sample and log it for the Serial Plotter.
  int32_t sample = 0;
  int bytes_read = i2s_pop_sample(I2S_PORT, (char *)&sample, portMAX_DELAY); // no timeout
  if (bytes_read > 0) {
    Serial.println(sample);
  }
  
}

// actually we would need to call `i2s_driver_uninstall(I2S_PORT)` upon exit. 
```

lbernstone, 2020-01-17:
Have you tried switching it back to I2S_CHANNEL_FMT_ONLY_LEFT?  I seem to recall someone mentioning that it had switched.
If not, you will probably need to take the issue to [IDF](https://github.com/espressif/esp-idf/issues) for resolution.

tmsd2001, 2020-01-18:
Switch back to I2S_CHANNEL_FMT_ONLY_LEFT fix the problem.

closed 2020-01-18 by tmsd2001