espressif/arduino-esp32 #3645: Filters.h package is not compatible for ESP32. Is there any alternative?

dayanand-123, 2020-01-18:
The below code is working fine for Arduino uno, but returns error for ESP32-Wroover/ESP32.

#include <Filters.h>

float testFrequency = 60;                     // test signal frequency (Hz)
float windowLength = 20.0/testFrequency;     // how long to average the signal, for statistist
int sensorValue = 0;
float intercept = -0.1129; // to be adjusted based on calibration testing
float slope = 0.0405; // to be adjusted based on calibration testing
float current_amps; // estimated actual current in amps

unsigned long printPeriod = 1000; // in milliseconds
// Track time in milliseconds since last reading 
unsigned long previousMillis = 0;

void setup() {
  Serial.begin( 57600 );    // start the serial port
}

void loop() {
RunningStatistics inputStats;                 // create statistics to look at the raw test signal
  inputStats.setWindowSecs( windowLength );
   
  while( true ) {   
    sensorValue = analogRead(A0);  // read the analog in value:
    inputStats.input(sensorValue);  // log to Stats function
        
    if((unsigned long)(millis() - previousMillis) >= printPeriod) {
      previousMillis = millis();   // update time
      
      // display current values to the screen
      Serial.print( "\n" );
      // output sigma or variation values associated with the inputValue itsel
      Serial.print( "\tsigma: " ); Serial.print( inputStats.sigma() );
      // convert signal sigma value to current in amps
      current_amps = intercept + slope * inputStats.sigma();
      Serial.print( "\tamps: " ); Serial.print( current_amps );
      Serial.print( "\tSensor Value: " ); Serial.print( sensorValue );
    }
  }
}

THE ERROR

C:\Users\Murali\Documents\Arduino\libraries\Filters-master\FilterTwoPole.cpp: In member function 'void FilterTwoPole::test()':

C:\Users\Murali\Documents\Arduino\libraries\Filters-master\FilterTwoPole.cpp:198:33: error: 'analogWrite' was not declared in this scope

     analogWrite(10,osc.output() ); // hardcoded the dial pin

                                 ^

Multiple libraries were found for "Filters.h"
 Used: C:\Users\Murali\Documents\Arduino\libraries\Filters-master
exit status 1
Error compiling for board ESP32 Wrover Module.

me-no-dev, 2020-01-20:
AnalogWrite is not "yet" implemented the same way as on AVR. Please see the examples on how to get it going.

closed 2020-01-20 by me-no-dev

shah5109, 2020-11-28:
U need to modified filters.h library.. by open the filterstwopole.cpp file using notepad.. and delete analogWrite code.