espressif/arduino-esp32 #3646: Feature Request: Port CertStoreBearSSL to ESP32?

yknivag, 2020-01-19:
Is it possible to handle https connections on the ESP32 **without** knowing in advance the root CA for the service one is trying to connect to?

For example, on the ESP8266 this is possible using `CertStoreBearSSL` (see https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WiFi).

This is particularly important for https services which redirect from one domain to another where one would need to know all the certificates in advance on the ESP32 whereas on the ESP8266 one needs only to add `certs.ar` to SPIFFS and it is possible to connect to almost any https service.

I may have missed a way that this is already possible but I haven't been able to find one. How complex would it be to port the `CertStoreBearSSL` functionality to the ESP32?

lbernstone, 2020-01-19:
Porting BearSSL would be a very significant task, particularly to use the hardware acceleration.  However, if you attempt to create an https connection in WiFiClientSecure with no CA/cert information, it will make permissive requests, and should accept any certificate offered.  Perhaps posting your actual issue (with code and logs) instead of your proposed fix can resolve the root problem.

yknivag, 2020-01-19:
> make permissive requests, and should accept any certificate offered.
That really isn't a solution though is it?  That's just leaving a key under the doormat because you keep losing your keys - you may as well just leave the door open.

What I want to be able to do is connect to any arbitrary https web service from within a sketch without having to load an individual root CA for each domain and keep track of which is which.

I can do that on the ESP8266 using `CertStoreBearSSL` as the `certs.ar` file allows any certificate to be authenticated against it's root CA without knowing which domains are required at the time of writing the code.

I have no code to post as I have, as yet, found not a single example nor any documentation that proposes any method of securely hitting an arbitrary https web service.  I have code that works on the ESP8266 (and there are many official examples of it for the ESP8266) but, whilst I can happily follow the examples and documentation to use the `CertStoreBearSSL` library I will readily admit that I lack the knowledge to recreate it's functionality.

Essentially I would like to recreate (this ESP8266 sketch)[https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/BearSSL_CertStore/BearSSL_CertStore.ino] on the ESP32 and have it function with the same level of security.

When I replace `#include <ESP8266WiFi.h>` with `#include <WiFi.h>` I get the following error in the Arduino IDE:

_Arduino: 1.8.10 (Windows 7), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

Test:26:30: error: CertStoreBearSSL.h: No such file or directory_

lbernstone, 2020-01-19:
Ok, so you want a certificate store, not BearSSL.  I'd suggest you change the title if you want anybody to look at it- a cert store is significantly easier than an encryption library.  The feature set of esp32 is completely different than esp8266, so won't be an exact analogue, but it should be possible to do it with less resources.

yknivag, 2020-01-19:
Thank you for your guidance, I have changed the title as you suggested.

> won't be an exact analogue

It would make sense for it to be implemented with the same "API" as the `CertStoreBearSSL` for the ESP8266 so that the interaction is the same.  In the same way that the WiFi client offers a similar interface for ESP8266, ESP32, Arduino and others.  That way it would be possible to pass a reference to a certStore object into a function and have it work the same way no matter the underlying hardware in the same way as one may now pass a reference to a WiFi client.

pranavb25, 2020-03-07:
> Porting BearSSL would be a very significant task, particularly to use the hardware acceleration. However, if you attempt to create an https connection in WiFiClientSecure **with no CA/cert [...]

@lbernstone I just observed this behaviour. Could you please shed some light into why this is allowed and why this happens? If no certificate/a corrupted certificate is provided, shouldn't the library throw an error? It currently says "Certificate verified" which is incorrect. 

I am not an expert on SSL, just wanting to understand the implications of this behaviour and the need of even providing a cert in light of this (and associated security concerns). I personally noticed this while trying to corrupt my ca cert and noticing what happens.

yknivag, 2020-05-11:
Would be better if this could be flagged as a feature request rather than a stale issue.

kinafu, 2020-06-22:
Did you already take a look at [https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_crt_bundle.html](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_crt_bundle.html)?

yknivag, 2020-06-22:
@kinafu - that's a promising start.  Unfortunately only available in the ESP-IDF though.  If that functionality (particularly the ability to store the cert bundle in SPIFFS/LittleFS) were to be ported here to Arduino-ESP32 then that would, I think, close this feature request.

meltdown03, 2020-06-23:
> @kinafu - that's a promising start. Unfortunately only available in the ESP-IDF though. If that functionality (particularly the ability to store the cert bundle in SPIFFS/LittleFS) were to be ported here to Arduino-ESP32 then that would, I think, close this feature request.

With a few modifications to the `ssl_client.cpp/h` files and adding the `esp_crt_bundle.c/h` component, I was able to use the `esp_crt_bundle_attach()` function to use specific certificates for the [Google IOT MQTT Smart Outlet Example](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino).

I had to manually create the `x509_crt_bundle` binary with the the `esp_crt_bundle/gen_crt_bundle.py` script and embedded into my sketch. I use platformio, so I'm going to make an extra_script to automate that process. In IDF, the CMakeLists.txt file does this for you.

atanisoft, 2020-06-23:
> I had to manually create the `x509_crt_bundle` binary with the the `esp_crt_bundle/gen_crt_bundle.py` script and embedded into my sketch. I use platformio, so I'm going to make an extra_script to automate that process. In IDF, the CMakeLists.txt file does this for you.

PlatformIO offers similar functionality for embedding files in the binary: http://docs.platformio.org/en/latest/platforms/espressif32.html#embedding-binary-data

meltdown03, 2020-06-23:
Yeah, that's what I used to embed the binary that the script created.

kinafu, 2020-06-23:
That sounds promising!

Would you mind sharing your modified `ssl_client` implementation?
I guess the solution would be 

- to add a method which does

```
esp_crt_bundle_set(myx509_bundle)
esp_crt_bundle_attach(&ssl_client->ssl_conf)
```

- and call that method in the `ssl_init()` or `start_ssl_client()` (?)

meltdown03, 2020-06-23:
I added the modified `esp_cert_bundle.c` file to libraries/WiFiClientSecure/src. (Modified so logging would work). The diff is for the entire arduino-esp32 directory. Just drop the txt extensions.
[esp_cert_bundle.diff.txt](https://github.com/espressif/arduino-esp32/files/4820805/esp_cert_bundle.diff.txt)
[esp_crt_bundle.c.txt](https://github.com/espressif/arduino-esp32/files/4820807/esp_crt_bundle.c.txt)

meltdown03, 2020-06-23:
I changed the #defines in my platformio.ini file like this:
```
board_build.embed_txtfiles =
     x509_crt_bundle
build_flags =
     -DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG
     -DCONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_NONE=1
     -DCONFIG_MBEDTLS_CUSTOM_CERTIFICATE_BUNDLE=1
     -DCONFIG_MBEDTLS_CUSTOM_CERTIFICATE_BUNDLE_PATH=""
```

You could also edit the sdkconfig.h file that Arduino uses (`tools/sdk/esp32/include/config/sdkconfig.h`)

meltdown03, 2020-06-24:
It seems like changing the `#defines` is not necessary. I wrote a simple extra_script for PlatformIO to automatically generate the `x509_crt_bundle` file:
```
Import("env")

env.Execute("python3 gen_crt_bundle.py --input ./certs")
```

It assumes the certificates are in a `./certs` subdirectory of the project dir, and the `gen_crt_bundle.py` is in the project directory. The resulting `x509_crt_bundle`  ends up in the project directory also. I'm not that familiar with the Arduino IDE, so maybe someone else can figure out how to implement this and the embedding of binary files like PlatformIO does.

meltdown03, 2020-06-24:
> That sounds promising!
>
> Would you mind sharing your modified `ssl_client` implementation?
> I guess the solution would be
>
> * to add a method which does
>
> ```
> esp_crt_bundle_set(myx509_bundle) [...]

It's my understanding that the `esp_crt_bundle_attach()` method calls `esp_crt_bundle_set(embedded_certs)`, so it is not required to call on it's own.

kinafu, 2020-09-23:
[I modified the arduino-esp32](https://github.com/kinafu/arduino-esp32) according to the discussed / proposed changes (thx @meltdown03).

To test the modifications [I created an example sketch](https://github.com/kinafu/https-test-sketch) (as platformio project).
It's based on the [BasicHttpsClient example](https://github.com/espressif/arduino-esp32/blob/1.0.4/libraries/HTTPClient/examples/BasicHttpsClient/BasicHttpsClient.ino).

I looked into the problem for some hours now and did not get it to work unfortunately.
At runtime on the device I always get the message: 
`[E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():216]: (-12288) X509 - A fatal error occured, eg the chain is too long or the vrfy callback failed`

You can try my changes quickly by checking out [my test repository ](https://github.com/kinafu/https-test-sketch)in your editor with platformio installed.
The test sketch will compile using my changes (set in platformio.ini).
Just do not forget to enter your own `SSID `and `PASSWORD`.

The example sketch should be fine. The failure probably lies in [my arduino-esp32 modifications](https://github.com/kinafu/arduino-esp32).

kinafu, 2020-10-02:
hey `stale bot`, reopen the issue 🥺

kubo6472, 2021-01-04:
Can someone please reopen this issue? @igrr
@kedars @mahavirj @Spritetm @projectgus @me-no-dev
Thanks.

Duckle29, 2021-02-03:
poking the issue a bit so stale bot doesn't get angry. I think this is a nice feature request :)

kubo6472, 2021-02-03:
> 
> 
> poking the issue a bit so stale bot doesn't get angry. I think this is a nice feature request :)

thanks, I almost forgot.

Duckle29, 2021-03-18:
I have put together a PlatformIO project based on BasicHttpClient example like @kinafu. It uses a local copy of the WiFiClientSecure library, which I have modified like @meltdown03, however with additional modifications as the client wouldn't call start SSL without either a key or a cert being passed, or it'd return -1 immediately. An extra "useBundled" boolean was added.

This example works for me, but I'd love feedback from others.

https://github.com/Duckle29/esp32-certBundle

Onno-Dirkzwager, 2021-03-20:
@Duckle29 looking good! I test / use / study and get back.

Onno-Dirkzwager, 2021-03-23:
@yknivag @Duckle29 works very well!
I'll see this weekend if I can do my part and add the loadCertBundle(Stream& stream, size_t size); function to the modified library (by @Duckle29). Which would let us load / stream the bundle from SPIFFS SD etc.

Onno-Dirkzwager, 2021-03-26:
@yknivag @Duckle29 I made it work. We can now load the certificate bundle from SPIFFS using loadCertBundle(Stream& stream, size_t size); 

https://github.com/Onno-Dirkzwager/esp32-certBundle

It does need some work before we can offer this as a PR to espressif. Because we have introduced some compile errors if you do not use the bundle.

@Duckle29 let's tackle this in your issue section: https://github.com/Duckle29/esp32-certBundle/issues/2

Onno-Dirkzwager, 2021-04-04:
@meltdown03 based on your modifications @Duckle29 got the embedded cert bundle to work with PIO. And I added the missing loadCertBundle(Stream& stream, size_t size); function to the WiFiClientSecure library.

This has the potential to close this issue were it not that we run into the following PIO compile error when not using the embeded bundle (when using the other clientSecure options):

.pio\build\esp32dev\lib499\libWiFiClientSecure.a(esp_crt_bundle.c.o):(.literal.esp_crt_bundle_attach+0x8): undefined reference to `_binary_x509_crt_bundle_start'

This happens as soon as we include esp_crt_bundle.c/h
For you and me this is probably not really an issue as we will include it when we want it.
But for others and the library as a whole, this would be....

Any thoughts on how to prevent this?

meltdown03, 2021-04-04:
You'd have to make a new #define in sdkconfig.h most likely. Something
like "#define CUSTOM_MBEDTLS_USE_CUSTOM_CERTIFICATE_STORE", then use
#ifdef's around the changes, and add to board.build_flags in platformio.ini.

    Sorry, I haven't really been pursuing this much anymore. Busy with
other things. I can try some things next week if I get a chance.

Onno-Dirkzwager, 2021-04-04:
@meltdown03 please do! I'm interested to see what you come up with. I've failed miserably in solving this ;)

Keep in mind that we would like to make a PR that should work for Arduino users out of the box and not everyone will use the bundle.

For our personal cases this already workes. So thanks for your original post pointing us in this direction!

Onno-Dirkzwager, 2021-04-18:
poking the issue a bit so stale bot doesn't get angry.

lbernstone, 2021-04-22:
Just an fyi, and it isn't there yet in arduino, but esp-idf now includes a certificate bundle as [cert binaries](https://github.com/espressif/esp-idf/blob/master/components/mbedtls/esp_crt_bundle/cmn_crt_authorities.csv).  It is not a comprehensive set, but will get you into many cloud services, and will be 64kb of code space.

Onno-Dirkzwager, 2021-04-22:
@lbernstone not sure but I believe we are already using what you are referring to:
- For PIO @Duckle29 used @meltdown03 's modified esp_cert_bundle.c to enable using the [embeded bundle](https://github.com/Duckle29/esp32-certBundle)
esp_crt_bundle_attach
- To prevent compile error's in Arduino I added a [build flag](https://github.com/Duckle29/esp32-certBundle/issues/2#issuecomment-813111309).
- And to enable loading the bundle from FS I modified Duckl29's work [here](https://github.com/Onno-Dirkzwager/esp32-certBundle/blob/main/src/main.cpp#L55)

If this was not what you meant or we are on the wrong track please let me know!?

Otherwise, I'll try and make a PR this weekend as @Duckle29 seems too busy and does not have the time...

lbernstone, 2021-04-22:
I mean it will be included in the compiled libraries once the weird naming from esp-idf is resolved.  You will not need any external library, just an include.

Onno-Dirkzwager, 2021-04-22:
@lbernstone thank you for clarifying!
I think a perfect / easier alternative for @Duckle29 

Would you still be interested in my PR regarding client.loadCertBundle() which lets you basically load the same bundle but then from a FS? I would not mind this feature myself as this lets the user update the bundle separately OTA or with an SD card.

lbernstone, 2021-04-22:
Anything goes on external libraries.  If you make an example using one of those root CAs, then it should be usable as an example here too.

Onno-Dirkzwager, 2021-04-22:
@lbernstone thank you, Ill make a PR this weekend.

kubo6472, 2021-05-19:
Bumping the issue :)

witnessmenow, 2021-08-01:
Also interested in this feature, @yknivag has a nice approach for doing OTA based on GitHub releases that relies on this and I think it would be a really worthy addition to the esp32

zekageri, 2021-09-09:
Is there any news on this?

esp32wrangler, 2022-01-05:
I was also trying to solve this issue, and this ticket was very helpful.
However, I think the proposed solution of loading the certificates from SPIFFS is very wasteful of RAM, 64k is a large chunk of the usable heap. As far as I can tell, if the the data is embedded with the embed_txtfiles approach (https://docs.platformio.org/en/latest/platforms/espressif32.html#embedding-binary-data), then it is read directly from flash through the cache mechanism (https://www.exploreembedded.com/wiki/File:Address_map.JPG) and the only RAM needed is the index that esp_crt_bundle_init builds. 
Perhaps it would be best to just use the approach the original Espressif authors intended (see the EMBEDDED_BUNDLE == 1 section of esp_crt_bundle_attach).

esp32wrangler, 2022-01-05:
I cleaned up the changes and make a change set that may be acceptable upstream. I have zero experience with this project, so I would appreciate if someone could check if the changes follow the project norms.

https://github.com/esp32wrangler/arduino-esp32/commit/974a7d9bef7002b4ca6b1aba9ff9a920afcd270e

Twilight-Logic, 2022-04-12:
I am also interested in this capability. I am developing an IoT project which communicates with Google App Script and am unfortunately getting this error whenever I perform a GET request:

[E][ssl_client.cpp:36] _handle_error(): [data_to_read():287]: (-29184) SSL - An invalid SSL record was received

I am using the HTTPSRedirect library, which just wraps the WiFiClientSecure GET method without any modification. A client.connect() works provided that .setInsecure() is used first, however a subsequent GET request just hangs the ESP32 although I still get the above message in the Serial Monitor when 'debug' log level is selected. I am assuming at this point that the problem is that the GET request can't be performed without having a valid certificate chain in place? Whether that's down to Google increasing security, changing their cert or the library not working as it should be, I don't know. I have no problem accessing the app web page in a standard browser on the PC. Google does re-direct from script.google.com to script.googleusercontent.com hence the use of the HTTPSredirect library.

closed 2025-01-16 by Parsaabasi