espressif/arduino-esp32 #3647: too many warnings are disabled

liebman, 2020-01-20:
### Hardware:
Board:							Any
Core Installation version:			current git
IDE name:						Platform.io
Flash Frequency:					Any
PSRAM enabled:                                    no
Upload Speed:					any
Computer OS:                                        Mac OSX

### Description:
Very many warnings are disabled in platform.txt and I believe this is hiding many bugs and all warnings should be enabled.
 For instance in `WiFiClientSecure` I find https://github.com/espressif/arduino-esp32/blob/2195109ecc0c7bdfa17cc60aa5a03ae78ed87205/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp#L140-L142 
Where the parameters `pskIdent` and `psKey` are unused and similarly named object members are used instead.  I don't believe that is intended.

If PRs will be accepted to re-enable these and clean up the code I am willing to generate it.

### Sketch:  
```cpp
#include "Arduino.h"
#include "Wire.h"
#include "WiFi.h"
#include "WiFiClientSecure.h"
void setup()
{
    Wire.begin();
    WiFi.begin();
}

void loop()
{
    WiFiClientSecure x;
    x.connect("fake.host.com", 4242);
}
```
I've added the following to my `platformio.ini` to re-enable the warnings that are disabled in the core:
```
build_flags = 
    -Wall
    -Werror=all
    -Wextra
build_unflags =
    ; esp32 core disabled these warnings, re-enable them
    -Wno-maybe-uninitialized
    -Wno-unused-function
    -Wno-unused-but-set-variable
    -Wno-unused-variable
    -Wno-deprecated-declarations
    -Wno-unused-parameter
    -Wno-sign-compare
    -Wno-error=maybe-uninitialized
    -Wno-error=unused-function
    -Wno-error=unused-but-set-variable
    -Wno-error=unused-variable
    -Wno-error=deprecated-declarations
```
### Debug Messages:
```
N/A
```

me-no-dev, 2020-01-20:
we do accept PRs. You can at least try and we can discuss there.

closed 2020-04-03 by stale[bot]

liebman, 2020-04-04:
This still has an open PR against it. Please consider to reopen.